kiibohd/controller #94: ": 500 mA" displayed in debug console

mattventura, 2016-02-29:
Repro steps:
Power on keyboard
Open debug console
Press enter once
": 500 mA"  appears after the normal prompt. 
Doesn't seem to happen again until the keyboard is restarted. Looks like debug output from Output/pjrcUSB/output_com.c, but the message seems to be incomplete.

haata, 2016-02-29:
Yeah, this is an artifact of the USB debug buffer being slow. The message is fully shown in UART debug mode. This message is to indicate what the power level changed to (on plug-in, I start at 100 mA, then attempt to negotiate to 500 mA, otherwise I try 100 mA, and if that's still problematic I can go lower).

This message shows up when the power level changes (which can change dynamically if the host puts the USB link into suspend or sleep mode).

mattventura, 2016-02-29:
Out of curiosity, what do I actually need for non-USB debugging? Just a USB to UART conver attached to RX0 and TX0?

haata, 2016-02-29:
Basically, you need a UART adapter.

I use a typical USB to serial adapter + an RS232 to UART level converter (because the most convenient)
Say, this http://www.amazon.com/RS232-converter-board-Female-3-3V/dp/B005D5T292
and http://www.amazon.com/Plugable-Adapter-Prolific-PL2303HX-Chipset/dp/B00425S1H8

Another option is to use a bus pirate http://dangerousprototypes.com/bus-pirate-manual/bus-pirate-uart-guide/

Baud Rate: 115200
On Linux, screen /dev/ttyUSB0 115200 8N1

On an IC60, you'll want to connect to Tx0/Rx0, but on all other keyboards you'll want to connect to Tx2/Rx2.

You'll need to compile with the usbMuxUart Output module to enable the UART for debugging mode. However, I do have it enabled by default in the bootloader for messages only on mk20dx256vlh7 keyboards (all keyboards expect the IC60, IC60 doesn't have enough bootloader space for it).

haata, 2016-05-17:
I've removed these messages (were affecting going into sleep state).

closed 2016-05-17 by haata