kiibohd/controller #97: infinity ergodox doesn't seem to have any effect in bios and grub portions of boot on a HP z420 workstation

razamatan, 2016-03-04:
here's a log of the irc:
14:56  razamatan> infinity ergodox.  during boot, my keyboard can't trigger getting into the bios.  
                  it's only "active" during OS boot...
14:56  razamatan> is there a workaround aside from plugging in another keyboard?
14:56  HaaTa> hmm
14:56  HaaTa> so, I have gotten a few reports about this
14:56  HaaTa> there are a couple things to try
14:57  HaaTa> bios options related to usb
14:57  HaaTa> (bios may not be following the spec and requesting a boot keyboard)
14:57  HaaTa> force keyboard into boot mode
14:58  HaaTa> (I need to see if I have any problemmatic bios's around)
14:58  HaaTa> for whatever reason I tend to pick computers with less shitty ones, lol
14:58  razamatan> i'm using a hp z420.. i snooped around its bios, and there's no usb keyboard 
                  settings
14:59  HaaTa> usb legacy?
14:59  razamatan> nope..
14:59  HaaTa> ehci?
14:59  razamatan> it's a more recent workstation class machine
14:59  razamatan> nope
14:59  razamatan> unless i'm missing something...
15:00  razamatan> https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwi4_OOVkajLAhUP52MKHXdaDswQtwIIHDAA&url=https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DGOsaR8S35mg&usg=AFQjCNFND9N9P2XKwwwVPK4aryKCoLPYew&sig2=T5iRY3R6Xfc5AAuLvUDXQQ&bvm=bv.116274245,d.cGc
15:02  razamatan> around the 3m mark is probably the closest it gets to anything that could head 
                  toward usb settings
15:02  HaaTa> haha, yeah I saw that
15:02  HaaTa> nothing I can see for settings
15:02  HaaTa> so, one thing that can be done is to default the keyboard in 6KRO mode
15:02  HaaTa> then set to NKRO
15:03  HaaTa> (it might be a firmware bug)
15:03  razamatan> i'm guessing that's a KLL specific setting... no option in the web configurator?
15:03  HaaTa> but unless I have the computer in person and attach a usb logic analyzer I can't know 
              for sure
15:04  HaaTa> yeah, I haven't added a configurator option yet
15:04  HaaTa> (mostly because I'm a terrible web dev)
15:04  HaaTa> there are quite a few configuration options for the keyboard
15:05  razamatan> is there a list?
15:05  HaaTa> so, a list can be generated by looking at capabilities.kll files
15:06  HaaTa> hmm, there might not be a default setting option...shouldn't take me more than 5 
              minutes to add though
15:06  HaaTa> should be here -> 
              https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/capabilities.kll
15:06  HaaTa> you can swap between modes though
15:06  HaaTa> U"A" : kbdProtocolBoot();
15:06  HaaTa> U"A" : kbdProtocolNKRO();
15:07  HaaTa> is how the kll configuration would look like
15:07  razamatan> is the default NKRO?
15:08  HaaTa> yes
15:09  razamatan> i noticed that hitting my layer shift key did nothing
15:09  HaaTa> https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/output_com.c#L117
15:11  HaaTa> hmm?
15:11  razamatan> even in the grub screen the keyboard did nothing..
15:11  razamatan> only when the os kernel is loaded does it respond normally
15:11  razamatan> dunno if that helps anything..
15:12  razamatan> i'll have to try the boot toggle for now
15:12  razamatan> but how would that work?
15:12  razamatan> b/c whenever i rebooted, the keyboard would go into this state
15:12  razamatan> do i have to go to 6KRO before i reboot to test?
15:14  HaaTa> so
15:14  HaaTa> generally when you reboot
15:14  HaaTa> usb is reset
15:14  HaaTa> then, bios hands usb control over to the bios
15:14  HaaTa> erm
15:15  HaaTa> grub/bootloader
15:15  HaaTa> (this is why it doesn't work in grub either)
15:15  HaaTa> then
15:15  HaaTa> when the OS starts, the OS is supposed to reset the usb bus
15:15  HaaTa> which is how NKRO starts working
15:15  HaaTa> so, you should be able to flash the firmware
15:15  HaaTa> then press the 6KRO mode button
15:15  HaaTa> then the keyboard will just work
15:16  HaaTa> no usb bus resetting is required
15:16  razamatan> but does that mean i lose NKRO?
15:16  HaaTa> (I do fanciness with multiple usb descriptors to make this work)
15:16  HaaTa> temporarily
15:16  HaaTa> you can just re-enable NKRO\
15:16  HaaTa> the same way
15:16  razamatan> sorry for my slowness.. please bear w/ me..
15:16  razamatan> :)
15:17  HaaTa> haha, np, this is annoyingly complicated
15:17  razamatan> i map some key to NKRO toggling
15:17  HaaTa> because BIOS engineers suck
15:17  razamatan> probably not in a layer
15:17  HaaTa> so, layers shouldn't matter
15:17  HaaTa> because, they are handled inside the keyboard firmware
15:17  razamatan> do i even bind to a key?
15:17  razamatan> i'm pretty confused at this point..
15:18  HaaTa> I'd bind it to function key
15:18  razamatan> if there's some prewritten doc that i could read
15:18  HaaTa> yeah, for 6KRO/NKRO, not currently....
15:18  razamatan> all teh function keys are layer-1 number keys?
15:18  HaaTa> https://github.com/kiibohd/controller/issues file me a bug for writting something on 
              that :P
15:19  HaaTa> f1 is function shift 1
15:19  HaaTa> to layer 1
15:19  HaaTa> F1 is normal F1
15:22  razamatan> issue 96
15:23  razamatan> hitting f1 for me during boot does nothing
15:25  razamatan> so in layer0, defing left f1 to be kdbdProtocolBoot()
15:25  razamatan> ?
15:25  razamatan> in my kll
15:25  razamatan> compile and load
15:25  razamatan> then hit what is now kbdProtocolBoot()
15:25  razamatan> then reboot?
15:26  HaaTa> yes
15:26  HaaTa> so
15:26  HaaTa> you can test to make sure it's in 6KRO mode two ways
15:26  HaaTa> use this page http://www.gigahype.com/nkey-rollover-test-page/
15:26  HaaTa> make sure you can only press 6 keys at a time (not including modifiers)
15:27  HaaTa> the other way is to use the debug cli
15:27  HaaTa> (there's a wiki on the github page for setting that up)
15:27  razamatan> i've been in the debug cli before
15:29  razamatan> once i'm in 6kro, then i reboot..?
15:29  razamatan> won't it reinit to nkro?
15:31  HaaTa> yes
15:31  HaaTa> you can press the key whenever you want to go into 6KRO mode
15:33  razamatan> so that setting persists even w/ the usb being reset on boot..
15:33  HaaTa> also, a neat feature for the ergodox would be to have an LCD screen message when in 
              6kro mode
15:33  razamatan> yeah
15:33  HaaTa> razamatan: it may
15:33  HaaTa> it all depends on how the usb host computer does the reset
15:33  razamatan> man.. so this isn't guaranteed..
15:33  razamatan> heh
15:33  HaaTa> there's nothing I can really do about that
15:33  HaaTa> unless I want to possible destroy the keyboard
15:33  razamatan> right..
15:34  razamatan> having the lcd show the mode would help here
15:34  HaaTa> plz file github issue :D
15:34  HaaTa> this way I can tell you it's ready
15:34  razamatan> to determine if the reboot is futzing w/ the 6kro
15:34  HaaTa> yep
15:34  HaaTa> doesn't work with all keyboards
15:34  HaaTa> but it's a start
15:34  razamatan> i'm probably going to cut and paste this chat
15:34  HaaTa> yep
15:34  HaaTa> that's totes coo
15:35  razamatan> coo

Guppster, 2016-05-01:
Any update on this? I need my ergodox infinity to be in 6rko

razamatan, 2020-04-25:
the configurator has support for a 6/n-kro toggle for awhile now.  i forgot to update this issue on what i've been doing in the wild since haata doesn't have a repro.  i'll do this now.

i have been happily using this to toggle to 6kro when in bios/bootloader contexts.  whenever i need to interact in these contexts, i wait until i'm in them, then i toggle over the 6kro by using the toggle key which is mapped on layer 2 for me (so my concern about mapping this toggle in non-base layers is gone).  i then let it go on.

it turns out, as soon as the kernel/os is fully booted, it resets the usb connection such that i don't need to toggle back to nkro.  since the default on init for the ergodox is to be nkro, the reload of the usb connection during os boot sequences work fine to get it back to the default nkro state.  verified w/ various *nix and windows 10 booting.

haata, feel free to resolve this issue since there is a well supported workaround, unless you want to keep this to track coming up with a way to work around bios issues.

razamatan, 2022-12-21:
another update.  i moved to a computer that is fully non-csm uefi, and the infinity is still not mapped correctly until i switch to 6kro.  the behavior when under bios and uefi as far as nkro infinity is inconsistent between the two, but consistent individually.