kiibohd/controller #100: Firmware freezes when computer sleeps

adamsaj, 2016-04-10:
I'm using an ErgoDox Infinity with OSX, and my keyboard is freezing when my computer goes to sleep. The LCDs stay on, but when I wake my computer I can't type anything and the layer buttons don't work (they don't add layers to the LCD). I'm not noticing this behavior with my Windows computer, it's just my Mac.

csquared, 2016-05-01:
I think this is happening to me as well. Will update when I can test on another machine on Monday. Specific observation is: use mouse to wake up computer and can't type. Unplug and replug then can type. OS X El Captian, 10.11.4. Works until computer goes to sleep again. Happy to install something to observe what's happening - can use a little guidance there. Thanks!

These seem to be around this time.
```
5/1/16 3:41:49.712 PM WindowServer[189]: no sleep images for WillPowerOffWithImages
5/1/16 3:41:53.000 PM kernel[0]: IOPMrootDomain: idle cancel, state 1
5/1/16 3:41:53.421 PM loginwindow[95]: magsafeStateChanged state changed old 2 new 1
5/1/16 3:41:58.000 PM kernel[0]: USBMSC Identifier (non-unique): 000000264001 0x424 0x4063 0x191, 2
5/1/16 3:42:07.000 PM kernel[0]: 158671.903416 AppleUSB20HubPort@14121000: AppleUSBHostPort::disconnect: persistent enumeration failures
5/1/16 3:42:08.872 PM loginwindow[95]: CoreAnimation: warning, deleted thread with uncommitted CATransaction; set CA_DEBUG_TRANSACTIONS=1 in environment to log backtraces.
5/1/16 3:42:13.000 PM kernel[0]: AppleCamIn::systemWakeCall - messageType = 0xE0000340
5/1/16 3:42:13.326 PM CommCenter[260]: Telling CSI to exit low power.

```

here's what i got just searching for USB:

```
5/1/16 3:41:58.000 PM kernel[0]: USBMSC Identifier (non-unique): 000000264001 0x424 0x4063 0x191, 2
5/1/16 3:42:07.000 PM kernel[0]: 158671.903416 AppleUSB20HubPort@14121000: AppleUSBHostPort::disconnect: persistent enumeration failures
5/1/16 3:42:29.000 PM kernel[0]: 158694.103184 AppleUSB20HubPort@14121000: AppleUSBHostPort::disconnect: persistent enumeration failures
```

This time from sleep the left hand just came up in DFU mode, but saw that as a separate issue.

csquared, 2016-05-01:
On a hunch I reverted the dynamic usb support and that seems to have fixed it so far:

https://github.com/csquared/controller/tree/c2-logo-no-dynamic-usb

haata, 2016-05-17:
I've done a bunch of work on sleep again. Tested on a Mac Book Pro with El Capitan. Should be working now. Do let me know if there are still issues.

https://github.com/kiibohd/controller/commit/ba58010a7582b11b5fff2b995eb406e27fa44ecf

anvit, 2016-05-31:
I seem to be having this issue as well. I have a windows PC though and me ergodox stops responding if my system goes to sleep or is rebooted. Does the layout configuration use the latest firmware? If it does, I have the latest build and am still facing the issue. I tried playing around with XHCI settings in the bios but that doesn't make a difference.

haata, 2016-05-31:
I updated the firmware as of 7-8 hours ago. Please try that, if that doesn't work I'd like to know the make and model of the laptop (also which version of Windows).

anvit, 2016-05-31:
Yes, it is working now. Thank you!

F1LT3R, 2016-07-31:
> I updated the firmware as of 7-8 hours ago. Please try that, if that doesn't work I'd like to know the make and model of the laptop (also which version of Windows).

@haata, @anvit,  what are the steps to test this update?

doodzik, 2016-09-11:
updating the firmware fixed this issue for me on a mac.
I'm going to verify if it works on a windows machine tomorrow.

I think this issue and the related issues can be closed.

@F1LT3R you can use the webconfigurator as it uses the correct controller [version](https://configurator.input.club/stats.json)

doodzik, 2016-09-12:
I can confirm that it works under windows as well

haata, 2017-11-22:
Seems to be working now.

closed 2017-11-22 by haata