kiibohd/controller #104: Modifiers on daisy chained half not working after reboot

typorian, 2016-05-01:
Hi,
on my linux machine (Linux Mint 17), the daisy chained half does not register modifiers correctly. This means that after a reboot, I have to unplug the keyboard to be able to capitalize letters with the shift key on the daisy chained half. I can see in a keyboard testing tool that the shift key actually gets registered as such, it just doesn't do anything.
Cheers

Lknechtli, 2016-05-03:
I can confirm that this happens, it seems to occur most frequently when the keyboard is plugged in before the computer is turned on.

typorian, 2016-05-08:
In between when I posted this and now, I did a git pull once and I'm now running firmware I actually compiled myself. The issue has changed now: I am able to press enter in grub, but during boot I get a few USB related errors and the Keyboard does not work again until I unplug and re plug it.

haata, 2017-10-03:
I've fixed a lot of USB issues recently, please re-open if you still have problems after trying master.

closed 2017-10-03 by haata