kiibohd/controller #105: Modifiers not working in boot mode

cryham, 2016-05-03:
I experience not working modifiers (shift, ctrl, alt etc) when USBKeys_Protocol is 0 (Boot Mode).

So far I found that in Output\pjrcUSB\output_com.c around line 656:
`case 0: // Boot Mode`
`	// Clear modifiers only in boot mode`
`	USBKeys_Modifiers = 0;`
seems to be IMO wrong, I commented out USBKeys_Modifiers = 0;

and tried at line 359:
`	switch ( USBKeys_Protocol )`
`	{`
`	case 0: // Boot Mode`
`		// Set the modifier bit if this key is a modifier`

adding the code for modifier Release:
`	else // Release`
`	{`
`		USBKeys_Modifiers &= ~(1 << (key ^ 0xE0)); // Left shift 1 by key XOR 0xE0`
`	}`

just like in NKRO mode block but that didn't solve the issue.

eblanton, 2016-11-03:
I also have this problem in ba3c38b.  Setting usbProtocol = 0; in my KLL causes all modifiers to stop working.

cryham, 2017-01-07:
OK, now they seem working, without my modifications I had in code.

closed 2017-10-03 by haata