kiibohd/controller #109: Unable to make project in Archlinux VM

primitiveType, 2016-05-06:
Trying to get set up to build my own custom layouts, and ran into an issue that I can't find a way around. I am seeing exactly the same problem that was in this closed issue : https://github.com/kiibohd/controller/issues/13  
I downloaded the arch linux VM, logged in, did a git pull, then followed the exact instructions here:
https://github.com/kiibohd/controller/wiki/Building

```
$ mkdir build
$ cd build
$ cmake ..
$ make
```
the output is exactly the same as in the link posted above, but I'll screen cap it here as proof. I also tried changing the chip in CMakeLists.txt (and cleaned the build directory) but it made no difference.

[image]

haata, 2016-05-06:
It looks like the kll directory might be stale.

cd kll
git pull

then try again.

primitiveType, 2016-05-06:
Thanks, I think that did it. I also had to clean the build directory and redo the cmake.

closed 2016-05-06 by primitiveType