kiibohd/controller #116: Bug with sequences involving Shift keys to toggle CapsLock

nomaded, 2016-05-16:
With the Ben Blazak and TMK firmwares for the original ErgoDox, I was able to toggle CapsLock while holding down both LShift and RShift. On the ICED, with KLL 0.3d, I was not able to get the combination of U"LSHIFT" + U"RSHIFT" : U"CAPSLOCK"; to work at all. Is this a bug?

I ended up changing it to two sequences of hold LShift then tap RShift, and hold RShift then tap LShift to toggle CapsLock. This seems to work pretty well, except I seem to be hitting a different bug where if I tap (hit then release) LShift then tap RShift (and same in the other order), it engages CapsLock, but repeating the same set of taps does not release CapsLock. At this point, I need to hold LShift and then tap RShift (or the other order) in order to engage CapsLock again before I can hold LShift and tap RShift to disengage CapsLock.

-----

TL;DR:
U"LSHIFT" + U"RSHIFT" : U"CAPSLOCK"; doesn't work.

Therefore, I have defined:
U"LSHIFT", U"RSHIFT" : U"CAPSLOCK";
U"RSHIFT", U"LSHIFT" : U"CAPSLOCK";

Hit and release LShift, then hit and release RShift will turn on CapsLock, but repeating will not turn it off. It is necessary to Hold LShift, then hit and release RShift to turn on CapsLock again, and repeating the sequence to turn off CapsLock.

nomaded, 2016-05-21:
This is even weirder. It seems the combination doesn't work on layer0, but works on all my other layers.

In the ergodox.bash file for compiling the firmware, I have:

DefaultMap="layer0 lcdFuncMap"
PartialMaps[1]="layer1 lcdFuncMap"
PartialMaps[2]="layer2 lcdFuncMap"
PartialMaps[3]="layer3 lcdFuncMap"
PartialMaps[4]="layer4 lcdFuncMap"

The following combination is defined only in layer0:

U"LSHIFT" + U"RSHIFT" : U"CAPSLOCK";

This combination works in layers 1-4, but not on layer0. What am I doing wrong?

haata, 2016-05-31:
Hmm, this may well be a bug. Have you tried mapping LSHIFT and RSHIFT to one hand of the ergodox and trying this? (I'm trying to make sure it's not an interconnect bug)

Which OS are you using? And I'm assuming you are using the interconnect instead of connecting both sides directly to the computer?

nomaded, 2016-06-03:
I have tried mapping both LSHIFT and RSHIFT to two different keys on the right hand and the combination still doesn't work. I have the right side connected to my MacBook Pro, and the left side is connected to the right.

haata, 2016-06-03:
Interesting, ok this is definitely a bug, I'll look into it.