kiibohd/controller #117: Feature request: LCD indicator of CapsLock

nomaded, 2016-05-16:
Any thoughts on having CapsLock indicated in the LCD screens, both as a graphic, and with the colored backlighting?

I had a brief glance over the code for controlling the LCD screens for indicating layers, but I haven't dug down into the logic to figure out everything that I would need to do to add CapsLock support into it. It would be nice to have the graphics on the LCD shift over with "CAPS" much like how it does for the layers.

ckaros, 2016-10-05:
I have been able to accomplish this by assigning a blank last layer ( layer 6 for me) and setting the lcd image to a C as well as boosting the back light brightness. It is fairly straightforward. My kll line for capslock is 

U"TAB" : U:"CAPSLOCK" + layerLock(6) + LCDLayerDisplay()

The LCD image and color are set in the lcdFuncMap.kll file.

The hex matrix for a 32x32px letter c:
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x80, 0xc0, 0xc0, 0xe0, 0xe0, 0x60, 0x60, 0x60, 0x00, 0x60, 0x60, 0x60, 0x60, 0xc0, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0xf8, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x0f, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x03, 0x03, 0x07, 0x06, 0x06, 0x04, 0x00, 0x04, 0x06, 0x06, 0x07, 0x07, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,

bkerley, 2016-11-08:
I got this working on my setup, and it's pretty nice, especially since [I made the caps-lock state of the LCD a skull.](https://twitter.com/BonzoESC/status/796103584039501824)

The downside is that Mac OS X requires the caps-lock key be held down for longer than a normal key to register a key press, which means that the alternate layer may get out of sync with the caps-lock state. Fixing this a way that's compatible with the caps-lock LED state (readable using the debug console on the keyboard) is going to be more involved.