kiibohd/controller #121: Make the mouse and joystick interfaces optional

bburky, 2016-05-27:
It doesn't currently seem possible to completely disable the mouse and joystick interfaces including their USB descriptors. Please make it possible to disable them. An ifdef would be enough.

I had issues with (I think) the mouse or joystick interface before. Steam on Windows was continuously acting like I was pressing the right button on a joystick (I think). That was with an old build, I'll report it as a separate issue if I reproduce it.

I can't seem to get the uartOut OutputModule to work (usbMuxUart works fine) but it may make sense to not have a CDC serial port USB device then.

If I don't plan on using an interface at all, it would be nice to just disable it. Possibly this could help computers confused by the current list of interfaces. This would also help things like #60.

haata, 2016-05-27:
The Joystick and Mouse is reasonable, however the CDC interface is for the virtual serial port interface which is completely separate from the usbMuxUart debug interface.

I'll look into how to get them disabled, however this is a big change because of the way USB descriptors and you then have to test compatibilty of every single combination for every OS which is already very time consuming with my limited set of devices. If everyone actually followed the USB spec there wouldn't be issues like always going right (I'm not using the Joystick endpoint atm, so it's definitely a bug on the OS side).

bburky, 2016-05-27:
Removing the CDC interface isn't that important. I was just mentioning it for completeness. I can't see any situation when it might cause a problem.

haata, 2016-05-27:
Nah, it's cool.

For KLL 0.6 I'm going to have to figure out dynamically generated Mouse/Joystick USB HID descriptors, which will be a bunch more complicated.

haata, 2016-05-27:
Screw-it, I'm on a USB kick.
Bbl, adding all the things.

Hopefully will have something by the end of long-weekend.

haata, 2016-05-31:
https://github.com/kiibohd/controller/commit/358d480ea9862106f851cbb68c1d68da7369ddd5

Aaaand, there you have it. Everything is now optional!
I've also tested the different combos, at least the ones that tend to die the most easily and it seems to work fine. Mac OSX boot screen gave me the most trouble as per usual, but I got that fixed up.

I've also disabled Joystick by default (because it's not being used).

bburky, 2016-06-11:
Oh I meant to say thanks for this a while ago. This looks great.

You can probably close this issue now.

closed 2017-10-03 by haata