kiibohd/controller #123: Macro for letter ñ (and other international characters)

YamiFrankc, 2016-06-01:
Hello, I am trying to add a macros to be able to write the letter ñ but I am not able to do it.
In Windows you are able to do it by pressing ALT+164 in the numpad, so I added this to my settings:
`U"N"  :   U"ALT" + U"P1",U"P6",U"P5";`

But it doesnt work, it only does ALT+1, and 65.
Is there a way to do this?

haata, 2016-06-01:
Oh, you are trying to use the numpad numbers.

Try this instead: `U"N" : U"Alt" + U"1", U"6", U"5";`

YamiFrankc, 2016-06-01:
That still doesnt work.
It prints 65.
It looks like what is happening is that the keyboard sends ALT+1, 6 and 5 as individual parts of the sequence instead of a sequence of 165 while holding ALT

haata, 2016-06-01:
Hmm, try this:

`U"N" : U"Alt" + U"1", U"Alt" + U"6", U"Alt" + U"5";`

If this doesn't work, it's definitely a bug (I do something special for repeated keys in a sequence, but I'm not sure I exclude modifiers).

gromwalh, 2016-06-01:
I could be wrong, but I thought alt codes only worked with the numpad.  Wouldn't they want this:
`U"N" : U"Alt" + U"P1", U"Alt" + U"P6", U"Alt" + U"P5";`

I just tried manually typing this on my ergodox, first with the standard numbers, then with numpad numbers.  The standard numbers didn't work, but the numpad numbers did.

haata, 2016-06-01:
Oops, yeah, I don't really use Alt codes so I wouldn't know....

gromwalh, 2016-06-01:
Follow-up question: Would this
`U"N" : U"Alt" + U"P1", U"Alt" + U"P6", U"Alt" + U"P5";`
simulate lifting the alt key in between each number?  Because if so, you won't get Ñ, you'll get ☺♠♣.

haata, 2016-06-01:
Hmm, as I implemented it, yes it would. KLL 0.5 would make this much easier.

But I had an idea. Time to use some more obscure syntax, lol (also simpler).

```
U"N" : U"P1", U"P6", U"P5";
U"N" :+ U"Alt";
```

Ok, so what this means. First setting the sequence P1, P6, P5 to the n key. Then appending Alt to the n key as well. This means both of the macro results will be played at the same time (order does not matter in the same file).

This should work now, if not, it's a bug which I need to fix.

YamiFrankc, 2016-06-01:
Unfortunately that still doesnt do it, it prints "☺65"

haata, 2016-06-01:
kk, I shall look into this.

nsuffield, 2017-05-15:
I just wondered if there had been any progress on this. I have had similar problems trying to get German characters on an English keyboard.

`U"U" : U"LALT"+U["P0","P2","P5","P2"]; >>>> ♣ (like alt+5)
U"U" : U"LALT"+U"P0"+U"P2"+U"P5"+U"P2"; >>>· (no idea what that is)
U"U" : U"ALT"+U["P1","P2","P9"];>>> ☺ (like alt+1)`

YamiFrankc, 2017-05-16:
Not much, I do not use that letter that much so I stopped trying.
Earlier this year I got a Genovation CP48 that is really easy to configure and I just put my special characters( Ññ¿¡) there.

TheOtherDave, 2017-05-16:
> On May 15, 2017, at 04:14, Neil Suffield <notifications@github.com> wrote:
> 
> I just wondered if there had been any progress on this. I have had similar problems trying to get German characters on an English keyboard.
> 
> U"U" : U"LALT"+U["P0","P2","P5","P2"]; >>>> ♣ (like alt+5) U"U" : U"LALT"+U"P0"+U"P2"+U"P5"+U"P2"; >>>· (no idea what that is) U"U" : U"ALT"+U["P1","P2","P9"];>>> ☺ (like alt+1)
> 
I was under the impression that this couldn't be done solely with the keyboard (that is, without some sort of help from the OS) because the USB HID spec for keyboards pretends Unicode doesn't exist. Is that not the case? Because I'd *love* to put a "mathematical symbols" layer in my ergodox.

- Dave Sweeris

nsuffield, 2017-05-16:
Dave - as far as my understanding is, yes, you are correct that the USB HID spec doesn't allow for unicode.

However, it is possible to use alt codes for lots of symbols and characters (which you probably already know about but I'll post here for reference: https://tools.oratory.com/altcodes.html). In fact, I use them all the time for German, even though I use an English keyboard. However, it is a bit cumbersome to type them and I do have the occasional mistypes.

The fact that I can get the symbols in my example above suggests that it is almost working, but somehow the keyboard is producing the first character it comes to, rather than waiting for the complete sequence/combination to be completed. 

It's a shame, because I think it could me very powerful, and it also looks like it is almost working.

nsuffield, 2017-05-16:
@YamiFrankc - That's a good solution, but I don't need another keyboard on my desk!!!!
☺  (alt+1)

TheOtherDave, 2017-05-16:
> On May 16, 2017, at 01:04, Neil Suffield <notifications@github.com> wrote:
> 
> Dave - as far as my understanding is, yes, you are correct that the USB HID spec doesn't allow for unicode.
> 
> However, it is possible to use alt codes for lots of symbols and characters (which you probably already know about but I'll post here for reference: https://tools.oratory.com/altcodes.html). In fact, I use them all the time for German, even though I use an English keyboard. However, it is a bit cumbersome to type them and I do have the occasional mistypes.
> 
> The fact that I can get the symbols in my example above suggests that it is almost working, but somehow the keyboard is producing the first character it comes to, rather than waiting for the complete sequence/combination to be completed.
> 
> It's a shame, because I think it could me very powerful, and it also looks like it is almost working.
> 
Huh... I've heard of alt-codes, but never used them, and it didn't click that that's what you were trying to do. I know they won't work on macOS because their keyboards don't even have an "alt" key (Mac keyboards have "control", "option", and "command" modifiers). Does anyone know if alt-codes work on Linux? Or if that list is expandable?

- Dave Sweeris

nsuffield, 2017-05-16:
Looking here [https://ubuntuforums.org/showthread.php?t=2000359] it looks like it is available on Linux, but I haven't tried it. It appears that alt isn't available on Mac [http://www.wikihow.com/Type-Symbols-Using-the-ALT-Key]. For info, this list of unicodes also looks more complete [https://en.wikipedia.org/wiki/List_of_Unicode_characters]  

(For info, you just have to make sure that you prefix some of the codes with '0' ... and you must also use the keypad numbers, not the top row, at least on Windows anyway. Hence the use of U"P1" above, rather than U"1").

TheOtherDave, 2017-05-16:
Sent from my iPhone
> On May 16, 2017, at 02:41, Neil Suffield <notifications@github.com> wrote:
> 
> Looking here [https://ubuntuforums.org/showthread.php?t=2000359] it looks like it is available on Linux, but I haven't tried it. It appears that alt isn't available on Mac [http://www.wikihow.com/Type-Symbols-Using-the-ALT-Key].
> 
Those characters are tied to the selected keyboard layout file, though... Dunno how it works on Windows (or Linux), but on macOS, the keyboard layout file itself has separate layers for pressing keys with various modifiers held down. It just so happens that Apple's default QWERTY layout has those Unicode characters defined on the opt-layer... when you type "option-8", that's what the keyboard sends, and the OS then says "oh that means •". It's still OS-dependent, and, AFAIK, not part of the QWERTY "standard".

I'm not criticizing you or anyone else here (unless you're on the USB HID committee*), just bemoaning the lack of any standard or cross-platform way to input non-ascii characters.

- Dave Sweeris

* I mean, really, Unicode predates USB HID by seven years... I get that some embedded systems and such don't have the capability to handle non-ascii characters, but couldn't they have at least made a "unicode"/"non-ascii" keyboard spec in addition to the basic spec for everyone else?

nsuffield, 2017-05-17:
Nope, not part of the USB HID, and if I was, I'd probably be the guy that brings the tea and biscuits and tidies up the chairs after the meeting. I feel your pain, and it looks like you've been concerned about this issue for a while now [https://github.com/kiibohd/KiiConf/issues/30], but if this could work, it would be a stop-gap until something moves within USB HID.

TheOtherDave, 2017-05-17:
> On May 16, 2017, at 23:49, Neil Suffield <notifications@github.com> wrote:
> 
> Nope, not part of the USB HID, and if I was, I'd probably be the guy that brings the tea and biscuits and tidies up the chairs after the meeting. I feel your pain, and it looks like you've been concerned about this issue for a while now [https://github.com/kiibohd/KiiConf/issues/30], but if this could work, it would be a stop-gap until something moves within USB HID.
> 
Haha, I'd forgotten about that thread! Yeah, it's been bugging me ever since Apple released Swift and I realized there wasn't really an easy way to type Unicode (Swift supports Unicode operators/function names). With that in mind, if the computer industry would notice that it's not the 1980s anymore WRT input devices, it'd become practical (as opposed to merely "possible") to write math code in the normal mathematical nomenclature. 

- Dave Sweeris

haata, 2017-10-03:
I'm working on something that will support this: https://github.com/hid-io/hid-io