kiibohd/controller #124: Hanging keys/resetting still after "Major USB update" commit

mkke, 2016-06-01:
My Ergodox is resetting itself (causing Windows to do the driver reinstall dance) and has hanging keys (e.g. backspace) even with the changes from the "Major USB update" commit. The PC doesn't sleep in between, it happens right while im writing something. Local modifications are only a different layout and LCD brightness changes.

haata, 2016-06-01:
How often does the reset occur?
Are you able to type for any length of time?
Have you tried a different USB cable?
Have you tested NKRO? https://www.microsoft.com/appliedsciences/KeyboardGhostingDemo.mspx
Can you access the keyboard cli? https://github.com/kiibohd/controller/wiki/Debugging

mkke, 2016-06-02:
It usually happens after a few minutes typing.
I have two complete Ergodox pairs with the cables that came with them, and it's happening with both of them. I also tried different USB ports, only the PC is the same.
I did the NKRO test, and it registers as many keys as I can hold down.

I can access the CLI with putty via USB; I'm not really sure what is helpful for you, so I'm attaching some output I have captured.

: outputDebug
: DEBUG - NKRO USB: 00 010000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 010004000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090004000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 080004000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 080000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000001000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 600000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 600200000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 680200000000 0000000000000000000000000000 00 000000000000
WARNING - USB Transmit Timeout...auto-restart disabled
WARNING - USB Transmit Timeout...auto-restart disabled
WARNING - USB Transmit Timeout...auto-restart disabled
[...]

and another one:
[...]
DEBUG - NKRO USB: 00 200000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200000000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 280000000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 280004000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 290004000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090004000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090404000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090404000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090C04000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 080C04000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 080C00000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000C00000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000400000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000C00000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000400000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200400000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200000000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200000000000 0000001000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200400000000 0000001000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 200400000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000400000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000404000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 010404000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090404000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090004000000 0000000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 090004000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 010004000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 010000000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000000000000 0800000000000000000000000000 00 000000000000
DEBUG - NKRO USB: 00 000800000000 0800000000000000000000000000 00 000000000000
WARNING - USB Transmit Timeout...auto-restart disabled
WARNING - USB Transmit Timeout...auto-restart disabled
WARNING - USB Transmit Timeout...auto-restart disabled
[...]

The two resets above came within 2 minutes, now I've been typing for 5 more without problems.

haata, 2016-06-03:
Hmm, this is definitely due to something occurring on the USB bus. (thanks for the log, it's very helpful).

Some possibilities

1. Some odd USB power saving event (there's an event for this, I'm planning on adding the cli message back in a future commit). Though, this is unlikely because the resume code would be triggered while you're typing. And, the USB host should only request suspend if the usb comm link has been quite for a number of seconds (you're typing and getting debug messages). But I wouldn't rule this out. Try to see if you can disable USB power saving.

2. Power usage. Do you have through-hole LEDs on your ergodox? If you only plug in one half does this still happen? I've seen some reports of USB controllers that can't produce enough power. The Ergodox should only be drawing around 200-250 mA based on my measurements. I request 500 mA in the USB descriptor, which should be enough for 4 halves. Since the keyboard initialized, the power budget (if one is being kept track of) should be ok. Easy thing to try, turn off the LEDs and see if you run into this issue.
Another option is to grab a decent powered USB hub, then you can burn your retinas to your heart's content :P 

Also, try the usbInitTime command after each of these events, this will let you know if the host has re-initialized the keyboard (the time and # of ticks will increase if the host requests USB configuration data).

troy-schellenbacher, 2016-06-06:
TLDR: Had similar problems to reported ones here. Took my powered USB hub out of the loop and it seems to be working a lot better. Will post again if the issue reoccurs.

Not sure if this is helpful, but I'm seeing similar issues with I version that I downloaded and flashed a few minutes ago. 

I was connected through a powered USB hub (Vantec UGT-PH217-BK) and don't have LEDs installed. Also, the system that I'm connected to is a laptop (an older HP with Window 7) and I have the same issues whether I've got the USB hub connected to a USB port on the dock or on the side of the laptop itself. In my case, the keyboard was resetting similarly to the logs above every few minutes (3-5).

I gave the ghost test a shot and don't see anything strange there.

I tried bypassing my USB hub and plugged directly into the laptop dock. So far, things seem a lot better. It's probably been the better part of an hour since I had an issue while I was having very frequent problems before.

I'll try to do a bit more testing with and without the hub and post an update if I find anything interesting.

mkke, 2016-06-07:
Sorry for the delay, I can only try it on the pc at work...
I've tried disabling USB power saving in the extended energy settings, but it didn't make a difference.
I have no LEDs, and the ergodox is connected to a port on the PC itself.
One unusual thing is that I have a Digitus USB 3.0 to DVI Adapter connected.

mkke, 2016-06-09:
I've captured more output with usbInitTime, but the keyboard didn't reset (now I'm not sure if it ever did with the new firmare).

: usbInitTime
INFO - USB Init Time: 13307 ms - 43 ticks
: WARNING - USB Transmit Timeout...auto-restart disabled
WARNING - USB Transmit Timeout...auto-restart disabled
WARNING - USB Transmit Timeout...auto-restart disabled
usbInitTime
INFO - USB Init Time: 13307 ms - 43 ticks

haata, 2017-10-03:
I've done a lot more USB updates since. Would you mind testing?

haata, 2019-02-18:
Closing unless this issue still persists with the latest firmware.

closed 2019-02-18 by haata