kiibohd/controller #128: Wiki on OS driver logs

haata, 2016-06-20:
We need a wiki for debugging system events for the various OSs.

- lsusb -v on Linux
- dmesg on Linux
- system log? on Windows
- Console? on macOS

Info that's needed.
- Insertion/removal events (or something close to it)
- USB device tree
- Possibly some indication of the driver that was loaded

Please use this bug to collect various utilities. I'll post utilities and commands as I find them.

bburky, 2016-06-20:
For macOS, System Profiler or `system_profiler SPUSBDataType` on the command line gives some USB device tree information. The USB Prober app has more info than either of those, but you have to [download some iOUSBFamily tools](http://superuser.com/questions/781982/how-can-i-install-usb-prober-from-the-developer-sdk-on-mac-os-x) to get it. It's not in the standard Xcode install.

haata, 2016-07-03:
Started the page here -> https://github.com/kiibohd/controller/wiki/Reporting-Issues