kiibohd/controller #129: Port to Raspberry Pi

sigboe, 2016-07-06:
Would it be feasible to port the controller firmware to the Raspberry Pi? To do scan on the GPIO and then run it as a program or keyboard driver in Linux? Or then connect the raspberry further to another computer to use as a USB HID? **Please** just close the ticket if it is not feasible.

There are so many cool project that come to mind if this was a thing. Every thing from a 40% keyboard with a Pi Zero, having an integrated battery, just plug in a monitor. To a full size keyboard running Ubuntu straight on the keyboard.

haata, 2016-07-06:
In theory it should be possible to port to a Raspberry Pi. Would likely require some sort of kernel module to handle custom keyboard input though. The scan and macro portions should be pretty straight-forward though.

sigboe, 2016-07-06:
The project that came to mind was making a very simple case like [your case right here](https://input.club/wp-content/uploads/2015/07/massdrop-infinity-side-2.jpg) CNC it and bend the corners. And the case would be extended to fit a Pi B+, A+ 2B or 3B with HDMI and Audio jack hole on the front, mounted with motherboard standoffs. Then butchering a [GPIO cable](https://www.adafruit.com/product/1988) just opening it up to break out the GPIO pins to the rows and columns. And using [DC converter](https://www.adafruit.com/products/2465) along with any Lipo battery for a instant 65% keyboard computer, just plug a monitor. :smile:

wpc009, 2016-10-14:
A raspberry pi is way too powerfull for a keyboard. Shotting flies using a canon.

wpc009, 2016-10-14:
I say adopting to STM32 or some cortex-m SoC with native BLE4.0 support is the right thing to do.

wpc009, 2016-10-14:
Also a raspberry pi comsumming much more power and the it require more complex manufacturing techs.