kiibohd/controller #131: Text Macros not working

M1K3L08, 2016-07-24:
When assigning a text macro to a key by manually editing a kll file, the macro will not activate when the key is pressed.

haata, 2016-07-24:
Please give a short kll example.

M1K3L08, 2016-07-24:
U"F13" : 'myemail@test.net';

When pressing F13, this is not printed

dfaught, 2017-10-10:
I experience an issue very similar to this one.  The following macros do activate when the key is pressed but not all of the result is delivered.  For example, I have these macros defined:

U"T" : 'ptype ';
U"E" : 'explore ';
U"W" : 'whatis ';

The output received when activating "T" is "ptye ".  Notice the second 'p' is stripped.  The "E" macro has similar behavior, the last 'e' is stripped.

The "W" macro works as expected, but it does not duplicate any letters within the result.  Is there some sort of "debounce" algorithm at work which is preventing all of the characters in the result from being processed.