kiibohd/controller #133: infinity md1.1 media keys stop working after reboot in windows.

b0lly, 2016-07-28:
Media keys stop working every time after reboot and I must re-plug the keyboard for them to work again.  Tested in vista and win 10.

haata, 2016-07-28:
Sounds like your BIOS is being too aggressive and doing USB handoff for EHCI devices.
Sometimes there's an option to disable this.

Basically, Windows 10 isn't re-initializing the keyboard so it still thinks it's in BIOS mode (which doesn't have Media key support right now). Also, you don't have NKRO support when it's like this (only 6KRO).

Sadly, this is Windows 10 not following the USB spec...and there's not a whole lot I can do.

However! There is a simple fix that you can do every time you reboot.
Assign a key (usually on a function layer) to 6/N-KRO. This will toggle between 6KRO and NKRO mode. Much nicer than having to unplug USB.

b0lly, 2016-07-29:
Are you sure this isn't a firmware problem? My easyAVR based board does not seem to have any issues.

haata, 2016-07-29:
Is it NKRO?

haata, 2016-07-29:
Anyways, it is possible for me to do some odd things to hack around this. Where I still remain in 6KRO mode but send media keys from the NKRO USB descriptor.
But it not really guaranteed to work, and it's not the real problem. Which is that Windows 10, didn't re-initialize the USB device as the HID spec requires.

b0lly, 2016-07-29:
Yes it's NKRO and works in Bios. (https://deskthority.net/wiki/Easy_AVR_USB_Keyboard_Firmware)

And, just to double check I tried with an old thinkpad running linux and the problem is the same. So it does appear to be a firmware problem not windows.

haata, 2016-07-29:
You know, you're right. I'll fix this right now.

Not sure I can fix the NKRO thing without breaking BIOS support.

haata, 2016-07-29:
https://github.com/kiibohd/controller/commit/9309889fad2fd31518eff6fded681d1281d66dd9

This fixes the media keys (or at least it should). Doesn't fix the NKRO thing, which I can't debug unless I have your specific computer (I don't have any computers with this problem).

b0lly, 2016-07-29:
It was just media keys not working after reboot, I'm not sure what NKRO thing you are referring to. Maybe you've got me confused with someone else with similar issues?

haata, 2016-07-29:
Nah, what's was happening, was that I only enable media keys in NKRO mode. If the keyboard was in boot/6KRO mode media keys wouldn't work.
I've just removed this restriction. So it's not a 100% fix, but it should work for you.

haata, 2017-10-03:
Please try a recent firmware, many of the NKRO vs. 6KRO issues have been fixed.

closed 2017-10-03 by haata