kiibohd/controller #134: usb_desc.c:44:19: error: expected expression before ')' token  #define LSB(n) ((n) & 255)

littletinyfish, 2016-07-29:
I'm very gingerly working my way through installing everything required to compile KLLs on Windows 10.

Seems like I'm nearly there, but now when I type "make" I keep getting stuck at 50%.

The output (with errors) is below.

```
$ make
[  4%] Generating KLL Layout
fatal: Not a git repository (or any of the parent directories): .git
Scanning dependencies of target kiibohd.elf
[  8%] Building C object CMakeFiles/kiibohd.elf.dir/main.c.obj
[ 12%] Building C object CMakeFiles/kiibohd.elf.dir/Lib/mk20dx.c.obj
[ 16%] Building C object CMakeFiles/kiibohd.elf.dir/Lib/delay.c.obj
[ 20%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/ISSILed/led_scan.c.obj
[ 25%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/MatrixARM/matrix_scan.c.obj
[ 29%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/WhiteFox/scan_loop.c.obj
[ 33%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.obj
[ 37%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/result.c.obj
[ 41%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/trigger.c.obj
[ 45%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/output_com.c.obj
[ 50%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.obj
C:/cygwin64/home/jason/kiibohd_controller/Output/pjrcUSB/arm/usb_desc.c:44:19: error: expected expression before ')' token
 #define LSB(n) ((n) & 255)
                   ^
C:/cygwin64/home/jason/kiibohd_controller/Output/pjrcUSB/arm/usb_desc.c:63:2: note: in expansion of macro 'LSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
  ^
C:/cygwin64/home/jason/kiibohd_controller/Output/pjrcUSB/arm/usb_desc.c:45:20: error: expected expression before ')' token
 #define MSB(n) (((n) >> 8) & 255)
                    ^
C:/cygwin64/home/jason/kiibohd_controller/Output/pjrcUSB/arm/usb_desc.c:63:20: note: in expansion of macro 'MSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
                    ^
make[2]: *** [CMakeFiles/kiibohd.elf.dir/build.make:316: CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:68: CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

Thanks for your help.

zassen, 2016-08-02:
I meet this error too.
Do you download the codebase by zip file ?
there is may some end symbol to be add when file unzip.

how  I did fix this error:
git clone this codebase in cygwin64.

littletinyfish, 2016-08-02:
Actually, the first time I downloaded it as a ZIP. The second time I cloned it with Git.

I thought my issue was that the folder was named something other than `controller`.

Thanks for the hint.

haata, 2016-08-31:
Turns out ancient git is to blame here. Fixed.

b7b7fd1ee724b8831bb9a870efa76a26a8d080a8

closed 2016-08-31 by haata

p-callaghan, 2016-08-31:
I can confirm that the fix works now.

Thanks for the help.

Peter

IMPORTANT NOTICE: This message is intended for the individual or entity to which it is addressed and may contain information that is privileged, confidential, and/or exempt from disclosure under applicable law. If you are not the intended recipient, you are hereby notified that copying, 
forwarding or other dissemination or distribution of this message is prohibited and that taking any action in reliance on the content of this message is to be avoided. Should you receive this e-mail in error, please notify the sender immediately via e-mail or call (403) 699-4000 and delete this message from your system. Thank you. 
************************************************************