kiibohd/controller #135: Windows setup doesn't mention need for kll.py

littletinyfish, 2016-07-29:
In order to run `make` you need `kll/kll.py` in your `kiibohd_controller` folder. However, this isn't mentioned anywhere in the Windows Setup Wiki.

I'm not sure what other dependencies are required for `kll.py` to run, so the Wiki might recommended you download the kll compiler into that directory. 

Related: During initial setup, I wasn't sure where the `kiibohd_controller` folder should go, so the Wiki might recommend it be placed in the `gygwin64\home\my_name` folder to make things easier.

So maybe:

- install cygwin
- install cmake
- kiibohd controller
- kll compiler
- serial drivers
- etc.

haata, 2016-07-31:
Hmm, I wonder. The build process expects git to be installed on the system (and in the path).
Perhaps CMake couldn't find it?

The CMake configuration does label it as required, so I'm not sure how the build proceeded. The kll compiler should clone automatically if it isn't found.
Do you, by chance have any build logs? Or able to reproduce it?
(I'm setting up another computer with Cygwin right now to see if there are any build issues)

littletinyfish, 2016-08-01:
I just tried cloning the controller with Git this time, but it still did not clone the `kll` folder.

In cmd:
`git clone "https://github.com/kiibohd/controller.git"`

Output:
```
Cloning into 'controller'...
remote: Counting objects: 4442, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 4442 (delta 0), reused 0 (delta 0), pack-reused 4437 eceiving objects: 100% (4442/4442), 1.45 MiB | 2.88 MReceiving objects: 100% (4442/4442), 1.55 MiB | 2.88 MiB/s, done.

Resolving deltas: 100% (2851/2851), done.
Checking connectivity... done.
```

haata, 2016-08-01:
It's not part of the clone.

It's part of the cmake process -> https://github.com/kiibohd/controller/blob/master/Lib/CMake/kll.cmake#L22

littletinyfish, 2016-08-01:
Okay, I think the problem was that I had changed the `controller` folder to `kiibohd_controller` ("controller" felt like too generic a name and being new to this, I didn't want to lose track of where I was).

When I re-cloned the `controller` directory and left it as-is, re-edited the CMakeLists.txt, ran wincmake and make, it compiled everything successfully.

littletinyfish, 2016-08-01:
This probably explains why my build was failing in this issue as well: #134

littletinyfish, 2016-08-05:
A few other suggestions to make the installation Wiki more clear for novice users:

---
When installing cygwin, there will be an option to select packages to install. In the search bar, search for the following packages. For each, click the word "Skip" (to un-skip) and make sure the Bin box is ticked.

make (in the Devel folder)
git (in the Devel folder)
cmake (in the Devel folder)
gcc-core (in the Devel folder)
gcc-g++ (in the Devel folder)
libusb1.0 (in the Libs folder)
libusb1.0-devel (in the Libs folder)
Python3 (in the Python folder)
ctags (in the Devel folder)

---
Clone or extract the Kiibohd controller folder to cygwin64\home\my_name.

---
Updating and loading layouts requires the kii-dfu. Maybe link out to here if users haven't already done so? https://github.com/kiibohd/KiiConf/wiki/Update-or-load-layouts-(Firmware)

haata, 2017-10-03:
Yep, I need to revisit the setup guide.