kiibohd/controller #138: dfu hangs, leaving ergodox in flash state

F1LT3R, 2016-08-02:
Steps to reproduce:

```bash
cd keyboards
./infinity.bash
cd IC60.gcc
# Put left Inifinity Ergodox controller into Flash (red LCD on left)
./load
```

I get the same effect when using `./load` as I do using `dfu-util -D kiibohd.dfu.bin`.

Output:

```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1c11
Match product ID from file: b007
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0

dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        28576 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
```

- The terminal hangs.
- The left LCD just hangs in red, keyboard is now unusable.
- After unplugging the keyboard and plugging it back in, it is still red as if it is ready to flash.

This behavior does not happen when using the bin file from the [Input Club Configurator](https://input.club/configurator-ergodox)

Any ideas what I am doing wrong here?

haata, 2016-08-02:
Oh, it sounds like you flashed the wrong firmware.

Use the `./ergodox.bash` script instead.

F1LT3R, 2016-08-02:
Ok thanks, I will give that a shot later.

F1LT3R, 2016-08-03:
Ok thanks @haata I am able to flash now. Do I have to flash each side separately with this repo, plugging each side in individually? (I ask because the Input Club Configurator seems to flash both at the same time.)

closed 2016-08-03 by F1LT3R

jceaser, 2017-09-29:
Oldish post, but if your here from google, @haata's Aug 2 comment fixed this for me.