kiibohd/controller #139: bootloader build fail

zassen, 2016-08-02:
I enter Bootloader folder and modify the CMakeLists.txt file.  Select the mk20dx256vlh7 chip.
after that, I enter Builds folder and rum below command:
cmake ..
make

finally. I get fail message about Linking C executable kiibohd_bootloader.elf.

-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Bootloader Source Files:
main.c;dfu.c;dfu.desc.c;flash.c;kinetis.c;usb.c;debug.c
-- Found Git: /usr/bin/git (found version "1.9.1") 
-- Found Ctags: /usr/bin/ctags (found version "5.9~svn20110310") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/heaven/Work/kiibohd/controller/Bootloader/Builds
heaven@ubuntu:~/Work/kiibohd/controller/Bootloader/Builds$ make
Scanning dependencies of target kiibohd_bootloader.elf
[ 10%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/main.c.o
[ 20%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/dfu.c.o
[ 30%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/dfu.desc.c.o
[ 40%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/flash.c.o
[ 50%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/kinetis.c.o
[ 60%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/usb.c.o
[ 70%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/debug.c.o
[ 80%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/home/heaven/Work/kiibohd/controller/Lib/mk20dx.c.o
[ 90%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/home/heaven/Work/kiibohd/controller/Lib/delay.c.o
Linking C executable kiibohd_bootloader.elf
/home/heaven/Work/kiibohd/controller/Bootloader/../Lib/mk20dx256vlh7.bootloader.ld:57 cannot move location counter backwards (from 00000000000004ba to 0000000000000400)
collect2: error: ld returned 1 exit status
make[2]: *** [kiibohd_bootloader.elf] Error 1
make[1]: *** [CMakeFiles/kiibohd_bootloader.elf.dir/all] Error 2
make: *** [all] Error 2

zassen, 2016-08-02:
I build success in windows and gcc version is 5.4.
if use gcc 4.8, this issue will occur in windows. thanks hint.
The wiki may need add  gcc verion for reference.

haata, 2016-08-03:
Yeah, I believe the minimum requirement is gcc 4.9

I think it was documented at one point, but I haven't had anyone trying to
compile the bootloader until recently (so it probably got lost).

zassen, 2016-08-04:
by the way， which gcc version that you use for firmware building ?’

haata, 2016-08-04:
I use whatever is on Arch Linux (which stays pretty up to date). Probably 6.1 was my latest try.

https://www.archlinux.org/packages/core/x86_64/gcc/

haata, 2017-10-03:
The bootloader should be much easier to build (at least for Linux).

haata, 2017-11-20:
Please use the build scripts to build bootloaders now.

For example, `Bootloader/Builds/mk20dx256vlh7.bash` is a generic mk20dx256vlh7 bootloader.

closed 2017-11-20 by haata