kiibohd/controller #140: Followed wiki, unexpected matrixDebug results w/ Teensy 3.1

alaricljs, 2016-08-03:
I followed [these directions](https://github.com/kiibohd/controller/wiki/Custom-Keyboard-using-a-Teensy-3.0-3.1-3.2).  I set a 4x4 matrix:
```
GPIO_Pin Matrix_cols[] = { gpio(D,7), gpio(D,4), gpio(D,2), gpio(D,3) };
GPIO_Pin Matrix_rows[] = { gpio(C,3), gpio(C,4), gpio(C,6), gpio(C,7) };
```
Here's where things are off:
```
: matrixInfo
INFO - Columns:  0x9
INFO - Rows:     0x7
INFO - Max Keys: 0x3F
```

Testing the pins with a probe tells me that somehow the Infinity 60% pinout is still being used.

Heeeelp?

haata, 2016-08-03:
Would you mind showing me the full compiler log output?
As well as the modifications to the kiilulz.bash script?

alaricljs, 2016-08-03:
```
BuildPath="ljs-tkl"
BaseMap="scancode_map"
DefaultMap="md1Overlay stdFuncMap"
PartialMaps[1]="hhkbpro2"
PartialMaps[2]="colemak"
ScanModule="ljs-tkl"
MacroModule="PartialMap"
OutputModule="pjrcUSB"
DebugModule="full"
Chip="mk20dx256"
Compiler="gcc"
if [ ! -f cmake.bash ]; then
	echo "ERROR: Cannot find 'cmake.bash'"
	exit 1
fi
source cmake.bash
```

```
-- Compiler Family:
arm
-- Compiler Selected:
CMake Deprecation Warning at /usr/share/cmake-3.6/Modules/CMakeForceCompiler.cmake:79 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  Lib/CMake/arm.cmake:18 (cmake_force_c_compiler)
  Lib/CMake/initialize.cmake:53 (include)
  CMakeLists.txt:52 (include)

CMake Deprecation Warning at /usr/share/cmake-3.6/Modules/CMakeForceCompiler.cmake:93 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  Lib/CMake/arm.cmake:19 (cmake_force_cxx_compiler)
  Lib/CMake/initialize.cmake:53 (include)
  CMakeLists.txt:52 (include)

gcc
-- Chip Selected:
mk20dx256
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
Teensy
-- Detected Scan Module Source Files:
Scan/MatrixARM/matrix_scan.c;Scan/ljs-tkl/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Detected Layout Files:
/home/ljs/Code/kiibohd/controller/Scan/MatrixARM/capabilities.kll
/home/ljs/Code/kiibohd/controller/Macro/PartialMap/capabilities.kll
/home/ljs/Code/kiibohd/controller/Output/pjrcUSB/capabilities.kll
/home/ljs/Code/kiibohd/controller/Scan/ljs-tkl/scancode_map.kll
/home/ljs/Code/kiibohd/controller/kll/layouts/md1Overlay.kll
/home/ljs/Code/kiibohd/controller/kll/layouts/stdFuncMap.kll
/home/ljs/Code/kiibohd/controller/kll/layouts/hhkbpro2.kll
/home/ljs/Code/kiibohd/controller/kll/layouts/colemak.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ljs/Code/kiibohd/controller/Keyboards/ljs-tkl.gcc
Scanning dependencies of target kiibohd.elf
[  4%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/output_com.c.o
[  8%] Building C object CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.o
[ 13%] Linking C executable kiibohd.elf
Creating iHex file to load:    kiibohd.teensy.hex
Creating Extended Listing:     kiibohd.lss
Creating Symbol Table:         kiibohd.sym
[ 95%] Built target kiibohd.elf
[100%] Chip usage for mk20dx256
	 SRAM: 	11% 	7396/65536	bytes
	Flash: 	11% 	29640/262144	bytes
[100%] Built target SizeAfter
Firmware has been compiled into: 'ljs-tkl.gcc'
/home/ljs/Code/kiibohd/controller/Keyboards
```

haata, 2016-08-03:
Hmm. It looks like you are doing it correctly.

I think...this is just a build system bug. Sometimes when you change .h files it doesn't get picked up properly.

Try this:

```bash
cd /home/ljs/Code/kiibohd/controller/Keyboards/ljs-tkl.gcc
make clean
make
```

This should force the build to regenerate everything and source the .h files.

Alternatively you can just remove the ljs-tkl.gcc directory.

alaricljs, 2016-08-03:
Success, thanks for that
```
matrixInfo
INFO - Columns:  0x4
INFO - Rows:     0x4
INFO - Max Keys: 0x10
```

closed 2016-08-03 by alaricljs