kiibohd/controller #142: Creating another endpoint (for computer interaction)

danieloosthuizen, 2016-08-05:
[On the ’dox subreddit][1], I asked whether it was possible for the computer to change the ’dox’s current layer – the only way currently is to have a key mapped to a particular layer (e.g., `[+L2]`).

The answer I got was twofold. For this to happen, there must be:

 1. **another endpoint** (this time, OUT), and
 2. **a program which interacts with the keyboard**.

For (2) to happen, (1) needs to be there first; which is why I’ve brought this up.

For someone unfamiliar with this project (especially myself!), this seems a daunting task. 

So: **are there any plans to bring about this kind of functionality (1), given that it would have a very large scope?** (Beyond just changing layers! – it ~~would~~ _should_ allow you to change anything on the keyboard, without having to go through the laborious process of re-flashing.) 

[1]: https://www.reddit.com/r/ergodox/comments/4vtaoi/computercontrolled_layer_changing/

haata, 2016-08-05:
This is the Infinity Ergodox firmware (the original ergodox uses different firmware). But, yes (1) is planned. And already implemented to an extent (the endpoint code is there, just it doesn't do anything right now).

danieloosthuizen, 2016-08-05:
In that case:

 -  **how ‘high on the agenda’ is it?** (I’m guessing there’s lots more to do aside from implementing this particular feature.), and
 - **what can people do to help?** (A kind of – relatively – detailed to-do list.)

haata, 2016-08-05:
It's somewhat high. There are a few big ticket items that need to be finished before I'll get to this.
Specifically
- K-Type LED Support
- KLL 0.5 support
- Client-side configurator

Likely what will happen is that the client-side configurator will hook into this endpoint and handle all the control. However, there's nothing preventing, say a Python implementation, to handle the OS side control.

I started on a spec for the interface already (https://deskthority.net/workshop-f7/hid-io-sideband-input-device-protocol-spec-t12927.html). The idea is that more commands can be added easily, rather than just trying to think of everything at once.

Things I can use help with
- Python interface to HID-IO (client side configurator will likely be done in Javascript/Electron, really my requirement is cross platform)
- Help implementing some of the firmware side functions (this will be in C, if there is interest, I can do some quick examples and have a command or two working rather quickly)
- Review/suggestion of new commands (with a focus on simple a concise instructions with as little duplication as possible, flash can be limited on microcontrollers)
- Testing of interface (once there is some code available)
- Keyboard interface simulator (to unit test the interface), then setup with Travis to validate new commands

Some side projects related to this include
- Unicode system driver (able to type Unicode from your keyboard directly)
- Automated driver installation procedures (I have something sorta working with Zadig on Windows. Nothing is needed right now for macOS and Linux but it might be for Unicode)

danieloosthuizen, 2016-08-05:
Thank you for putting this together. :relaxed: The impression I got from reddit is that there _is_ a fair amount of interest in this; but I'll get on with reading [that spec][1]!

[1]: https://deskthority.net/workshop-f7/hid-io-sideband-input-device-protocol-spec-t12927.html

haata, 2017-10-03:
BTW, I'm actually working on this now: https://github.com/hid-io/hid-io