kiibohd/controller #145: Compile Errors in usb_desc.c

p-callaghan, 2016-08-19:
I just downloaded the new firmware and I am getting the following error on build:

Everything worked fine for the previous versions:

Scanning dependencies of target kiibohd.elf
[  4%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/output_com.c.o
[  8%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o
../../Output/pjrcUSB/arm/usb_desc.c:44:19: error: expected expression before ')' token
 #define LSB(n) ((n) & 255)
                   ^
../../Output/pjrcUSB/arm/usb_desc.c:63:2: note: in expansion of macro 'LSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
  ^
../../Output/pjrcUSB/arm/usb_desc.c:45:20: error: expected expression before ')' token
 #define MSB(n) (((n) >> 8) & 255)
                    ^
../../Output/pjrcUSB/arm/usb_desc.c:63:20: note: in expansion of macro 'MSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
                    ^
make[2]: *** [CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2

haata, 2016-08-19:
Can you remove the build directory and show me the output of the cmake
portion of the build? ( This might fix the problem too)

p-callaghan, 2016-08-19:
I wipe out the whole controller directory and pull down the controller and kll repo from scratch.

I tried it again with the same results.

Here is the entire build output.

-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Detected Scan Module Source Files:
Scan/ISSILed/led_scan.c;Scan/MatrixARM/matrix_scan.c;Scan/STLcd/lcd_scan.c;Scan/UARTConnect/connect_scan.c;Scan/Infinity_Ergodox/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "1.7.1")
-- Found Ctags: /usr/bin/ctags (found version "5.8")
-- Detected Layout Files:
/ergodox/controller/Scan/ISSILed/capabilities.kll
/ergodox/controller/Scan/MatrixARM/capabilities.kll
/ergodox/controller/Scan/STLcd/capabilities.kll
/ergodox/controller/Scan/UARTConnect/capabilities.kll
/ergodox/controller/Macro/PartialMap/capabilities.kll
/ergodox/controller/Output/pjrcUSB/capabilities.kll
/ergodox/controller/Scan/Infinity_Ergodox/scancode_map.kll
/ergodox/controller/Scan/Infinity_Ergodox/leftHand.kll
/ergodox/controller/Scan/Infinity_Ergodox/slave1.kll
/ergodox/controller/Scan/Infinity_Ergodox/rightHand.kll
/ergodox/controller/kll/layouts/callaghp.kll
/ergodox/controller/kll/layouts/lcdFuncMap.kll
/ergodox/controller/kll/layouts/callaghp1.kll
/ergodox/controller/kll/layouts/callaghp2.kll
-- Found DFU_SUFFIX: /usr/bin/dfu-suffix (found version "0.7(C) 2011-2012 Stefan SchmidtThis program is Free Software and has ABSOLUTELY NO WARRANTYPlease report bugs to dfu-util@lists.gnumonks.org")
-- Configuring done
-- Generating done
-- Build files have been written to: /ergodox/controller/Keyboards/ICED-L.gcc
[  4%] Generating KLL Layout
Scanning dependencies of target kiibohd.elf
[  8%] Building C object CMakeFiles/kiibohd.elf.dir/main.c.o
[ 12%] Building C object CMakeFiles/kiibohd.elf.dir/Lib/mk20dx.c.o
[ 16%] Building C object CMakeFiles/kiibohd.elf.dir/Lib/delay.c.o
[ 20%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/ISSILed/led_scan.c.o
[ 24%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/MatrixARM/matrix_scan.c.o
[ 28%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/STLcd/lcd_scan.c.o
[ 32%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/UARTConnect/connect_scan.c.o
[ 36%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/Infinity_Ergodox/scan_loop.c.o
[ 40%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.o
[ 44%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/result.c.o
[ 48%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/trigger.c.o
[ 52%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/output_com.c.o
[ 56%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o
../../Output/pjrcUSB/arm/usb_desc.c:44:19: error: expected expression before ')' token
#define LSB(n) ((n) & 255)
                   ^
../../Output/pjrcUSB/arm/usb_desc.c:63:2: note: in expansion of macro 'LSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
  ^
../../Output/pjrcUSB/arm/usb_desc.c:45:20: error: expected expression before ')' token
#define MSB(n) (((n) >> 8) & 255)
                    ^
../../Output/pjrcUSB/arm/usb_desc.c:63:20: note: in expansion of macro 'MSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
                    ^
make[2]: *** [CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [all] Error 2
Error in make. Exiting...

IMPORTANT NOTICE: This message is intended for the individual or entity to which it is addressed
and may contain information that is privileged, confidential, and/or exempt from disclosure under 
applicable law. If you are not the intended recipient, you are hereby notified that copying, 
forwarding or other dissemination or distribution of this message is prohibited and that taking 
any action in reliance on the content of this message is to be avoided. Should you receive this 
e-mail in error, please notify the sender immediately via e-mail or call (403) 699-4000 
and delete this message from your system. Thank you. 

************************************************************

haata, 2016-08-19:
Hmm. Looks like some combo of build tools and OS I didn't test recently...
(I rewrote the compiler selection code recently)

Which OS? (uname -a)
cmake --version
arm-none-eabi-gcc --version

p-callaghan, 2016-08-19:
Here is the info:

[root@ergodox /]# cat /etc/redhat-release
Red Hat Enterprise Linux Server release 6.6 (Santiago)

[root@ergodox /]# uname -a
Linux ergodox 2.6.32-504.el6.x86_64 #1 SMP Tue Oct 14 01:47:47 PDT 2014 x86_64 x86_64 x86_64 GNU/Linux

[root@ergodox /]# cmake --version
cmake version 2.8.12.2

[root@ergodox /]# arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848]

************************************************************ 

IMPORTANT NOTICE: This message is intended for the individual or entity to which it is addressed
and may contain information that is privileged, confidential, and/or exempt from disclosure under 
applicable law. If you are not the intended recipient, you are hereby notified that copying, 
forwarding or other dissemination or distribution of this message is prohibited and that taking 
any action in reliance on the content of this message is to be avoided. Should you receive this 
e-mail in error, please notify the sender immediately via e-mail or call (403) 699-4000 
and delete this message from your system. Thank you. 

************************************************************

p-callaghan, 2016-08-19:
I will try upgrading some of the software as well, and let you know if it makes a difference.  

I see there is a new gcc-arm-non-eabi package available version 5.4.

Thanks,
     Peter

************************************************************ 

IMPORTANT NOTICE: This message is intended for the individual or entity to which it is addressed
and may contain information that is privileged, confidential, and/or exempt from disclosure under 
applicable law. If you are not the intended recipient, you are hereby notified that copying, 
forwarding or other dissemination or distribution of this message is prohibited and that taking 
any action in reliance on the content of this message is to be avoided. Should you receive this 
e-mail in error, please notify the sender immediately via e-mail or call (403) 699-4000 
and delete this message from your system. Thank you. 

************************************************************

alexisorg1, 2016-08-24:
I am getting the exact same error on multiple different operating systems/builds. At this point I am convinced that I must be doing something wrong. 

```
arm
— Compiler Selected:
gcc
— Chip Selected:
mk20dx128vlf5
— Chip Family:
mk20dx
— CPU Selected:
cortex-m4
— Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
— Bootloader Type:
dfu
— The C compiler identification is GNU 6.1.1
— Check for working C compiler: /usr/bin/arm-none-eabi-gcc
— Check for working C compiler: /usr/bin/arm-none-eabi-gcc — works
— Detecting C compiler ABI info
— Detecting C compiler ABI info – done
— Detecting C compile features
— Detecting C compile features – done
— Detected Scan Module Source Files:
Scan/MatrixARM/matrix_scan.c;Scan/Infinity_60%/scan_loop.c
— Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
— Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
— Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
— Found Git: /usr/bin/git (found version “2.9.3”)
— Found Ctags: /usr/bin/ctags (found version “5.8”)
— Detected Layout Files:
/home/alexis/Desktop/Test/Scan/MatrixARM/capabilities.kll
/home/alexis/Desktop/Test/Macro/PartialMap/capabilities.kll
/home/alexis/Desktop/Test/Output/pjrcUSB/capabilities.kll
/home/alexis/Desktop/Test/Scan/Infinity_60%/scancode_map.kll
/home/alexis/Desktop/Test/kll/layouts/md1Overlay.kll
/home/alexis/Desktop/Test/kll/layouts/stdFuncMap.kll
/home/alexis/Desktop/Test/kll/layouts/hhkbpro2.kll
— Found DFU_SUFFIX: /usr/bin/dfu-suffix (found version “0.9”)
— Configuring done
— Generating done
— Build files have been written to: /home/alexis/Desktop/Test/Keyboards/IC60.gcc
[ 4%] Generating KLL Layout
fatal: ambiguous argument ‘HEAD’: unknown revision or path not in the working tree.
Use ‘–‘ to separate paths from revisions, like this:
‘git <command> [<revision>…] — [<file>…]’
Scanning dependencies of target kiibohd.elf
[ 8%] Building C object CMakeFiles/kiibohd.elf.dir/main.c.obj
[ 13%] Building C object CMakeFiles/kiibohd.elf.dir/Lib/mk20dx.c.obj
[ 17%] Building C object CMakeFiles/kiibohd.elf.dir/Lib/delay.c.obj
[ 21%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/MatrixARM/matrix_scan.c.obj
[ 26%] Building C object CMakeFiles/kiibohd.elf.dir/Scan/Infinity_60%/scan_loop.c.obj
[ 30%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.obj
[ 34%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/result.c.obj
[ 39%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/trigger.c.obj
[ 43%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/output_com.c.obj
[ 47%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.obj
/home/alexis/Desktop/Test/Output/pjrcUSB/arm/usb_desc.c:44:19: error: expected expression before ‘)’ token
#define LSB(n) ((n) & 255)
^
/home/alexis/Desktop/Test/Output/pjrcUSB/arm/usb_desc.c:63:2: note: in expansion of macro ‘LSB’
LSB(BCD_VERSION), MSB(BCD_VERSION), // bcdDevice
^~~
/home/alexis/Desktop/Test/Output/pjrcUSB/arm/usb_desc.c:45:20: error: expected expression before ‘)’ token
#define MSB(n) (((n) >> 8) & 255)
^
/home/alexis/Desktop/Test/Output/pjrcUSB/arm/usb_desc.c:63:20: note: in expansion of macro ‘MSB’
LSB(BCD_VERSION), MSB(BCD_VERSION), // bcdDevice
^~~
make[2]: *** [CMakeFiles/kiibohd.elf.dir/build.make:292: CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Error in make. Exiting…
```

Can send cygwin/ ubuntu log if necessary.

haata, 2016-08-24:
Have you tried removing the IC60.gcc build directory and trying again?

alexisorg1, 2016-08-25:
Yes, just now once again and to no avail. Perhaps I should try with older versions of the software or maybe change the build directory. All this aimless trying and testing has completely taken up my time in the last few days though. I will probably be waiting for a solution or explanation as I am hopelessly inexperienced.

p-callaghan, 2016-08-30:
I have upgraded the OS as well as the arm-compiler, with the same result.

[root@ergodox /ergodox]# cat /etc/redhat-release
Red Hat Enterprise Linux Server release 6.8 (Santiago)

[root@ergodox /ergodox]# uname -a
Linux ergodox 2.6.32-642.4.2.el6.x86_64 #1 SMP Tue Aug 23 10:42:14 PDT 2016 x86_64 x86_64 x86_64 GNU/Linux

[root@ergodox /usr/local]# arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

I also tried to use an older compiler: gcc-arm-none-eabi-4_9-2015q3 with the same issue.
[root@ergodox /ergodox]# arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

======== COMPILE OUTPUT =================
[ 56%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o
../../Output/pjrcUSB/arm/usb_desc.c:44:19: error: expected expression before ')' token
 #define LSB(n) ((n) & 255)
                   ^
../../Output/pjrcUSB/arm/usb_desc.c:63:2: note: in expansion of macro 'LSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
  ^
../../Output/pjrcUSB/arm/usb_desc.c:45:20: error: expected expression before ')' token
 #define MSB(n) (((n) >> 8) & 255)
                    ^
../../Output/pjrcUSB/arm/usb_desc.c:63:20: note: in expansion of macro 'MSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
                    ^
make[2]: *** [CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [all] Error 2
Error in make. Exiting...

haata, 2016-08-30:
Hmm.

Which version of Python3 do you have?

Also, can you post your generatedKeymap.h and kll_defs.h files? They should be in the build directory.
My guess is that something is going on with the kll compiler that's affecting the C macros.

p-callaghan, 2016-08-30:
[root@ergodox /ergodox]# python3 --version
Python 3.5.1

generatedKeymap.h and kll_defs.h files attached.

/* Copyright (C) 2014-2016 by Jacob Alexander
 *
 * This file is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This file is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this file.  If not, see <http://www.gnu.org/licenses/>.
 */

// This file was generated by the kll compiler, DO NOT EDIT.
// Generation Date:    2016-08-30
// KLL Backend:        kiibohd
// KLL Git Rev:        1a078b2b940709bc3c429c952d2f0d842927394f
// KLL Git Changes:    None
// Compiler arguments:
//    ../../kll/kll.py
//      -d
//      -p
//      -p
//    --backend
//      kiibohd
//    --templates
//      /ergodox/controller/kll/templates/kiibohdKeymap.h
//      /ergodox/controller/kll/templates/kiibohdDefs.h
//    --outputs
//      generatedKeymap.h
//      kll_defs.h
//
// - Base Layer -
//    ISSILedCapabilities
//    STLcdCapabilities
//    pjrcUSBCapabilities
//    MDErgo1
//    MDErgo1LeftHand
//    MDErgo1Slave1
//    MDErgo1RightHand
// - Default Layer -
//    MDErgo1
//    lcdFuncMap
// - Partial Layers -
//    Layer 1
//     MDErgo1
//    Layer 2
//     MDErgo1

#pragma once

// ----- Includes -----

// KLL Include
#include <kll.h>

// ----- Capabilities -----

// Capability function declarations
void LCD_layerStackExact_capability( uint8_t state, uint8_t stateType, uint8_t *args );
void LCD_layerStack_capability( uint8_t state, uint8_t stateType, uint8_t *args );
void LED_control_capability( uint8_t state, uint8_t stateType, uint8_t *args );
[... 1057 lines cut ...]

#define enableUSBResume_define 1
#define STLcdNumber2_define  \
0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x3f, 0x00, 0x00, 0x00, \
0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0xe0, 0x00, 0x00, 0x00, \
0x00, 0x00, 0x00, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0x07, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, \
0x00, 0x00, 0x00, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xfc, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, \

#define MinDebounceTime_define 5
#define ISSILedMask1_define  \
	0xFF, 0x00, /* C1-1 -> C1-16 */ \
	0xFF, 0x00, /* C2-1 -> C2-16 */ \
	0xFF, 0x00, /* C3-1 -> C3-16 */ \
	0xFF, 0x00, /* C4-1 -> C4-16 */ \
	0x3F, 0x00, /* C5-1 -> C5-16 */ \
	0x00, 0x00, /* C6-1 -> C6-16 */ \
	0x00, 0x00, /* C7-1 -> C7-16 */ \
	0x00, 0x00, /* C8-1 -> C8-16 */ \
	0x00, 0x00, /* C9-1 -> C9-16 */ \

#define enableDeviceRestartOnUSBTimeout_define 0
#define CapabilitiesNum_KLL 17
#define ResultMacroNum_KLL 116
#define TriggerMacroNum_KLL 132
#define LayerNum_KLL 3

p-callaghan, 2016-08-30:
No difference with python3.5.2.

Interestingly enough, I am not able to compile C code any more.  It is coming up that only the ARM type is supported.

/usr/local/bin/ld: unrecognised emulation mode: elf_x86_64
Supported emulations: armelf

It didn’t occur to me until you mentioned the C macros.   I am not sure how it got into this state, but I will do some investigation.

Peter

From: Callaghan, Peter (Calgary)
Sent: Tuesday, August 30, 2016 4:41 PM
To: 'kiibohd/controller'; 'kiibohd/controller'
Cc: 'p-callaghan'; 'Author'
Subject: RE: [kiibohd/controller] Compile Errors in usb_desc.c (#145)

[root@ergodox /ergodox]# python3 --version
Python 3.5.1

generatedKeymap.h and kll_defs.h files attached.

p-callaghan, 2016-08-30:
The Supported Emulation is unrelated.

I setup a new Linux install, added the tool and the compiler is giving me the exact same error.

I confirmed I can still compile normal C code on the new system as well.

Just in case there is something off with the first system here are those files and the error again from the new system.

[ 56%] Building C object CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o
../../Output/pjrcUSB/arm/usb_desc.c:44:19: error: expected expression before ')' token
#define LSB(n) ((n) & 255)
                   ^
../../Output/pjrcUSB/arm/usb_desc.c:63:2: note: in expansion of macro 'LSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
  ^
../../Output/pjrcUSB/arm/usb_desc.c:45:20: error: expected expression before ')' token
#define MSB(n) (((n) >> 8) & 255)
                    ^
../../Output/pjrcUSB/arm/usb_desc.c:63:20: note: in expansion of macro 'MSB'
  LSB(BCD_VERSION), MSB(BCD_VERSION),     // bcdDevice
                    ^
make[2]: *** [CMakeFiles/kiibohd.elf.dir/Output/pjrcUSB/arm/usb_desc.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [all] Error 2
Error in make. Exiting...

From: Callaghan, Peter (Calgary) [mailto:Peter.Callaghan@nexencnoocltd.com]
Sent: Tuesday, August 30, 2016 5:23 PM
To: 'kiibohd/controller'; 'kiibohd/controller'
Cc: 'p-callaghan'; 'Author'
Subject: RE: [kiibohd/controller] Compile Errors in usb_desc.c (#145)

No difference with python3.5.2.

Interestingly enough, I am not able to compile C code any more.  It is coming up that only the ARM type is supported.

/usr/local/bin/ld: unrecognised emulation mode: elf_x86_64
Supported emulations: armelf

It didn’t occur to me until you mentioned the C macros.   I am not sure how it got into this state, but I will do some investigation.

Peter

From: Callaghan, Peter (Calgary)
Sent: Tuesday, August 30, 2016 4:41 PM
To: 'kiibohd/controller'; 'kiibohd/controller'
Cc: 'p-callaghan'; 'Author'
Subject: RE: [kiibohd/controller] Compile Errors in usb_desc.c (#145)

[root@ergodox /ergodox]# python3 --version
Python 3.5.1

generatedKeymap.h and kll_defs.h files attached.

haata, 2016-08-31:
Where are you getting arm-none-eabi-gcc from? I haven't seen anything available for CentOS (6.8 or 7.2).

I'm attempting to setup a test environment to see what's up.

haata, 2016-08-31:
Aha! I've managed to reproduce on CentOS 6.8.

No issue on CentOS 7.2 using the exact same arm-none-eabi compiler (https://launchpad.net/gcc-arm-embedded/+download).

Now let's see if I can find the issue.

haata, 2016-08-31:
Yep, found it. Was due to...an ancient version of git. Hah.

#define BCD_VERSION wasn't getting the value defined in buildvars.h

I'm working on a fix now to support git 1.7.1 which doesn't have git rev-list --count HEAD

haata, 2016-08-31:
Fixed! b7b7fd1ee724b8831bb9a870efa76a26a8d080a8

Let me know if this issue still shows up.

closed 2016-08-31 by haata