kiibohd/controller #147: Macro result is interrupted by another macro.

wpc009, 2016-08-24:
For the following marco definition
```
U"A" : U"9" + U"LShift";
U"S" : U"LBrace";
```

when I hit 'A' and 'S' separately, it works fine.
The KeyBoard Event Viewer Shows
```
eventType:FlagsChanged    code:0x38       name:Shift_L         flags:Shift                              misc:KeyCode::SHIFT_L	 characters:    
eventType:KeyDown         code:0x19       name:(               flags:Shift                              misc:KeyCode::9	 characters:(   
eventType:FlagsChanged    code:0x38       name:Shift_L         flags:                                   misc:KeyCode::SHIFT_L	 characters:    
eventType:KeyUp           code:0x19       name:9               flags:                                   misc:KeyCode::9	 characters:9   
eventType:KeyDown         code:0x21       name:[               flags:                                   misc:KeyCode::BRACKET_LEFT	 characters:[   
eventType:KeyUp           code:0x21       name:[               flags:                                   misc:KeyCode::BRACKET_LEFT	 characters:[   

```

But when I hit 'A' then 'S' very quickly, the 'S' macro will interrupt the 'A' macro, and output an error result "{".
Here is the event viewer result.
```
eventType:FlagsChanged    code:0x38       name:Shift_L         flags:Shift                              misc:KeyCode::SHIFT_L	 characters:    
eventType:KeyDown         code:0x19       name:(               flags:Shift                              misc:KeyCode::9	 characters:(   
eventType:KeyDown         code:0x21       name:{               flags:Shift                              misc:KeyCode::BRACKET_LEFT	 characters:{   
eventType:FlagsChanged    code:0x38       name:Shift_L         flags:                                   misc:KeyCode::SHIFT_L	 characters:    
eventType:KeyUp           code:0x19       name:9               flags:                                   misc:KeyCode::9	 characters:9   
eventType:KeyUp           code:0x21       name:[               flags:                                   misc:KeyCode::BRACKET_LEFT	 characters:[   
```

wpc009, 2016-08-24:
Shouldn't all the input event be queued up and write to USB one by one?
My understanding is the 'LBrace' input should be queue up, and send after the 'A' macro finished.

haata, 2017-10-03:
Hmm, this is interesting. I think I'll have to look into this further.