kiibohd/controller #157: Combination Keypress Order

Joshuaweiss, 2016-10-10:
When making a simple shift macro in any of these ways:
```U"1" : U"LSHIFT" + U"1";```
```U"1" : U"1" + U"LSHIFT"```
```U"1" :+ U"LSHIFT"```

It produces the keypresses:
```U"1"(P), U"LSHIFT"(P), U"1"(R), U"LSHIFT"(R)```

I need it to instead produce either:
```U"LSHIFT"(P), U"1"(P), U"1"(R), U"LSHIFT"(R)```
OR
```U"LSHIFT"(P), U"1"(P), U"LSHIFT"(R), U"1"(R)```

I am compiling it for the latest Infinity Ergodox.
Is there an easy way to ensure one of these orders without waiting for kll 0.4 or 0.5?

haata, 2016-10-10:
Hmm. Yeah this will be much easier with KLL 0.4/0.5

So, in theory `U"1" : U"LSHIFT" + U"1";` should work fine because LSHIFT and 1 are sent in the same USB packet. It's up to the OS to determine the order on the other side.

`U"1" : U"LShift", U"LShift" + U"1", U"LShift";` is sorta what you want. You'll have to look at the generatedKeymap.h to see if the datastructure looks correct. 

The kll compiler is inserting break packets to  make sequences behave normally.
https://github.com/kiibohd/kll/blob/master/backends/kiibohd.py#L172
https://github.com/kiibohd/kll/blob/master/backends/kiibohd.py#L211

If you want to get really fancy, you might be able to hand-edit the generatedKeymap.h file. In theory it **can** do what you want, just the build infrastructure will try to *help* you and overwrite it all the time...

Joshuaweiss, 2016-10-10:
I just tried using ```U"1" : U"LSHIFT" + U"1";``` on a computer running El Capitan and it worked fine. It seems the new keyboard drivers in Sierra must have changed the way the os handles these USB packets. I was only aware that apple changed their drivers because it broke Karabiner.

Joshuaweiss, 2016-10-11:
This is actually just an is issue with Karabiner-Elements.

closed 2017-08-22 by Joshuaweiss