kiibohd/controller #158: "Simple macros" don't work on Mac OS Sierra

rjhilgefort, 2016-10-11:
I just upgraded to OSX Sierra to find that the "simple macros" I had configured through the input club layout layout builder no longer fire consistently (fire once every 20 key presses or something, and only when alternating keys for some reason).

Anyone else having this issue? Is it a known issue- I couldn't find anything in the issues on this repo? Is this an issue I should be posting to the input club configurator instead?

TheEmpty, 2016-10-27:
If you have your KLL files, the configurator could easily be ruled out. I assume that's not the issue though. This was also working before Sierra and you haven't re-flashed since then, correct? (and you shouldn't need to unless you were changing bindings)

rjhilgefort, 2016-10-27:
Here's my KLL files: https://github.com/rjhilgefort/.ergodox

This was working before Sierra, and I haven't flashed since the OS change. I've engaged @haata  over on kiibohd/KiiConf in [this issue](https://github.com/kiibohd/KiiConf/issues/63). Based on his comments, it's likely a firmware issue, rather than the configurator.

TheEmpty, 2016-10-27:
The KLL files look good, have the proper F keys. That's all I was trying to rule out. So that would make it an issue here and not the configurator.  I'm not really part of this project just trying to help out, so I wouldn't say close the issue :P

rjhilgefort, 2016-10-31:
I believe this is an issue with https://github.com/tekezo/Karabiner-Elements so I'm closing this.

(Here's the issue on Ergodox Infinity that it was deduced to be KE.)[https://github.com/kiibohd/KiiConf/issues/63]

(Here's the issue I created over at KE)[https://github.com/tekezo/Karabiner-Elements/issues/380]

closed 2016-10-31 by rjhilgefort