kiibohd/controller #163: How does this compare to TMK firmware?

patrickdepinguin, 2016-10-22:
This is not really an 'issue', but I think the answer could be added to a README or FAQ.

Both the TMK firmware as this kiibohd firmware are supposed to support the Infinity keyboard. How do they differ? Why would I choose one over the other?

jceaser, 2017-03-06:
do you want to have this conversation over on input.club?  In short (from what I can tell) this project is trying to create a standard keyboard configuration file to define what the keyboard should do.

patrickdepinguin, 2017-03-12:
@jceaser Not sure what you mean with having the conversation on input.club, do you mean in the forums?

I had a chat with @haata some time ago where he already clarified a bit. I think it would be great to have such description somewhere for others, but if you don't agree this issue can be closed as far as I'm concerned.

jceaser, 2017-03-12:
it's just that this does not seam like an "issue" in terms of a ticket to a bug tracking system.