kiibohd/controller #164: Freezing keystrokes

jhawthorne2, 2016-10-22:
Hey all,

I have a MD infinity I think 1.1 with LEDs, and if I type for any extended period of time eventually one of the key presses will "freeze" and that key will register as pressed down until I unplug the keyboard and plug it back in(some times it's a letter key, sometimes space, sometimes delete).  I brought this up in the MD comments and someone there said it was probably a firmware issue.  I've looked into another issue someone posted about key's getting stuck at startup, but I'm not sure if this is related as my issue appears while typing.

Any help is much appreciated as I really want to be able to use this keyboard.

jceaser, 2017-05-01:
this sounds like an issue I have which I submitted as issue #196. I think it is software and not hardware. I can't even assign the key None to turn it off.

haata, 2017-10-03:
Have you tried a recent firmware?

schliem, 2019-05-07:
I have the same Issue on a Whitefox from the second massdrop run (v1.1). Since I flashed it using the new configurator random keys register permanently until any other key is pressed. This occurs randomly during daily operation, regardless of application.

devopsdeluxe, 2020-07-25:
I can reproduce this issue consistently on reboot from systemd-boot's menu, and cryptsetup's password entry prompt. In both cases, the first key to be pressed is the key that is repeated until it is pressed again, or another key is pressed.

If I wait for 3-5 seconds at either of those menus, the key pressed does not repeat, however I experience the same issue as schliem, where keys that was pressed have chance to get "stuck".

Let me know if there is more information I can provide regarding this issue, as I would be happy to help out now that I can reporduce the issue fairly reliably.

haata, 2020-07-25:
Some things to try.

1) Update to the latest firmware (v0.5.7)
2) Try forcing NKRO mode (and 6KRO) to see if that makes any difference

This may have something to do with sleep modes, but I'm not entirely sure. Do you have LEDs on your WhiteFox?
If you do, are the LEDs on when you press the first key when the problem happens?

Most likely what's happening is that systemd-boot or cryptsetup is throwing away/missing one of the packets (unless the keyboard is being reset for another reason right around that time). The keyboard has no understanding of the OS or what state the computer is in so the behaviour shouldn't be any different.

devopsdeluxe, 2020-07-26:
I'll give your two suggestions a try!

> Do you have LEDs on your WhiteFox?

No, unfortunately.

> The keyboard has no understanding of the OS or what state the computer is in so the behaviour shouldn't be any different.

Right, my guess is the motherboard USB ports are reinitialized during the EFI -> OS handoff. On a related note, it seems that I have about a 3 second window (in systemd-boot, or after the handoff to the initramfs) to reproduce the issue before all keypresses are registered as single keypresses (until the issue randomly occurs during normal desktop activities).