kiibohd/controller #165: Bootloader compile failure

axiiiofevil, 2016-10-23:
Hi,
I've been trying to compile the bootloader for an MK20DX256VLH7 chip, to no avail. Below is the output from the cmake and make commands:

root@PRODUCTION-PC:/mnt/c/Users/Production/Desktop/controller-master/Bootloader/build# cmake ..
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Bootloader Source Files:
main.c;dfu.c;dfu.desc.c;flash.c;kinetis.c;usb.c;debug.c
-- Found Git: /usr/bin/git (found version "1.9.1")
fatal: Not a git repository (or any parent up to mount point /mnt/c)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- Found Ctags: /usr/bin/ctags (found version "5.9~svn20110310")
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/Users/Production/Desktop/controller-master/Bootloader/build
root@PRODUCTION-PC:/mnt/c/Users/Production/Desktop/controller-master/Bootloader/build# make
Scanning dependencies of target kiibohd_bootloader.elf
[ 10%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/main.c.o
[ 20%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/dfu.c.o
[ 30%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/dfu.desc.c.o
[ 40%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/flash.c.o
[ 50%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/kinetis.c.o
[ 60%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/usb.c.o
[ 70%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/debug.c.o
[ 80%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/mnt/c/Users/Production/Desktop/controller-master/Lib/mk20dx.c.o
[ 90%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/mnt/c/Users/Production/Desktop/controller-master/Lib/delay.c.o
Linking C executable kiibohd_bootloader.elf
/mnt/c/Users/Production/Desktop/controller-master/Bootloader/../Lib/mk20dx256vlh7.bootloader.ld:57 cannot move location counter backwards (from 00000000000004ba to 0000000000000400)
collect2: error: ld returned 1 exit status
make[2]: *** [kiibohd_bootloader.elf] Error 1
make[1]: *** [CMakeFiles/kiibohd_bootloader.elf.dir/all] Error 2
make: *** [all] Error 2

It's similar to a problem posted earlier, but I've updated to gcc 6.2, so I don't think that that's the problem....

Any help would be greatly appreciated.

axiiiofevil, 2016-10-23:
Just tried gcc 5.4.1 and no luck....
Also I can compile the 128K version no problems..... Strange but true.

haata, 2016-10-23:
Hmm.

Have you tried using the scripts under Bootloader/Builds?
Both appveyor and Travis (Linux and Mac) bootloader builds are working for me. 
https://travis-ci.org/kiibohd/controller/jobs/166643352

Also, please clone the git repo instead of downloading the archive. There's no way to figure out which version of the bootloader you are running otherwise which makes it even harder to debug issues.

You may also try the ktype branch, but it has a few untested features in the bootloader that may cause issues for non-ktype keyboards.

axiiiofevil, 2016-10-24:
Firstly, thanks for getting back so fast!
I've not tried the scripts, but they look like they don't compile the boot loader? Or am I missing something.
I'm actually trying to get the boot loader to work on it's own to run some experiments for a completely different project using the boot loader.

What I've tried so far...
1. I've cloned the repo now and have access to the branches.. 
2. Also I've now re-installed the linux subsystem under windows 10 (completely fresh), installed all of the packages you recommend in the wiki and re-compiled and... same result which is weird right?
3. I've tried gcc 5.4, 4.9 and 6.1 now  as well. What's strange is it compiles the 128K version of the processor (MK20DX128.....) fine, but bugs out on the 256K one.
4. Tried the K type branch, same thing.
5. Are there binaries for the boot loader? I can't find them if there are, maybe I'm looking in the wrong place. I just want to run a couple of experiments to see if using this boot loader for another project is feasible. It should be right?

axiiiofevil, 2016-10-24:
BTW I checked out your Travis build, and mine is the same, up to the linking. The output below is the K branch, which has the full compiler output.

I've been looking in the linker for a problem with the memory addressing, but the Freescale literature is DIABOLICAL, so after I'd nodded off a couple of times I gave up. I'll probably go back into it unless you think this is something else. I've also got a macbook pro I can (probably) set up to compile this if you think it's useful.

Here's the output for the last build I did (K branch with the MK20DX256 selected).

root@PRODUCTION-PC:/mnt/f/10_Linux_Builds/00_Bootloader/Bootloader/build# cmake ..
-- Build CPU Detected:
x86_64
-- Build OS Detected:
Linux 3.4.0+
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Bootloader Source Files:
main.c;dfu.c;dfu.desc.c;flash.c;kinetis.c;usb.c;debug.c
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc
-- Found Git: /usr/bin/git (found version "1.9.1")
-- Found Ctags: /usr/bin/ctags (found version "5.9~svn20110310")
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/f/10_Linux_Builds/00_Bootloader/Bootloader/build
root@PRODUCTION-PC:/mnt/f/10_Linux_Builds/00_Bootloader/Bootloader/build# make
Scanning dependencies of target kiibohd_bootloader.elf
[ 10%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/main.c.o
[ 20%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/dfu.c.o
[ 30%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/dfu.desc.c.o
[ 40%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/flash.c.o
[ 50%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/kinetis.c.o
[ 60%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/usb.c.o
[ 70%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/debug.c.o
[ 80%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/mnt/f/10_Linux_Builds/00_Bootloader/Lib/mk20dx.c.o
[ 90%] Building C object CMakeFiles/kiibohd_bootloader.elf.dir/mnt/f/10_Linux_Builds/00_Bootloader/Lib/delay.c.o
Linking C executable kiibohd_bootloader.elf
/mnt/f/10_Linux_Builds/00_Bootloader/Bootloader/../Lib/mk20dx256vlh7.bootloader.ld:57 cannot move location counter backwards (from 00000000000004d8 to 0000000000000400)
collect2: error: ld returned 1 exit status
make[2]: *** [kiibohd_bootloader.elf] Error 1
make[1]: *** [CMakeFiles/kiibohd_bootloader.elf.dir/all] Error 2
make: *** [all] Error 2
root@PRODUCTION-PC:/mnt/f/10_Linux_Builds/00_Bootloader/Bootloader/build#

haata, 2016-10-24:
This definitely isn't a Freescale thing. I wrote the entire build system and the .ld scripts are custom.

Yes, this bootloader should work for other projects. It's just a standard dfu-util bootloader with a cleaned-up usb descriptor (also has upload/download support). 

What version of arm-none-eabi-ld are you using?
My Arch Linux system has 2.27.

I vaguely recall seeing that error a while ago. IIRC I just ignored it and flashed the kiibohd_bootloader.bin file anyways.

Looking closer at mk20dx256vlh7.bootloader.ld:57 it seems like the linker is having trouble fitting both the vector table and startup function before 0x400. 
In theory, you can just do this (less space efficient, but we still have around 2K in the bootloader protected region, protection is in 8k chunks for the mk20dx256vlh7, 4k for the mk20dx128vlf5).

```c
.text :
{
	. = 0; KEEP(* (.vectors)) /* MUST BE AT 0 */
	. = 0x400; KEEP(* (.flashconfig)) /* MUST BE AT 0x400 */
	*(.startup*)
	*(.rodata*)
	*(.text*)
	. = ALIGN(4);
	KEEP(*(.init))
} > FLASH
```

axiiiofevil, 2016-10-24:
Thank you sir! Are you in the Northern or Southern hemisphere? If your up north it's VERY late for you!
> Looking closer at mk20dx256vlh7.bootloader.ld:57 it seems like the linker is having trouble fitting both the vector table and startup function before 0x400. 

This is why I was looking at the HORRENDOUS Freescale processor guide :) I'm sure you've done some time with them and know what I mean.... I was trying to work out what size the protected regions were, and whether they are different in the 128K and 256K versions.
Funny thing is, the linker file for the 128K version is almost identical to the 256K version and the 128K version compiles no problem.

> What version of arm-none-eabi-ld are you using?
My Arch Linux system has 2.27.

You'r running ver 2.27 of arm-none-eabi-ld ? Isn't the latest version 5.4?
My system tells me it's got 4.8.2 installed.
`root@PRODUCTION-PC:/mnt/f/10_Linux_Builds/00_Bootloader/Bootloader/build# arm-none-eabi-gcc --version
arm-none-eabi-gcc (4.8.2-14ubuntu1+6) 4.8.2`
I can't even find 2.2....... Could this be the problem?

```
.text :
	{
		. = 0; KEEP(* (.vectors)) /* MUST BE AT 0 */
		*(.startup*)
		. = 0x400; KEEP(* (.flashconfig)) /* MUST BE AT 0x400 */
		*(.rodata*)
		*(.text*)
		. = ALIGN(4);
		KEEP(*(.init))
	} > FLASH
```
This is what's currently in the mk20dx256vlh7.bootloader.ld file. It's no different (except in order) from what you have suggested. I tried the your new order in the previous post to no avail.....

haata, 2016-10-24:
I'm in California, not quite 10 PM yet.

binutils is a different version than gcc is. The linker is what's failing here, though your gcc version might be at fault here. The bootloader has always been a bit finicky and needed a very recent version of gcc (4.9.1 was known to work at one point). Here are the arm-none-eabi packages I have installed.

```
local/arm-none-eabi-binutils 2.27-1
    A set of programs to assemble and manipulate binary and object files for the ARM EABI (bare-metal) target
local/arm-none-eabi-gcc 6.2.0-1
    The GNU Compiler Collection - cross compiler for ARM EABI (bare-metal) target
local/arm-none-eabi-gdb 7.11.1-1
    The GNU Debugger for the ARM EABI (bare-metal) target
local/arm-none-eabi-newlib 2.4.0-3
    A C standard library implementation intended for use on embedded systems (ARM bare metal)
```

Hmm, I don't think kiibohd_bootloader.lss is generated yet (it would be helpful...).

haata, 2017-10-03:
I have made a lot of enhancements to the bootloader build this summer.
I'm going to close this, but feel free to re-open it if there are still issues in master.

closed 2017-10-03 by haata