kiibohd/controller #168: Adding a trackpad

micser, 2016-11-10:
I have successfully added mouse control following the steps in https://github.com/kiibohd/controller/issues/4, I was wondering now if it is even possible technically to a trackpad to the keyboard?
I have one salvaged from an old laptop that I would like to reuse...

haata, 2016-11-10:
Technically, yes!

You'll need to figure out what sort of protocol the trackpad communicates over, then select some available pins in order to communicate with it.
Also, more than likely a bunch of USB work would be necessary in order to expose more of the trackpad functionality. Though that's just secondary stuff that needs to be done anyways at some point.

micser, 2016-11-13:
Thanks!
I'll let you know how it goes . I have no idea what I'm doing, so I might be a while :)

CrazyByDefault, 2017-04-09:
Hey @micser have you made any progress?

micser, 2017-04-11:
Hey @CrazyByDefault, no sorry, I gave up on the idea because I couldn't figure out the wiring and didn't really know how to get started... but if you have any ideas, I'm listening! The trackpad I was trying to add to the keyboard is out of an old Compaq Presario laptop and looks very similar to the one on [this page](http://runawaybrainz.blogspot.com/2013/04/hacks-reuse-laptop-trackpad.html). A wireless Logitech trackpad does the job for me at the moment.