kiibohd/controller #170: Sound Teensy 3.2/3.1 Dac

alex-d-fox, 2016-11-28:
Is it possibly to use kiibohd as a composite usb device with sound card and keyboard perhaps even make something like a touchscreen bar similar to the 2016 MacBook Pros

haata, 2016-11-28:
I don't see why not.

The main concern when using MCUs for audio DACs is how many clock cycles are required (any sound processing is rather expensive). 

A touchbar is mostly just finding a touchscreen that would work. Then having a way for the computer to talk to the keyboard to control it. My guess is that Apple didn't make it easy, so I'm not sure how portable their approach is.

alex-d-fox, 2016-11-28:
Someone has already implemented a touchbar emulation for iPad and older Macs with https://github.com/bikkelbroeders/TouchBarDemoApp

alex-d-fox, 2016-11-28:
Also Teensy has audio features already http://www.pjrc.com/teensy/gui/ and a built in DAC for mono audio.

alex-d-fox, 2016-11-28:
I am also building a keyboard based on your firmware and modifying an old ibm model-m to work with your keyboard firmware

haata, 2016-11-28:
Neat!

Feel free to drop into our IRC channel if you're having issues or want to chat. #input.club @ irc.freenode.net

jceaser, 2017-04-18:
the audio on the Planck is helpful to inform you when you have entered into a new layout, the ErgoDox Infinity has a screen for this, but all the others could really benefit from audio.