kiibohd/controller #175: [Bug] Windows 10 restart 'crashes'

rammoo, 2017-01-04:
Hi,
I'm using the WhiteFox and while setting up my environment for compiling layouts I've been experiencing issues with the keyboard 'crashing' after restarting Windows 10. 
In the past week this issue only occurs, when Windows 10 is in a restarted state. The issue fixes itself after shutting down the PC and booting manually. 

Steps to reproduce the issue:
1. Have Windows 10
2. Connect keyboard to PC
3. Use the RESTART functionality in Windows 10 (not the hardware button)
4. Start typing
5. Wait for keyboard to crash (happens in irregular intervals, sometimes multiple times a minute and other times once every ~5 minutes)

[Demonstration](https://www.youtube.com/watch?v=70KImRuMFsE)

This happens on key down and key up events, and there doesn't seem to be any indicator of the issue being related to any keys/switches.

A [comment](https://www.massdrop.com/buy/the-whitefox-keyboard/talk/621442) on massdrop indicates that this might also happen on other operation systems.

haata, 2017-10-03:
Hmm. This sounds like it could possibly be a firmware crash or a corrupted USB packet.

Have you tried the master branch of the firmware lately?

rammoo, 2017-10-26:
I haven't had time to manually compile the firmware myself lately and have been using the configurator (latest flash was 1 day ago). 
The issue is still happening and actually seems to have gotten worse, outside of just the keyboard crashing, now sometimes the entire USB hub appears to get overloaded by the keyboard (keyboard and mouse stop working, constant reconnecting and have to force reboot). 

This has also started happening during regular use (not just in the restarted state). 
I've also noticed that the keyboard defaults to boot mode (6KRO?) and manually has to be changed to NKRO, I'm not sure if that might be related in any way. I'm using a GA-Z77X-D3H motherboard.