kiibohd/controller #176: MSI 970A-G46 UEFI version 2.8: PC Firmware freezes. [Ergodox Infinity 1.2b]

ntzrmtthihu777, 2017-01-11:
So yeah. Odd issue, and is really the only thing keeping me from ergodoxing full time (aside from adjusting to the split and thumb keys). If at any point prior to the linux kernel executing (UEFI->GRUB2) the ergodox infinity is plugged into the machine (from any usb port as far as I can tell, its a real tedious test), the vendor BIOS/UEFI freezes up.

This occurs, for example, if I just let it go the normal boot route, or if I stop it and enter the firmware config menu, or get to the grub menu and plug it in and so on. Entirely unsure how in the world I would debug this.

ntzrmtthihu777, 2017-01-20:
Seems I spoke too soon: while it does freeze up, it will eventually unfreeze. Thanks to @elb on #input.club I have a debug capable firmware, and it outputs the following from the tx2 pin:
`USB Low Power Negotation Disabled, condition detected.`

This arises at [Output/pjrcUSB/arm/usb_dev.c#L244](https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_dev.c#L244).

What steps do I need to take to further debug this? Having the ergodox plugged in greatly increases my boot time as of right now. I realize this is likely the fault of a shoddy BIOS and not your fault, but it will be quite some time before I can replace the vendor BIOS with a coreboot one.

haata, 2017-01-20:
So, if you feel ambitious, you might be able to figure out which USB command/response is causing the delay using either:
https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_dev.c#L56
or
https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_dev.c#L54
uncommented

Possibly there's a stall or incorrect response for your particular BIOS.
In some cases, the correct action is a stall though.

https://github.com/kiibohd/controller/blob/master/Output/pjrcUSB/arm/usb_dev.c#L618
is an example of a command response.

I can't really give you much help remotely unless I have that motherboard (and BIOS version) on hand.
A USB analyzer also helps, but those are quite pricey (more than an ergodox...). However, they aren't a magical device, and the debug output tells you 80% of what you need to know anyways.

ntzrmtthihu777, 2017-01-20:
Will do, once I either get `arm-none-eabi` to build properly (with either `crossdev` or `ct-ng`) on gentoo or manage to coerce someone into building me a new debug firmware ;)

haata, 2019-02-18:
This may be resolved now (I've done a lot of work on the USB side of the firmware), though please re-open if this is still an issue after updating.

closed 2019-02-18 by haata

hanetzer, 2019-02-19:
@haata honestly I no longer have said motherboard so I can't test anyways.