kiibohd/controller #178: Ability to light up individual LEDs.

winneon, 2017-01-17:
Hello,

I have a simple question: is it possible to individually light up LEDs, perhaps within the KLL Spec, instead of every key being lit up per backlight toggle? I tried to find any reference of this in this repository and the KLL Spec, and have found nothing.

haata, 2017-01-17:
I'm actually working on this right now for the K-Type.

Take a look at the KLL 0.5 spec. https://github.com/kiibohd/kll-spec

Part of the reason this has taken me so long was I need to re-write the compiler to implement a lot of the new spec (that part is done!).
The next thing I need to do is unify the K-Type and LED drivers for the other keyboards (IC60, ICED, Whitefox). Adding in support for LEDs has shown a few weaknesses in the i2c driver so I'll be implementing a faster DMA version as well.

winneon, 2017-01-17:
Forgive me, as the KLL Spec is still slightly too technical for my understanding at the moment. Does the future 0.5 version Spec include different LED configurations per layer, say like a normal keymap, but for LEDs instead? If not, that seems to be the most user-friendly way of configuring certain LED indicators, and is very powerful in much the way the layered keymap KLL already provides.

haata, 2017-01-18:
Yeah, the intention is to give the user access to "animation sets". Each animation set can be just a single LED or multiple LEDs changing over a period of time.

When changing to, say, a new layer, it will be possible to trigger that animation, and thus change whichever LEDs you're interested in.

winneon, 2017-01-18:
Thank you for answering my questions! I eagerly await the finished product(s), i.e. the controller and compiler!

closed 2017-01-18 by winneon