kiibohd/controller #179: When attempting to follow instructions to make a keyboard layout I get stuck at the bash script.

trisimix, 2017-01-26:
So I was able to use an unedited bash script but when I change it to include the klls I made I get the following:
`[trisimix@localhost Keyboards]$ bash infinity_led.bash -f -o IC60_LED
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx128vlf5
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Detected Scan Module Source Files:
Scan/ISSILed/led_scan.c;Scan/MatrixARM/matrix_scan.c;Scan/Infinity_60%_LED/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "2.11.0") 
-- Found Ctags: /usr/bin/ctags (found version "5.8") 
-- Found PythonInterp: python3 (found suitable version "3.6", minimum required is "3") 
-- Detected Layout Files:
/home/trisimix/controller/Scan/ISSILed/capabilities.kll
/home/trisimix/controller/Scan/MatrixARM/capabilities.kll
/home/trisimix/controller/Macro/PartialMap/capabilities.kll
/home/trisimix/controller/Output/pjrcUSB/capabilities.kll
/home/trisimix/controller/Scan/Infinity_60%_LED/scancode_map.kll
/home/trisimix/controller/kll/layouts/md10.kll
/home/trisimix/controller/kll/layouts/md11.kll
/home/trisimix/controller/kll/layouts/md12.kll
/home/trisimix/controller/kll/layouts/md13.kll
/home/trisimix/controller/kll/layouts/md14.kll
/home/trisimix/controller/kll/layouts/md15.kll
/home/trisimix/controller/kll/layouts/md16.kll
/home/trisimix/controller/kll/layouts/md17.kll
-- Found DFU_SUFFIX: /usr/bin/dfu-suffix (found version "0.9") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/trisimix/controller/Keyboards/IC60_LED
[  4%] Generating KLL Layout
Traceback (most recent call last):
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 204, in _or
    return self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 182, in _add
    (v1, s2) = self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 182, in _add
    (v1, s2) = self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 224, in _shift
    (v, s2) = self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 224, in _shift
    (v, s2) = self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 348, in _some
    raise NoParseError('got unexpected token', s, t)
funcparserlib.parser.NoParseError: got unexpected token

During handling of the above exception, another exception occurred:

[... 254 lines cut ...]
    (v1, s2) = self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 182, in _add
    (v1, s2) = self.run(tokens, s)
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 206, in _or
    return other.run(tokens, State(s.pos, e.state.max))
  File "/home/trisimix/controller/kll/funcparserlib/parser.py", line 225, in _shift
    return f(v), s2
  File "../../kll/kll.py", line 313, in make_seqString
    usbCode = kll_hid_lookup_dictionary['USBCode'][ processedChar.upper() ]
KeyError: ' '

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "../../kll/kll.py", line 784, in <module>
    processKLLFile( filename )
  File "../../kll/kll.py", line 724, in processKLLFile
    if err.token.value != '<=':
AttributeError: 'KeyError' object has no attribute 'token'
make[2]: *** [CMakeFiles/kiibohd.elf.dir/build.make:74: generatedKeymap.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Error in make. Exiting...
[trisimix@localhost Keyboards]$ 
`

haata, 2017-01-28:
Would you mind posting your kll files here?

haata, 2017-10-03:
Have you gotten any farther?
Is it possible that you're working in a VM?

trisimix, 2018-10-15:
sorry for the zombie issue

closed 2018-10-15 by trisimix