kiibohd/controller #181: Error building kll from input-club configurator -- KEY_LCK2

leshow, 2017-02-10:
I have a layout I quite like for the ergodox infinity that I put together using the inputclub online configurator. I followed the instructions in the wiki, except instead of creating my own kll files I moved the `MDErgo-Default-*.kll` files to `kll/layouts/` and renamed them to to `mine*.kll`

I then edited the ergodox.bash file:

```
DefaultMap="mine0 lcdFuncMap"
PartialMaps[1]="mine1"
PartialMaps[2]="mine2"
PartialMaps[3]="mine3"
PartialMaps[4]="mine4"
```

```
Keyboards - master! ❯ bash ergodox.bash -f -o my_layout
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c
-- Bootloader Type:
dfu
-- Detected Scan Module Source Files:
Scan/ISSILed/led_scan.c;Scan/MatrixARM/matrix_scan.c;Scan/STLcd/lcd_scan.c;Scan/UARTConnect/connect_scan.c;Scan/Infinity_Ergodox/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Detected Layout Files:
/home/evan/personal/config/kiibohd/Scan/ISSILed/capabilities.kll
/home/evan/personal/config/kiibohd/Scan/MatrixARM/capabilities.kll
/home/evan/personal/config/kiibohd/Scan/STLcd/capabilities.kll
/home/evan/personal/config/kiibohd/Scan/UARTConnect/capabilities.kll
/home/evan/personal/config/kiibohd/Macro/PartialMap/capabilities.kll
/home/evan/personal/config/kiibohd/Output/pjrcUSB/capabilities.kll
/home/evan/personal/config/kiibohd/Scan/Infinity_Ergodox/scancode_map.kll
/home/evan/personal/config/kiibohd/Scan/Infinity_Ergodox/leftHand.kll
/home/evan/personal/config/kiibohd/Scan/Infinity_Ergodox/slave1.kll
/home/evan/personal/config/kiibohd/Scan/Infinity_Ergodox/rightHand.kll
/home/evan/personal/config/kiibohd/kll/layouts/mine0.kll
/home/evan/personal/config/kiibohd/kll/layouts/lcdFuncMap.kll
/home/evan/personal/config/kiibohd/kll/layouts/mine1.kll
/home/evan/personal/config/kiibohd/kll/layouts/mine2.kll
/home/evan/personal/config/kiibohd/kll/layouts/mine3.kll
/home/evan/personal/config/kiibohd/kll/layouts/mine4.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /home/evan/personal/config/kiibohd/Keyboards/my_layout
Scanning dependencies of target kiibohd.elf
[  3%] Building C object CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.obj
In file included from /home/evan/personal/config/kiibohd/Macro/PartialMap/macro.c:30:0:
/home/evan/personal/config/kiibohd/Keyboards/my_layout/generatedKeymap.h:241:27: error: 'KEY_LCK2' undeclared here (not in a function)
 Guide_RM( 88 ) = { 1, 15, KEY_LCK2, 0 };
                           ^
CMakeFiles/kiibohd.elf.dir/build.make:277: recipe for target 'CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.obj' failed
make[2]: *** [CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/macro.c.obj] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/kiibohd.elf.dir/all' failed
make[1]: *** [CMakeFiles/kiibohd.elf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Error in make. Exiting...
```

The offending line seems ot be in my layer 1 kll, I believe this is the part it doesn't like
```
U"FUNCTION1" : U"LCK2";
```

haata, 2017-02-14:
You didn't add stdFuncMap or lcdFuncMap as the last layer.

geeker342, 2017-06-01:
Haata, can you elaborate? It looks like leshow defined DefaultMap="mine0 lcdFuncMap" which has lcdFuncMap. Can you be specific as to what you mean by last layer? Do you have to add stdFuncMap or lcdFuncMap to each PartialMaps[x] definition?

haata, 2017-06-02:
The lcdFuncMap is a kll file that overrides the layer shift calls to do two
things:
1) Update the LCD screen with the layer information
2) convert non-standard USB codes into kll capabilities. I.e U"LOCK1" to
layerLock(1)

(2) is required for compilation to work.

closed 2017-10-03 by haata