kiibohd/controller #184: Build error: make[2]: execvp: ../kll/kll.py: Permission denied

wilecoyote2015, 2017-02-17:
Hello all,
I have followed the steps described in the build and installation guide to set up my system (Arch Linux, 64Bit). Using any .bash script as well as manual build with cmake fails with terminal output:

`[  4%] Generating KLL Layout
make[2]: execvp: ../kll/kll.py: permission denied
make[2]: *** [CMakeFiles/kiibohd.elf.dir/build.make:68: generatedKeymap.h] Error 127
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

haata, 2017-02-17:
It looks like you cloned or ran the build as root.

You may need to start with a fresh git clone.

wilecoyote2015, 2017-02-18:
Ah, thank you!
Solved it this way, so the issue can be closed.

closed 2017-02-18 by wilecoyote2015