kiibohd/controller #185: USB 2 vs USB 3 problems

patrickdepinguin, 2017-02-18:
I have a few problems when I plug in my Infinity 60% LED in an USB2 port, while these problems are not present when I plug into an USB3 port on the same computer:

* some special mappings do not work, among which: CONS"VOLUMEDOWN", CONS"VOLUMEUP", CONS"MUTE", SYS"SLEEP". When pressing the keys corresponding to these mappings, nothing happens.
* it is possible to freeze the keyboard by alternating between CONS"VOLUMEDOWN" and CONS"VOLUMEUP" a few times, for example 4 times the combination (8 presses in total). After this freeze, not any key works anymore, until you reset the keyboard by replugging it.
* I wanted to check the matrixDebug option to see more about these problems but: the debug mode does not work in USB2, nothing connects to /dev/ttyACM0. When I plug the keyboard, I see the following extra messages in dmesg:

```
[  +0.011209] input: failed to attach handler kbd to device input499, error: -16
[  +0.938186] cdc_acm 1-1.2:1.3: acm_port_activate - usb_submit_urb(ctrl irq) failed
```
where the last message repeats continuously every second.

The full plug log on USB2 is:
```
[260432.451205] usb 1-1.2: new full-speed USB device number 101 using ehci-pci
[260432.537893] usb 1-1.2: config 1 has an invalid interface number: 6 but max is 5
[260432.537901] usb 1-1.2: config 1 has no interface number 5
[260432.539274] usb 1-1.2: New USB device found, idVendor=1c11, idProduct=b04d
[260432.539280] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[260432.539284] usb 1-1.2: Product: Keyboard - Infinity_60%_LED PartialMap pjrcUSB full
[260432.539287] usb 1-1.2: Manufacturer: Input Club
[260432.539290] usb 1-1.2: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128vlf5
[260432.541428] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:1C11:B04D.0100/input/input501
[260432.592477] hid-generic 0003:1C11:B04D.0100: input,hidraw0: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:1a.0-1.2/input0
[260432.594462] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:1C11:B04D.0101/input/input502
[260432.645473] hid-generic 0003:1C11:B04D.0101: input,hidraw1: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:1a.0-1.2/input1
[260432.647868] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:1C11:B04D.0102/input/input503
[260432.659197] input: failed to attach handler kbd to device input503, error: -16
[260432.659355] hid-generic 0003:1C11:B04D.0102: input,hidraw2: USB HID v1.11 Device [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:1a.0-1.2/input2
[260432.659937] cdc_acm 1-1.2:1.3: ttyACM0: USB ACM device
[260432.662409] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.6/0003:1C11:B04D.0103/input/input504
[260432.662507] hid-generic 0003:1C11:B04D.0103: input,hidraw3: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:1a.0-1.2/input6
[260433.333074] cdc_acm 1-1.2:1.3: acm_port_activate - usb_submit_urb(ctrl irq) failed
[260434.334200] cdc_acm 1-1.2:1.3: acm_port_activate - usb_submit_urb(ctrl irq) failed
[260435.335318] cdc_acm 1-1.2:1.3: acm_port_activate - usb_submit_urb(ctrl irq) failed
...
```

The full log on USB3 is:
```
[260492.320500] usb 3-4: new full-speed USB device number 17 using xhci_hcd
[260492.486094] usb 3-4: config 1 has an invalid interface number: 6 but max is 5
[260492.486101] usb 3-4: config 1 has no interface number 5
[260492.487376] usb 3-4: New USB device found, idVendor=1c11, idProduct=b04d
[260492.487382] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[260492.487386] usb 3-4: Product: Keyboard - Infinity_60%_LED PartialMap pjrcUSB full
[260492.487389] usb 3-4: Manufacturer: Input Club
[260492.487392] usb 3-4: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128vlf5
[260492.489650] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1C11:B04D.0104/input/input505
[260492.540767] hid-generic 0003:1C11:B04D.0104: input,hidraw0: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:14.0-4/input0
[260492.542865] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.1/0003:1C11:B04D.0105/input/input506
[260492.593774] hid-generic 0003:1C11:B04D.0105: input,hidraw1: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:14.0-4/input1
[260492.595709] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.2/0003:1C11:B04D.0106/input/input507
[260492.646673] hid-generic 0003:1C11:B04D.0106: input,hidraw2: USB HID v1.11 Device [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:14.0-4/input2
[260492.647194] cdc_acm 3-4:1.3: ttyACM0: USB ACM device
[260492.649468] input: Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.6/0003:1C11:B04D.0107/input/input508
[260492.649577] hid-generic 0003:1C11:B04D.0107: input,hidraw3: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60%_LED PartialMap pjrcUSB full] on usb-0000:00:14.0-4/input6
```

I don't know if these three problems are actually related, but at least the common factor is USB2.
I want to stress that none of these problems happen on USB3: the volume keys work fine, no freeze, and I can use the debug console without problem.

This is on Linux 4.4.26-gentoo.

The firmware was built manually with kll commit fa41fab82e71e15f568fd23e6fdf6741442bfdc6 and controller commit 3b6723c40d1de6ea4223d01223cfbdf17a3ec004.

If there is anything I can do to help debug this, let me know.

einalex, 2017-02-23:
I have the same problem with my infinity ergodox regardless of USB2 or USB3. VOLUMEUP and VOLUMEDOWN don't send keycodes anymore.
both keys result in this xev capture:

KeymapNotify event, serial 34, synthetic NO, window 0x0,
    keys:  1   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 34, synthetic NO, window 0x1e00001,
    mode NotifyGrab, detail NotifyAncestor

FocusIn event, serial 34, synthetic NO, window 0x1e00001,
    mode NotifyUngrab, detail NotifyAncestor

haata, 2017-10-03:
@einalex which VOLUMEDOWN are you using?

USB or Consumer Control?

haata, 2017-10-03:
Hmm, Consumer Control should be the correct one.

The usb_submit_urb is concerning. Would either of you mind trying again with recent firmware?
This is one of those kind of issues where I'd love to see what my usb analyzer says about it.

einalex, 2017-10-03:
works for me now

einalex, 2017-10-03:
using

U"DOWN" : CONS"VOLUMEDOWN";
U"UP" : CONS"VOLUMEUP";

haata, 2017-10-03:
Awesome :+1:

patrickdepinguin, 2017-10-05:
This issue needs to be reopened, as I am still seeing the issues I reported. It seems @einalex was facing another problem.

Summary of the current situation:
Kernel: 4.12.5
USB3: everything works fine, USB2: mentioned problems still present.

dmesg log on USB2:
```
[ +13.534238] usb 1-1.6: new full-speed USB device number 4 using ehci-pci
[  +0.079911] usb 1-1.6: New USB device found, idVendor=1c11, idProduct=b04d
[  +0.000004] usb 1-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 1-1.6: Product: Keyboard - Infinity_60_LED PixelMap pjrcUSB full
[  +0.000002] usb 1-1.6: Manufacturer: Input Club
[  +0.000002] usb 1-1.6: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128vlf5
[  +0.002280] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:1C11:B04D.000C/input/input33
[  +0.052192] hid-generic 0003:1C11:B04D.000C: input,hidraw1: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input0
[  +0.001565] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.1/0003:1C11:B04D.000D/input/input34
[  +0.052440] hid-generic 0003:1C11:B04D.000D: input,hidraw2: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input1
[  +0.001546] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.2/0003:1C11:B04D.000E/input/input35
[  +0.017293] input: failed to attach handler kbd to device input35, error: -16
[  +0.000293] hid-generic 0003:1C11:B04D.000E: input,hidraw3: USB HID v1.11 Device [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input2
[  +0.000329] cdc_acm 1-1.6:1.3: ttyACM0: USB ACM device
[  +0.002241] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.6/0003:1C11:B04D.000F/input/input36
[  +0.000296] hid-generic 0003:1C11:B04D.000F: input,hidraw4: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input6
```

Notice the error line:
```[  +0.017293] input: failed to attach handler kbd to device input35, error: -16```

while on USB3 (good):
```
[  +6.345162] usb 3-4: new full-speed USB device number 10 using xhci_hcd
[  +0.169399] usb 3-4: New USB device found, idVendor=1c11, idProduct=b04d
[  +0.000004] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 3-4: Product: Keyboard - Infinity_60_LED PixelMap pjrcUSB full
[  +0.000002] usb 3-4: Manufacturer: Input Club
[  +0.000002] usb 3-4: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128vlf5
[  +0.001714] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1C11:B04D.0010/input/input37
[  +0.052282] hid-generic 0003:1C11:B04D.0010: input,hidraw1: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-4/input0
[  +0.001121] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.1/0003:1C11:B04D.0011/input/input38
[  +0.051879] hid-generic 0003:1C11:B04D.0011: input,hidraw2: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-4/input1
[  +0.001092] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.2/0003:1C11:B04D.0012/input/input39
[  +0.051828] hid-generic 0003:1C11:B04D.0012: input,hidraw3: USB HID v1.11 Device [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-4/input2
[  +0.000286] cdc_acm 3-4:1.3: ttyACM0: USB ACM device
[  +0.001488] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.6/0003:1C11:B04D.0013/input/input40
[  +0.000194] hid-generic 0003:1C11:B04D.0013: input,hidraw4: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-4/input6
```

Trying to use the debug tty /dev/ttyACM0 still doesn't work on USB2. When I try to make a connection with picocom, I see following message in dmesg:

```[  +3.908963] cdc_acm 1-1.2:1.3: acm_port_activate - usb_submit_urb(ctrl irq) failed```

I'm generally fine to have this restriction on USB port compatibility, but I think it better be documented. Of course, if it _can_ be fixed, all the better.

This was tested on kiibohd/controller@65252776370 and kiibohd/kll@ba099303d96 .

Here is a snippet from my kll file:
```
U"MINUS" : CONS"VOLUMEDOWN";
U"EQUAL" : CONS"VOLUMEUP";
U"BACKSLASH" : CONS"MUTE";
```

haata, 2017-10-05:
Hmm, let me think about how to debug this further.

@patrickdepinguin would you mind enabling HID debug mode? It's pretty harmless (I just have it permanently enabled on my Linux boxes).

https://github.com/kiibohd/controller/wiki/Reporting-Issues#usb-hid-debug-mode-linux 

dmesg looks like this

```bash
[346010.266347] usb 3-1.4.4.3: new full-speed USB device number 74 using xhci_hcd
[346010.368424] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 0
[346010.369471] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0200 wIndex=0x0000 wLength=1
[346010.369542] input: Input Club Keyboard - WhiteFox PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4.3/3-1.4.4.3:1.0/0003:1C11:B04D.0052/input/input89
[346010.423466] hid-generic 0003:1C11:B04D.0052: input,hidraw5: USB HID v1.11 Keyboard [Input Club Keyboard - WhiteFox PixelMap pjrcUSB full] on usb-0000:00:14.0-1.4.4.3/input0
[346010.423691] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 1
[346010.424482] input: Input Club Keyboard - WhiteFox PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4.3/3-1.4.4.3:1.1/0003:1C11:B04D.0053/input/input90
[346010.480074] hid-generic 0003:1C11:B04D.0053: input,hidraw6: USB HID v1.11 Keyboard [Input Club Keyboard - WhiteFox PixelMap pjrcUSB full] on usb-0000:00:14.0-1.4.4.3/input1
[346010.480295] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 2
[346010.481021] input: Input Club Keyboard - WhiteFox PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4.3/3-1.4.4.3:1.2/0003:1C11:B04D.0054/input/input91
[346010.536430] hid-generic 0003:1C11:B04D.0054: input,hidraw7: USB HID v1.11 Device [Input Club Keyboard - WhiteFox PixelMap pjrcUSB full] on usb-0000:00:14.0-1.4.4.3/input2
[346010.536698] cdc_acm 3-1.4.4.3:1.3: ttyACM0: USB ACM device
[346010.537425] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 6
[346010.538408] input: Input Club Keyboard - WhiteFox PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4.3/3-1.4.4.3:1.6/0003:1C11:B04D.0055/input/input92
[346010.538508] hid-generic 0003:1C11:B04D.0055: input,hidraw8: USB HID v1.11 Mouse [Input Club Keyboard - WhiteFox PixelMap pjrcUSB full] on usb-0000:00:14.0-1.4.4.3/input6
```

patrickdepinguin, 2017-10-07:
Sure, I will enable HID debug mode. As the keyboard is at my workplace, I won't be able to report back until start of next week.

patrickdepinguin, 2017-10-09:
dmesg log on USB2 with HID debug enabled:
```
[ +19.797196] usb 1-1.6: new full-speed USB device number 9 using ehci-pci
[  +0.080311] usb 1-1.6: New USB device found, idVendor=1c11, idProduct=b04d
[  +0.000005] usb 1-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 1-1.6: Product: Keyboard - Infinity_60_LED PixelMap pjrcUSB full
[  +0.000003] usb 1-1.6: Manufacturer: Input Club
[  +0.000001] usb 1-1.6: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128vlf5
[  +0.001216] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 0
[  +0.001116] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0200 wIndex=0x0000 wLength=1
[  +0.000118] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:1C11:B04D.0017/input/input162
[  +0.051757] hid-generic 0003:1C11:B04D.0017: input,hidraw0: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input0
[  +0.000253] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 1
[  +0.001121] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.1/0003:1C11:B04D.0018/input/input163
[  +0.052452] hid-generic 0003:1C11:B04D.0018: input,hidraw1: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input1
[  +0.000284] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 2
[  +0.000974] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.2/0003:1C11:B04D.0019/input/input164
[  +0.016341] input: failed to attach handler kbd to device input164, error: -16
[  +0.001013] hid-generic 0003:1C11:B04D.0019: input,hidraw2: USB HID v1.11 Device [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input2
[  +0.000335] cdc_acm 1-1.6:1.3: ttyACM0: USB ACM device
[  +0.001053] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 6
[  +0.000950] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.6/0003:1C11:B04D.001A/input/input165
[  +0.000167] hid-generic 0003:1C11:B04D.001A: input,hidraw3: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:1a.0-1.6/input6
[  +0.024025] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0200 wIndex=0x0000 wLength=1
[  +0.000012] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0201 wIndex=0x0001 wLength=2
```

Trying to access /dev/ttyUSBACM0 gives still (no additional output):
```
[ +12.390881] cdc_acm 1-1.6:1.3: acm_port_activate - usb_submit_urb(ctrl irq) failed
```

For USB3:
```
[Oct 6 03:29] usb 3-3: new full-speed USB device number 10 using xhci_hcd
[  +0.170141] usb 3-3: New USB device found, idVendor=1c11, idProduct=b04d
[  +0.000002] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 3-3: Product: Keyboard - Infinity_60_LED PixelMap pjrcUSB full
[  +0.000001] usb 3-3: Manufacturer: Input Club
[  +0.000001] usb 3-3: SerialNumber: FFFFFFFFFFFF0029000E001224404E45 - mk20dx128vlf5
[  +0.000555] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 0
[  +0.000772] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0200 wIndex=0x0000 wLength=1
[  +0.000069] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1C11:B04D.001B/input/input166
[  +0.051730] hid-generic 0003:1C11:B04D.001B: input,hidraw0: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-3/input0
[  +0.000181] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 1
[  +0.000694] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:1C11:B04D.001C/input/input167
[  +0.052137] hid-generic 0003:1C11:B04D.001C: input,hidraw1: USB HID v1.11 Keyboard [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-3/input1
[  +0.000170] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 2
[  +0.000604] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:1C11:B04D.001D/input/input168
[  +0.052120] hid-generic 0003:1C11:B04D.001D: input,hidraw2: USB HID v1.11 Device [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-3/input2
[  +0.000225] cdc_acm 3-3:1.3: ttyACM0: USB ACM device
[  +0.000558] drivers/hid/usbhid/hid-core.c: HID probe called for ifnum 6
[  +0.000533] input: Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.6/0003:1C11:B04D.001E/input/input169
[  +0.051829] hid-generic 0003:1C11:B04D.001E: input,hidraw3: USB HID v1.11 Mouse [Input Club Keyboard - Infinity_60_LED PixelMap pjrcUSB full] on usb-0000:00:14.0-3/input6
[  +0.017760] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0200 wIndex=0x0000 wLength=1
[  +0.000009] drivers/hid/usbhid/hid-core.c: submitting ctrl urb: Set_Report wValue=0x0201 wIndex=0x0001 wLength=2
```

Let me know what I can do to help further...

haata, 2019-02-18:
I've been doing a bunch of USB related fixes recently. Please try the latest firmware. Re-open if there are still issues.

closed 2019-02-18 by haata