kiibohd/controller #189: All keys on row 1 produce an extra scancode

abrahamssonmathias, 2017-03-05:
Hi,

I've used your guide for the Teensy microcontroller to connect an old Sharp PC4600 keyboard to a Teensy 3.2 board with your firmware.

When I type, I get extra scancodes for any key connected to Row 1. It seems to bring keystrokes from the next column with it, so for every key hit on column N, I get an extra scancode for same row but column N+1. I've checked my solders and code and can't seem to find anything wrong. Is this a timing issue?

Any help at all is appreciated.

abrahamssonmathias, 2017-03-07:
After some troubleshooting on my own, I've isolated the problem to somewhere on the PCB. Has anyone had a similar issue?

simonebracaloni, 2017-07-12:
We have exactly the same issue here. How do you solve that?

steinerlein, 2017-08-07:
If this is still an issue or for anyone coming here later:
This is typically an issue when you have a short between two of the rows.

simonebracaloni, 2017-08-18:
After checking multiple time the PCB. I solved the problem defining a delay in the scan, I added this in my matrix.h:
```
// delay in microseconds before and after each strobe change during matrix scan
#define STROBE_DELAY  10
```
I found that, in my case the issue is caused by the diodes I used for the matrix (ST BAT46). Using another diode model (e.g. 1N4005) fix the problem as well.
I guess is a timing issue, maybe a transient signal on the same row when the strobe switch to the following column.

abrahamssonmathias, 2017-09-10:
simonebracaloni, thanks! Will test this out as soon as I get my Teensy back.

haata, 2017-10-03:
Yeah, depending on the type of diodes being used, there may be lingering stray voltages.

Strobe delays definitely help in this case.
Feel free to re-open if you're still having issues.

closed 2017-10-03 by haata