kiibohd/controller #190: Cannot get key combinations to work with WhiteFox

aredondo, 2017-03-06:
My basic keyboard layout is working fine with a WhiteFox, but I can't seem to be able to make combinations to work. For example, I've got for the default layer:

```
U"HASH" : U"BACKSLASH";
U"BACKSPACE" : U"BACKTICK";
U"BACKTICK" : U"FUN2";
U"BACKSLASH" : U"BACKSPACE";
U"DELETE" : U"FUN3";
U"CAPSLOCK" : U"LCTRL";
U"CTRL" : U"FUN1";
U"FUNCTION1" : U"LALT";
U"LALT" : U"LGUI";
U"RALT" : U"RGUI";
U"GUI" : U"RALT";
```

And for the first functional one:

```
U"1" : U"F1";
U"2" : U"F2";
U"3" : U"F3";
U"4" : U"F4";
U"5" : U"F5";
U"6" : U"F6";
U"7" : U"F7";
U"8" : U"F8";
U"9" : U"F9";
U"0" : U"F10";
U"MINUS" : U"F11";
U"EQUAL" : U"F12";
U"BACKSLASH" : U"DELETE";
U"UP" : U"PAGEUP";
U"LEFT" : U"HOME";
U"DOWN" : U"PAGEDOWN";
U"RIGHT" : U"END";
```

And all works great. Keys map correctly and I can issue function keys by using the first functional layer. I have mapped multimedia keys using the second functional layer (not shown) and it also works well.

But adding the following combination to the default layer does nothing:

```
U"LALT" + U"8" : CONS"VOLUMEUP";
```

If I press left `ALT`+`F8` I just get `•`. The following combinations on the first alternative layer also don't work (which are what I really want; the previous is just a test):

```
U["ALT"] + U["7"] : CONS"SCANPREVIOUSTRACK";
U["ALT"] + U["8"] : CONS"PAUSEPLAY";
U["ALT"] + U["9"] : CONS"SCANNEXTTRACK";
U["ALT"] + U["0"] : CONS"MUTE";
U["ALT"] + U["MINUS"] : CONS"VOLUMEDOWN";
U["ALT"] + U["EQUAL"] : CONS"VOLUMEUP";
```

(Square brackets don't seem to matter.)

No error is raised, and the rest of the keyboard configuration is compiled and loaded fine. Combinations keys seem to be largely and simply ignored.

I cloned the repository and I am using the updated `master` branch. The syntax I am using is as described in the user guide for KLL.

I would appreciate any help about what I may be missing here.

TheOtherDave, 2017-03-06:
At least on a Mac, that key combination is how you'd normally type "•"... It's not a macro or anything; just part of the QWERTY keyboard layout (Dvorak, too). Maybe because you're on the default layer, the firmware is just passing the keystrokes through to the OS? Dunno, that's just the first thing that popped into my head.

Hope that helps,
- Dave Sweeris

aredondo, 2017-03-06:
Hi there Dave — Yes, you are correct. That key combination produces `•` by default in macOS. That’s why I say that key combinations seem to be entirely ignored when added to the keyboard configuration.

TheOtherDave, 2017-03-06:
What happens if you assign that key combination to a functional layer?

I'm just wondering if the firmware handles it differently since you're on the default layer.

- Dave Sweeris

aredondo, 2017-03-07:
The behavior is the same. The combination is just ignored, even though simple key assignments work well.

haata, 2017-03-07:
Try using `U"VolumeUp"` instead. The USB HID spec is annoying like this and Apple probably didn't implement the Consumer Control VolumeUp/Down codes.

aredondo, 2017-03-07:
Thanks for the feedback.

`CONS"VOLUMEUP"` works fine with a regular functional layer + key, that's not really the issue. For example, the following defined in a functional layer works well:

```
U["EQUAL"] : CONS"VOLUMEUP";
```

But this does not:

```
U["ALT"] + U["EQUAL"] : CONS"VOLUMEUP";
```

So my issue is not with what is on the right-hand side of the assignment, but with the key combination `U["ALT"] + U["EQUAL"]`—or any other.

Do you use key combinations with the WhiteFox without issues?

aredondo, 2017-03-11:
After more tinkering I was able to figure out why this was not working for me.

I have remapped all`CTRL`, `ALT` and `COMMAND` keys from the default configuration, and then tried to use them for the combinations. But when doing this, it is still necessary to refer to the keys using their initial code, not the remapped one. So the configuration needed to swap `COMMAND` and `ALT`, and then define `ALT + =` is actually the following:

```
U"LALT" : U"LGUI";
U["LGUI"] + U["EQUAL"] : CONS"VOLUMEUP";
```

After closing inspection, this behaviour is actually covered in the the KLL spec guide—but to me it was not so obvious to understand until actually running into the issue.

To simplify key remappings and be able to set further combinations using the labels that I see in the keyboard, what I have done is specify physical key changes in the `scancode_map.kll` file. Then I can refer to keys using their new code in the other configuration files.

Thanks for all the feedback.

closed 2017-03-11 by aredondo

haata, 2017-03-11:
Ahhh, yep. Sorry about that.

Yes, references refer to the basemap (or the previous file if you are merging layouts). Once I switch master over to the new KLL compiler it should be easier to debug this (lots of debug output, as well as compiling back to KLL files).