kiibohd/controller #192: macOS-specific functions

pointlessone, 2017-03-19:
Hi,
I'd like to map macOS-specific function such as Mission Control and Dashboard. I assume those are somewhere in Consumer Control Codes but I can not find anything relevant. Are they documented? Does Apple use custom CONS values?

JBonifacio, 2017-04-17:
I looked through the overleaf document and didn't see these either... you might want to use macros instead.

_TRIGGER_ : U["LCTRL","RCTRL"] + U"UP"

jceaser, 2017-09-19:
these actions are not listed in the Consumer control codes, I think you do a macro.

haata, 2017-10-03:
I'm not aware of any USB HID specific codes for macOS functions (though I really want to be able to send the Fn key in a generic way...).

If someone is aware of a standard/non-standard USB code for macOS function keys, let me know. :D

closed 2017-10-03 by haata