kiibohd/controller #193: Keyboard will not output # ~ signs

paramecium, 2017-04-09:
Current setup is 
```
S0x21 : '#';
S0x22 : '~';
S0x23 : U"Hash"
```
On the lower layer
Output is 

```
S0x21 = £ 
S0x22 = ¬
```

U"Hash" Gives this error in the debug serial connection
`
WARNING - USB Code not within 4-49 (0x4-0x31), 51-155 (0x33-0x9B), 157-164 (0x9D-0xA4), 176-221 (0xB0-0xDD) or 224-231 (0xE0-0xE7) NKRO Mode: 0x32
`
System language is set to English UK if that affects anything.

Anyone got any ideas?

DavidK-M, 2017-07-25:
I have the same problem.

It appears that in `controller/Output/pjrcUSB/output_com.c`
that the key "50" (0x38) is excluded from being placed in the `USBKeys_Changed` structure (in the USB NKRO mode).

I have not yet worked out why this key has special treatment. It is listed like any other key in the USB HID document. (Page 54, version 1.12 of the USB HID Usage Tables document).

Perhaps the reporting for NKRO is a bit special... and the HASH key is done differently.

I know that this has worked in the past - but there were some changes made to fix a problem with the keyboard not switching into the correct mode after booting - it was remaining in the boot keyboard mode and then the left-hand modifier keys didn't work (it had to be unplugged and plugged back in again).

For information, I'm using a custom keyboard driver on Windows, the machine is set to UK English.

haata, 2017-10-03:
I need to re-review this. Basically, Linux USB code handling gets a bit funky for some of the characters.

algernon, 2018-06-05:
FWIW, we used to disable the non-US pound scancode in Kaleidoscope too, due to reports of it not working well under Linux. We unmasked it a while ago, upon a user's request, and he's been using it happily since (on Windows, because Linux still translates it to Backslash - but we found no other issues other than that).

haata, 2018-06-07:
Thanks @algernon I've gone ahead and fixed this.
Closing, let me know if there are still issues.

closed 2018-06-07 by haata