kiibohd/controller #194: Combination trigger behaves unexpectedly

JBonifacio, 2017-04-11:
I set up the following in a KLL file:
`U["LSHIFT","RSHIFT"] + U"ESC" : '~';`

When using the trigger, if I let go of LSHIFT or RSHIFT, the backtick (`) repeats until another key is pressed.

[image]

In certain fields, it looks like the input is acting oddly.  For example, in Chrome, I get this result:

[image]

jceaser, 2017-04-18:
I had something similar when I set up U"RSHIFT","RSHIFT" : ')';

JBonifacio, 2017-04-25:
So I just made some changes I saw on the input.club forums to see if I could get it to work.  I modified the scan_map kll to add the blockHold() and blockKey() capabilities.  I added this to my default layout:

```
# Macro
# Example of blocking ESC (0x29) when holding either Shift key
# Currently, kll only supports numbers (this may be fixed in a future kll spec)
U["LShift", "RShift"] : blockHold( 0x29 );
U"Esc" : blockKey( 0x29 );

# Now that Esc is blocked, it's possible to use this macro
U["LShift", "RShift"] + U"Esc" : '~';
```

I'm able to get it to compile, but know LSHIFT and RSHIFT codes are sending out for anything but the ESC/Tilde macro I added.  

I'm not sure if I'm missing something.