kiibohd/controller #196: LGUI Crash

jceaser, 2017-04-17:
I have an ErgoDox Infinity with the LGUI key defined as an ESC. When pressing this key the keyboard seams to enter into an infinite loop issuing ESC keys as fast as it can. On the Mac, I open the Keyboard Viewer and the esc registerers as down constantly. I have to unplug the keyboard, click ESC on another keyboard or the Keyboard viewer then I can re plugin the Ergodox. I'm not sure what else I can post about this issue or what evidence would be needed here to help debug this. Would the JSON output of my keyboard layout help?

I have defined ESC on another key and I, so far, have no issue with that key.

jceaser, 2017-04-30:
some more testing: it's not a hardware issue (bad switch or something). If I install the same binary on the other side (right hand firmware on left hand keyboard) I get the same behavior with the matching switch while the previous misbehaving switch acts as expected (of course swapped around as the driver is on the wrong side).

haata, 2017-10-03:
Is this still an issue for you with the recent master branch updates?