kiibohd/controller #197: how to flash ergodox infinity with .hex file

tobsn, 2017-04-22:
There is literally no explanation anywhere about how to flash each side of the ergodox with the .hex the input club configurator downloads or inside the massdrop configurator. The FAQ on input club is almost a carbon copy from the 60% FAQ and has also no info.

How do you flash the downloadable .hex files or the source code onto each side?!

nomaded, 2017-04-22:
This is for the Infinity ErgoDox keyboard project, not the original ErgoDox project. For the Infinity ErgoDox, you should be using the configurator found at https://input.club/configurator-ergodox/, which will generate .kll files, not .hex files. More info for what to do with the .kll files can be found at https://github.com/kiibohd/controller/tree/master/Keyboards.

TheOtherDave, 2017-04-22:
> On Apr 22, 2017, at 1:00 PM, Tobias O. <notifications@github.com> wrote:
> 
> There is literally no explanation anywhere about how to flash each side of the ergodox with the .hex the input club configurator downloads or inside the massdrop configurator. The FAQ on input club is almost a carbon copy from the 60% FAQ and has also no info.
> 
> How do you flash the downloadable .hex files or the source code onto each side?!
> 
I'm not familiar with the Massdrop configurator.
On macOS with the Input Club configurator (https://input.club/configurator-ergodox/ <https://input.club/configurator-ergodox/>), you download & decompress the file (it’ll be called “MDErgo1-“ followed by some randomish string, then open a terminal window and type:
cd ~/Downloads/MDErgo1-some_long_string_of_text
(or wherever you saved it to), then put your keyboard into Flash mode, and type:
dfu-util -D left_kiibohd.dfu.bin
I know it only says “left”, but that’s all I need to do to get both halves flashed (and that worked as recently as maybe two weeks ago). I can’t recall if I had to install the dfu-util tool or it’s part of the OS. My understanding is that it’s a somewhat standard tool for programming micro-controllers, so it shouldn’t be too hard to find if it’s not part of the default os install.

I hope that helps!

- Dave Sweeris

TheOtherDave, 2017-04-22:
> On Apr 22, 2017, at 2:01 PM, David Sweeris <davesweeris@mac.com> wrote:
> 
> 
>> On Apr 22, 2017, at 1:00 PM, Tobias O. <notifications@github.com <mailto:notifications@github.com>> wrote:
>> 
>> There is literally no explanation anywhere about how to flash each side of the ergodox with the .hex the input club configurator downloads or inside the massdrop configurator. The FAQ on input club is almost a carbon copy from the 60% FAQ and has also no info.
>> 
>> How do you flash the downloadable .hex files or the source code onto each side?!
>> 
> I'm not familiar with the Massdrop configurator.
> On macOS with the Input Club configurator (https://input.club/configurator-ergodox/ <https://input.club/configurator-ergodox/>), you download & decompress the file (it’ll be called “MDErgo1-“ followed by some randomish string, then open a terminal window and type:
> cd ~/Downloads/MDErgo1-some_long_string_of_text
> (or wherever you saved it to), then put your keyboard into Flash mode, and type:
> dfu-util -D left_kiibohd.dfu.bin
> I know it only says “left”, but that’s all I need to do to get both halves flashed (and that worked as recently as maybe two weeks ago). I can’t recall if I had to install the dfu-util tool or it’s part of the OS. My understanding is that it’s a somewhat standard tool for programming micro-controllers, so it shouldn’t be too hard to find if it’s not part of the default os install.
> 
> I hope that helps!
> 
> - Dave Sweeris

Hehe, oops, you should probably type that 2nd command before putting your keyboard into flash mode. MacOS has a builtin on-screen keyboard you can use to press “enter” once you’re ready to flash it, if you don’t want to hook up another keyboard.

jceaser, 2017-05-01:
tobsn, you have not responded to any of the questions, so I thought I would ask if your still stuck. I have a massdrop ergo as well. Currently I use the software here to build my layout, but I started with the input.club GUI to get my initial KLL files.  I agree that documentation all around can use some improvement and I can help here. Here are some instructions I wrote that the kiibohd team incorporated, I'm willing to write more: https://github.com/kiibohd/controller/tree/master/Keyboards

Let me know where in the process you are and I or someone else here can try to help you through the process.

haata, 2017-10-03:
Closing, this firmware only works on the Infinity Ergodox (no .hex files).

closed 2017-10-03 by haata