kiibohd/controller #198: Some Control Codes do nothing

nazar-pc, 2017-05-09:
I found a few keys that do nothing, for instance:
```
U"Stop"
CONS"Stop"
CONS"AudioPlayer"
```

All of these (at least) do nothing when pressed, looks like they are not sent over USB or some garbage is sent, not sure how to confirm yet since keyboard doesn't show up as `/dev/usb/hiddevX` device unfortunately.

At the same time other codes work fine:
```
U"F1"  : CONS"Back";
U"F2"  : CONS"Forward";
U"F4"  : CONS"EmailReader";
U"F6"  : CONS"PausePlay";
```

jceaser, 2017-05-09:
what platform?

nazar-pc, 2017-05-09:
My OS is Ubuntu Linux

nazar-pc, 2017-05-09:
I have another keyboard and confirmed it really sends `CONS"Stop"` and system responds to the key correctly.

jceaser, 2017-05-09:
yea, sounds like a bug.

jceaser, 2017-05-10:
yea, I think this software get's confused in it's processing of layers or something, I have a key that throws the processor into an infinite loop

nazar-pc, 2017-05-10:
Yes, this `CONS"Stop"` is sent on Fn layer.
Here is keyboard config: https://github.com/nazar-pc/controller/commit/c357d76980d31d0d1b08b749e0cd50e9da35923c
And here is an Fn layer: https://github.com/nazar-pc/kll/commit/1d3722b7a984f10255e93da1f2a997fded098c39

Everything there works except effective `U"Function1" + U"F7" : CONS"Stop";`.

nazar-pc, 2017-05-11:
Out of curiosity assigned a few other actions to the layer and key where `CONS"Stop"` didn't work, and they are functional, so this particular `CONS"Stop"` is an issue, not a layout or wiring.

nazar-pc, 2017-05-15:
One more, `U"Keypad Decimal"` without any Fn layer also doesn't print anything

nazar-pc, 2017-05-25:
Here is one more: `SYS"ContextMenu"`.
Can anyone suggest how to debug this issue?

haata, 2017-05-25:
I've recently pushed a lot of updates, so you may want to try that.

Hmmm, have you tried using xev on Linux to see if you get those keypresses?
Also, the keyboard will need to be in NKRO mode iirc (that's something I can fix).

So using the debug cli you can check using the kbdProtocol command. Should return a 1 for NKRO (or 0 for bootmode).

nazar-pc, 2017-05-25:
Here is what xev shows when I press Fn+F7 (mapped to `CONS"Stop"`):
```
KeyPress event, serial 43, synthetic NO, window 0x7c00001,
    root 0x1e6, subw 0x0, time 63168646, (89,-21), root:(90,29),
    state 0x10, keycode 136 (keysym 0xff69, Cancel), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 43, synthetic NO, window 0x7c00001,
    root 0x1e6, subw 0x0, time 63168772, (89,-21), root:(90,29),
    state 0x10, keycode 136 (keysym 0xff69, Cancel), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

Keyboard is in NKRO mode.

I've tried to rebase against current master, but compilation fails now on Ubuntu 16.04 (I've removed old directory with generated files:
```
linux-gnu/Linux is supported.
Selected Generator: Unix Makefiles
linux-gnu.Rapoo_E9070.gcc.make
-- Build CPU Detected:
x86_64
-- Build OS Detected:
Linux 4.10.0-22-generic
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c;Lib/time.c
-- Bootloader Type:
Teensy
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc 
CMake Error at Lib/CMake/modules.cmake:103 (include):
  include could not find load file:

    /sources/Scan/MatrixARM/setup.cmake
Call Stack (most recent call first):
  Scan/Rapoo_E9070/setup.cmake:20 (AddModule)
  Lib/CMake/modules.cmake:103 (include)
  Lib/CMake/modules.cmake:152 (AddModule)
  CMakeLists.txt:169 (include)

-- Detected Scan Module Source Files:
Scan/Rapoo_E9070/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Found Ctags: /usr/bin/ctags (found version "5.9~svn20110310") 
-- Found PythonInterp: python3 (found suitable version "3.5.2", minimum required is "3") 
-- Detected Layout Files:
/sources/Macro/PartialMap/capabilities.kll
/sources/Output/HID-IO/capabilities.kll
/sources/Output/pjrcUSB/capabilities.kll
/sources/Scan/Rapoo_E9070/scancode_map.kll
/sources/kll/layouts/stdFuncMap.kll
/sources/kll/layouts/Rapoo_E9070_Fn.kll
-- Configuring incomplete, errors occurred!
See also "/sources/Keyboards/linux-gnu.Rapoo_E9070.gcc.make/CMakeFiles/CMakeOutput.log".
Error in cmake. Exiting...
```
What should I do in addition to simple rebasing?

nazar-pc, 2017-05-25:
OK, I've now rebased sources against master successfully and got broken keyboard:
```
$ lsusb
...
Bus 003 Device 029: ID 1c11:b04d  
...
```

Compilation output:
```
linux-gnu/Linux is supported.
Selected Generator: Unix Makefiles
linux-gnu.Rapoo_E9070.gcc.make
-- Build CPU Detected:
x86_64
-- Build OS Detected:
Linux 4.10.0-22-generic
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c;Lib/time.c
-- Bootloader Type:
Teensy
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc 
-- Detected Scan Module Source Files:
Scan/Devices/MatrixARM/matrix_scan.c;Scan/Rapoo_E9070/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Detected Layout Files:
/sources/Scan/Devices/MatrixARM/capabilities.kll
/sources/Macro/PartialMap/capabilities.kll
/sources/Output/HID-IO/capabilities.kll
/sources/Output/pjrcUSB/capabilities.kll
/sources/Scan/Rapoo_E9070/scancode_map.kll
/sources/kll/layouts/stdFuncMap.kll
/sources/kll/layouts/Rapoo_E9070_Fn.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /sources/Keyboards/linux-gnu.Rapoo_E9070.gcc.make
[  3%] Generating KLL Layout
kll ALPHA 0.5c.50ac60d70c9cc19bcaac69004731d2d268982ea9 - 2017-05-11 06:20:48 +0300
Scanning dependencies of target kiibohd
[  7%] Building C object CMakeFiles/kiibohd.dir/main.c.o
[ 11%] Building C object CMakeFiles/kiibohd.dir/Lib/mk20dx.c.o
[ 15%] Building C object CMakeFiles/kiibohd.dir/Lib/delay.c.o
[ 19%] Building C object CMakeFiles/kiibohd.dir/Lib/time.c.o
[ 23%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/MatrixARM/matrix_scan.c.o
[ 26%] Building C object CMakeFiles/kiibohd.dir/Scan/Rapoo_E9070/scan_loop.c.o
[ 30%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/macro.c.o
[ 34%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/result.c.o
[ 38%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/trigger.c.o
[ 42%] Building C object CMakeFiles/kiibohd.dir/Output/HID-IO/hidio_com.c.o
[ 46%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/output_com.c.o
[ 50%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_desc.c.o
[ 53%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_dev.c.o
[ 57%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_joystick.c.o
[ 61%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_keyboard.c.o
[ 65%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_mem.c.o
[ 69%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_mouse.c.o
[ 73%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_rawio.c.o
[ 76%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_serial.c.o
[ 80%] Building C object CMakeFiles/kiibohd.dir/Debug/cli/cli.c.o
[ 84%] Building C object CMakeFiles/kiibohd.dir/Debug/led/led.c.o
[ 88%] Building C object CMakeFiles/kiibohd.dir/Debug/print/print.c.o
[ 92%] Building C object CMakeFiles/kiibohd.dir/generatedPixelmap.c.o
[ 96%] Linking C executable kiibohd.elf
Creating iHex file to load:    kiibohd.teensy.hex
Creating Extended Listing:     kiibohd.lss
Creating Symbol Table:         kiibohd.sym
[ 96%] Built target kiibohd
Scanning dependencies of target SizeAfter
[100%] Chip usage for mk20dx256
         SRAM:  16%     10692/65536     bytes
        Flash:  12%     32444/262144    bytes
[100%] Built target SizeAfter
Firmware has been compiled into: 'linux-gnu.Rapoo_E9070.gcc.make'
/sources/Keyboards
```

eltang, 2017-05-25:
I believe this could possibly be related to #188.

nazar-pc, 2017-06-13:
@haata, I think you should have some keyboard with this firmware around, could you try to set some key to `U"Keypad Decimal"` and check if it works for you?

nazar-pc, 2018-01-14:
I haven't tried new versions of firmware since more than a half year ago. Should this be fixed in latest versions already?
The fact that some control codes are not working seems like a pretty major bug to me.

haata, 2018-01-16:
Not every OS implements all control codes.
The keyboard supports all the control codes, but it will be somewhat OS
dependent to which ones actually work.

nazar-pc, 2018-01-16:
Well, when on numpad `*` works, all other keys also work and `.` doesn't, I assume the issue is with keyboard, not control codes or OS.
Will you have time to debug this issue any time soon?

haata, 2018-01-16:
Which OS?

I'm actually getting close to debugging things like this.
I've been working on automating KLL testing and just got my first version
working: https://travis-ci.org/kiibohd/controller/jobs/329351666#L2063

nazar-pc, 2018-01-16:
Linux, can test on Windows if needed.

nazar-pc, 2018-06-08:
Any news on this issue?

haata, 2018-06-08:
Does xev do anything when these buttons are pressed?

nazar-pc, 2018-06-08:
Yes, here is how `Keypad Decimal` looks like (press/release):
```
KeyPress event, serial 30, synthetic NO, window 0x8e00001,
    root 0x145, subw 0x0, time 36353084, (520,-420), root:(2440,1020),
    state 0x10, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 33, synthetic NO, window 0x8e00001,
    root 0x145, subw 0x0, time 36353134, (520,-420), root:(2440,1020),
    state 0x10, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

nazar-pc, 2018-06-08:
`showkey --scancodes` shows scancodes for `Keypad Plus`, `Keypad Minus` and other keys, but not for `Keypad Decimal`, complete silence.
This is not specific to layout or OS, firmware sends `keysym 0x0, NoSymbol` for some keys. There are a few examples of those in first comments and I'm pretty sure you'll be able to reproduce it on any keyboard that is supported by this firmware with any host OS.

haata, 2018-06-08:
There's not a lot I can do for Linux as all Consumer Control keys are sent
the same way (at least from the keyboard). It's an OS bug/configuration
issue if they aren't working. Linux doesn't map every key in the USB HID
spec, as some keys are hardly ever used.

I'll try to figure out how to make them do something though. Usually it
just takes some effort setting up Linux to do this.

haata, 2018-06-08:
> Can you just confirm explicitly that Keypad Decimal specifically works
> for you on Mac or Windows?
>
> I suspect all keyboards to send keys the same way and this is the first
> keyboard with such behavior I've ever seen.

Backing up a second. Is this on the latest firmware?

nazar-pc, 2018-06-08:
No, on the firmware from about a year ago (approximately when this issue was reported, maybe a bit newer). If it works for you on latest, I'll try it too.

haata, 2018-06-08:
I tested Keypad Decimal a while back on the Kira prototype and it was
working. Though I've done a bunch of USB descriptor work earlier this week.

nazar-pc, 2018-06-08:
Is it all in master branch now?

haata, 2018-06-08:
Yep.
https://github.com/kiibohd/controller/commit/bac8dec83fc56cdf20dcc6f15629045145e4c413

nazar-pc, 2018-06-08:
Great, will try it and let you know if it works now

nazar-pc, 2018-06-15:
Well, it doesn't work at all just like the last time I've tried it.
Here is a commit that shows changes comparing to known working #200 so that it runs with latest master: https://github.com/nazar-pc/controller/commit/e3117e2642ad36a80c2c9140f523839c221c1862
Keys are not working, debugging interface prints enormous amounts of:
```
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Blow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 1ous Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overo Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
ERROR - Macro Trigger Event Overflow! Serious Bug! 144
```

nazar-pc, 2018-06-28:
@haata, any ideas why it behaves the way it does?

nazar-pc, 2018-08-12:
So sad, 15 months later still can't use some keys

haata, 2019-02-18:
I've done more work on this (sorry I haven't responded sooner, was handling a newborn then...).
See 3ba7436d86a81bca7dd69b467d78c7be6591fe2b.

nazar-pc, 2019-02-18:
Nice! Is Teensy 3.2 support back too? Last time I compiled it it gave me `ERROR - Macro Trigger Event Overflow! Serious Bug! 144` messages, I'd really like to try it again.

haata, 2019-02-18:
It should be working, though let me know if it isn't (I'll need to pull out one of my Teensy 3.2s).

nazar-pc, 2019-02-18:
I'll probably get to this on weekend and will definitely share my results here

nazar-pc, 2019-02-23:
So now I'm stuck with compilation errors and with pipenv setup it is not clear to me how to apply custom changes to KLL files to see how that can be fixed.

So just with https://github.com/kiibohd/controller/blob/bc0e707236d7f6dfc71aea2da349f641e1213790/Scan/Rapoo_E9070/scancode_map.kll it fails like this:
<details>

```
(Keyboards) root@f4d6a3e67108:/controller/Keyboards# ./Rapoo_E9070.bash 
linux-gnu/Linux is supported.
Unix Build
Selected Generator: Ninja
linux-gnu.Rapoo_E9070.gcc.ninja
VID: 0x1C11 (Boot: 0x1C11)
PID: 0xB04D (Boot: 0xB007)
-- Build CPU Detected:
x86_64
-- Build Kernel Detected:
Linux 5.0.0-rc7-acso
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256
-- Chip Family:
mk2x
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/kinetis.c;Lib/delay.c;Lib/entropy.c;Lib/gpio.c;Lib/periodic.c;Lib/storage.c;Lib/time.c;Lib/arm_cortex.c
-- Bootloader Type:
Teensy
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc 6.3.1
-- Detected Scan Module Source Files:
Scan/Devices/MatrixARMPeriodic/matrix_scan.c;Scan/Rapoo_E9070/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/kll.c;Macro/PartialMap/layer.c;Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/Interface/output_gen.c;Output/USB/output_com.c;Output/USB/output_usb.c;Output/USB/arm/usb_desc.c;Output/USB/arm/usb_dev.c;Output/USB/arm/usb_joystick.c;Output/USB/arm/usb_keyboard.c;Output/USB/arm/usb_mem.c;Output/USB/arm/usb_mouse.c;Output/USB/arm/usb_rawio.c;Output/USB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/latency/latency.c;Debug/led/led.c;Debug/print/print.c
-- Build OS Detected:
Ubuntu 18.04.1 LTS bionic
-- Checking for kll
-- kll Version Detected: kll 0.5.7.8
-- kll Installation Path: /root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll
-- kll Layouts Cache Path: /tmp/7-hid-io-layouts-c5410a1
-- Detected Layout Files:
/controller/Scan/Devices/MatrixARMPeriodic/capabilities.kll
/controller/Macro/PartialMap/capabilities.kll
/controller/Output/HID-IO/capabilities.kll
/controller/Output/USB/capabilities.kll
/controller/Debug/latency/capabilities.kll
/controller/Debug/led/capabilities.kll
/controller/Lib/arm_cortex.kll
/controller/Lib/kinetis.kll
/controller/Scan/Rapoo_E9070/scancode_map.kll
/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/layouts/stdFuncMap.kll
-- Configuring done
-- Generating done
-- Build files have been written to: /controller/Keyboards/linux-gnu.Rapoo_E9070.gcc.ninja
[1/36] Generating KLL Layout
FAILED: generatedKeymap.h kll_defs.h usb_hid.h generatedPixelmap.c kll.json 
cd /controller && python3 -m kll --kiibohd-debug --config /controller/Scan/Devices/MatrixARMPeriodic/capabilities.kll /controller/Macro/PartialMap/capabilities.kll /controller/Output/HID-IO/capabilities.kll /controller/Output/USB/capabilities.kll /controller/Debug/latency/capabilities.kll /controller/Debug/led/capabilities.kll /controller/Lib/arm_cortex.kll /controller/Lib/kinetis.kll --base /controller/Scan/Rapoo_E9070/scancode_map.kll --default /root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/layouts/stdFuncMap.kll --emitter kiibohd --def-output /controller/Keyboards/linux-gnu.Rapoo_E9070.gcc.ninja/kll_defs.h --map-output /controller/Keyboards/linux-gnu.Rapoo_E9070.gcc.ninja/generatedKeymap.h --hid-output /controller/Keyboards/linux-gnu.Rapoo_E9070.gcc.ninja/usb_hid.h --pixel-output /controller/Keyboards/linux-gnu.Rapoo_E9070.gcc.ninja/generatedPixelmap.c --json-output /controller/Keyboards/linux-gnu.Rapoo_E9070.gcc.ninja/kll.json
ERROR: 'CONTEXTMENU' (Layout(name=default)) is an invalid USB HID Code Lookup...
  File "/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
ERROR: OperationSpecificsStage:parse -> /tmp/kll/controllerScanRapoo_E9070@['scancode_map']_processed.kll:111
ERROR: Invalid stage status 'Incomplete' for 'OperationSpecificsStage'.
[... 69 lines cut ...]
ERROR: 'EMAILREADER' (Layout(name=default)) is an invalid USB HID Code Lookup...
ERROR: 'CONTEXTMENU' (Layout(name=default)) is an invalid USB HID Code Lookup...
  File "/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
ERROR: OperationSpecificsStage:parse -> /tmp/kll/controllerScanRapoo_E9070@['scancode_map']_processed.kll:111
  File "/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
ERROR: OperationSpecificsStage:parse -> /tmp/kll/root.localsharevirtualenvsKeyboards-41-guJFClibpython3.6site-packagesklllayouts@['Rapoo_E9070_Fn']_processed.kll:12
ERROR: 'CONSUMERCONTROLCONFIG' (Layout(name=default)) is an invalid USB HID Code Lookup...
  File "/root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
ERROR: OperationSpecificsStage:parse -> /tmp/kll/root.localsharevirtualenvsKeyboards-41-guJFClibpython3.6site-packagesklllayouts@['Rapoo_E9070_Fn']_processed.kll:13
ERROR: Invalid stage status 'Incomplete' for 'OperationSpecificsStage'.
ninja: build stopped: subcommand failed.
Error in build. Exiting...
```

</details>

So apparently things that worked when this issue was created are now broken.

I've merged current upstream master branch to #200 and running it under Ubuntu-based Docker environment as suggested in updated documentation.

nazar-pc, 2019-02-23:
You can try to compile https://github.com/nazar-pc/controller/tree/Rapoo_E9070-latest-upstream to reproduce my results

nazar-pc, 2019-03-09:
@haata should I expect any kind of solution any time soon?

haata, 2019-03-09:
I'll see if I can take a look this weekend.

Some of the USB code names have changed after I ported to using a separate repo for symbolic names: https://github.com/hid-io/layouts/blob/master/base/base.json

In your branch I'm not seeing the pipenv files. These are now required in order to use the KLL compiler now. Specifically   https://github.com/kiibohd/controller/blob/master/Keyboards/Pipfile

haata, 2019-03-09:
```
ERROR: 'EMAILREADER' (Layout(name=default)) is an invalid USB HID Code Lookup...
  File "/home/hyatt/.local/share/virtualenvs/Keyboards-YoFIrn1K/lib/python3.7/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
ERROR: OperationSpecificsStage:parse -> /tmp/kll/.local_share_virtualenvs_Keyboards-YoFIrn1K_lib_python3.7_site-packages_kll_layouts@['Rapoo_E9070_Fn']_processed.kll:12
ERROR: 'CONSUMERCONTROLCONFIG' (Layout(name=default)) is an invalid USB HID Code Lookup...
  File "/home/hyatt/.local/share/virtualenvs/Keyboards-YoFIrn1K/lib/python3.7/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
ERROR: OperationSpecificsStage:parse -> /tmp/kll/.local_share_virtualenvs_Keyboards-YoFIrn1K_lib_python3.7_site-packages_kll_layouts@['Rapoo_E9070_Fn']_processed.kll:13
ERROR: 'CONTEXTMENU' (Layout(name=default)) is an invalid USB HID Code Lookup...
  File "/home/hyatt/.local/share/virtualenvs/Keyboards-YoFIrn1K/lib/python3.7/site-packages/kll/common/parse.py", line 115, in hidCode
    hid_code = int(lookup[match_name], 0)
Invalid dictionary lookup, check syntax.
```

Looks like you just need to add some spaces. In prior versions of the KLL compiler, spaces were ignored. I removed this behaviour when I switched to JSON storage (otherwise it would have been confusing when parsing the JSON).

haata, 2019-03-09:
controller diff
```diff
diff --git a/Scan/Rapoo_E9070/scan_loop.c b/Scan/Rapoo_E9070/scan_loop.c
index 355c280..c2e03e9 100644
--- a/Scan/Rapoo_E9070/scan_loop.c
+++ b/Scan/Rapoo_E9070/scan_loop.c
@@ -1 +1 @@
-#include "../MD1/scan_loop.c"
+#include "../Infinity_60/scan_loop.c"
diff --git a/Scan/Rapoo_E9070/scan_loop.h b/Scan/Rapoo_E9070/scan_loop.h
index 2c5810b..24ad591 100644
--- a/Scan/Rapoo_E9070/scan_loop.h
+++ b/Scan/Rapoo_E9070/scan_loop.h
@@ -1 +1 @@
-#include "../MD1/scan_loop.h"
+#include "../Infinity_60/scan_loop.h"
diff --git a/Scan/Rapoo_E9070/scancode_map.kll b/Scan/Rapoo_E9070/scancode_map.kll
index d67d89c..f619028 100644
--- a/Scan/Rapoo_E9070/scancode_map.kll
+++ b/Scan/Rapoo_E9070/scancode_map.kll
@@ -108,7 +108,7 @@ S0x23 : U"LGui";
 S0x42 : U"LAlt";
 S0x44 : U"Space";
 S0x68 : U"RAlt";
-S0x66 : SYS"ContextMenu";
+S0x66 : SYS"Context Menu";
 S0x4C : U"RCtrl";
 S0x6D : U"Left";
 S0x47 : U"Up";
```

kll diff
```diff
diff --git a/kll/layouts/Rapoo_E9070_Fn.kll b/kll/layouts/Rapoo_E9070_Fn.kll
index 9a74daf..f850b8b 100644
--- a/kll/layouts/Rapoo_E9070_Fn.kll
+++ b/kll/layouts/Rapoo_E9070_Fn.kll
@@ -9,8 +9,8 @@ Date = 2017-05-11;
 U"F1"  : CONS"Back";
 U"F2"  : CONS"Forward";
 U"F3"  : CONS"Home";
-U"F4"  : CONS"EmailReader";
-U"F5"  : CONS"ConsumerControlConfig";
+U"F4"  : CONS"Email Reader";
+U"F5"  : CONS"Consumer Control Config";
 U"F6"  : CONS"PausePlay";
 U"F7"  : CONS"Stop";
 U"F8"  : CONS"ScanPreviousTrack";
```

You'll also need to fix your scancode_map.kll, I've removed matrix.h and replaced it with defines inside of scancode_map.kll. For example: https://github.com/kiibohd/controller/blob/master/Scan/Infinity_60/scancode_map.kll

nazar-pc, 2019-03-09:
OK, So I've managed to compile it again and again got tons of `ERROR - Macro Trigger Event Overflow! Serious Bug! 144` with completely otherwise broken keyboard.

Also it is quite frustrating that config generated for KLL 0.3d suddenly stops compiling, especially since even in 0.5's documentation can codes list doens't have spaces for `EmailReader` and `ConsumerControlConfig`.

Looks like versioning is useless here. Compiler should either compile the same KLL correctly or complain that KLL version is unsupported with links to upgrade guide of some sort.

It has been almost 2 years and I still can't get basic feature like sending a correct scan code to work.

nazar-pc, 2019-04-11:
Should I even expect this to work again or not? I mean, seriously, basic support for Teensy 3.2 is completely broken for about a year now, this issue is open for over 2 years.
I'm not commenting too often, but I'm still patiently waiting for this to be fixed and if it it not going to happen I'd like to be aware of that.

haata, 2019-04-11:
Sorry about that. I've merged the referenced patch.
I took a quick look at your repos and they seem a bit out of date right now.

I think I may have fixed the compilation issue (and maybe the non-sequential scancode problem too) recently. So you may have to rebase.

I tried just copying Scan/Rapoo_E9070 to my master (as well as Rapoo_E9070.bash to Keyboards).
It compiled (though there is one error so I had to remove the KLL = line from Scan/Rapoo_E9070/scancode_map.kll).

I need to locate my Teensy 3.2. I'm also available on our discord channel (and irc though not as responsive there). I think it's really close (if there are any remaining issues) and the firmware is probably more stable now than it's ever been.

nazar-pc, 2019-04-13:
Rebased https://github.com/nazar-pc/controller/tree/Rapoo_E9070-latest-upstream against master, compiles perfectly now.
However, still the same issue, it prints `ERROR - Macro Trigger Event Overflow! Serious Bug! 144` infinitely and I do not know why. On any key press screen session closes, apparently controller reboots or something.

I suspect, it can be somehow related to https://github.com/kiibohd/controller/blob/bc0e707236d7f6dfc71aea2da349f641e1213790/Scan/Rapoo_E9070/matrix.h#L42-L47

and apparently https://github.com/kiibohd/controller/blob/fcc79a0696ef5be2fd2e139fbe3f0ba79e91fea4/Scan/Rapoo_E9070/scancode_map.kll#L78-L79
is not the correct way of doing it now or I'm missing something else.

haata, 2019-04-15:
`GPIO_Config_Pullup` was a big hint. It seems I haven't tested that mode in a very very very long time. The comparison was inverted, so the error was showing up because it was like you were holding down every single key all the time and the firmware was getting a bit overwhelmed (still functioning, just printing the error message).

I've pushed a change, should be working now (managed to find my Teensy 3.2).
Sorry this took so long.

I'll leave this open for now (just in case I've missed some things).

nazar-pc, 2019-04-17:
Merged upstream master into https://github.com/nazar-pc/controller/tree/Rapoo_E9070-latest-upstream again, but getting the same exact result still.
Can you, please, try compiling https://github.com/nazar-pc/controller/blob/Rapoo_E9070-latest-upstream/Keyboards/Rapoo_E9070.bash and running on Teensy 3.2?

haata, 2019-04-18:
You need to add this option.

```diff
diff --git a/Scan/Rapoo_E9070/scancode_map.kll b/Scan/Rapoo_E9070/scancode_map.kll
index 1800bf3..9f5e871 100644
--- a/Scan/Rapoo_E9070/scancode_map.kll
+++ b/Scan/Rapoo_E9070/scancode_map.kll
@@ -78,6 +78,10 @@ ScanCodeSenseList = "
 ScanCodeMatrixType => ScanCodeMatrixType_define;
 ScanCodeMatrixType = "GPIO_Config_Pullup";

+# Invert Matrix
+# Needed to account for pull-up resistor values
+ScanCodeMatrixInvert = 1;
+
 # Each group is one row of keys on keyboard starting from the top
 # Comments are added where layout differs on ANSI and ISO layouts
 S0x3A : U"Esc";
```

nazar-pc, 2019-04-18:
Tried that, but have different issues now:
1) When connected via USB hub `dmesg` on Linux prints:
```
[  762.555809] usb 1-5.1.3: Not enough bandwidth for new device state.
[  762.555836] usb 1-5.1.3: can't set config #1, error -28
```
And keyboard doesn't work. Connecting directly without USB hub works, but...
2) Keyboards prints completely wrong characters, looks like something in firmware is upside down, like columns and rows switched places somehow; also not 100% consistent, pressing one key prints something between 8 and 11 characters

haata, 2019-04-18:
Then you should have specified "GPIO_Config_Pulldown" and ScanCodeMatrixInvert = 0.

nazar-pc, 2019-04-18:
Tried that any all other possible combinations of `ScanCodeMatrixType` and `ScanCodeMatrixInvert`.
It either doesn't print anything or prints wrong characters or doesn't recognize key presses on some keys at all. Also I have doubts that pull-up on one firmware version can turn into pull-down on the other.

I'm inclined to believe that `GPIO_Config_Pullup` or its combination with something else is still broken.

haata, 2019-04-18:
GPIO_Config_Pullup is not used in any of my keyboards. So I don't really have a good way of testing it (everything uses GPIO_Config_Pulldown by default -> https://github.com/kiibohd/controller/blob/master/Scan/Devices/MatrixARMPeriodic/capabilities.kll#L64)

There may be timing issues than need to be figured out when using Pullup which would require a full keyboard and an oscilloscope to test.

This is where the code is: https://github.com/kiibohd/controller/blob/master/Scan/Devices/MatrixARMPeriodic/matrix_scan.c#L312-L324 that does the scanning.

You may have another problem, when switching between Pullup and Pulldown you may also need to swap rows and columns depending on the direction of your diodes.

nazar-pc, 2019-04-18:
My keyboard doesn't have diodes, which is why I had `#define GHOSTING_MATRIX` previously. I've customized non-repairable off-the-shelf keyboard, this one: https://www.amazon.com/Rapoo-E9070-Wireless-Multimedia-Keyboard/dp/B00WJYU1HM

Is it possible that current code simply doesn't support keyboards without diodes?

haata, 2019-04-18:
Ahhh, yes that would be the problem.

Yeah, I removed the ghosting code a while back to simplify the rewrite. I haven't had a chance to add it back in as it's significantly more complicated than scanning with diodes.

nazar-pc, 2019-04-21:
So this effectively boils down to not being possible for me to use newer firmware on current hardware 😞 
I can't verify that initial issue from this thread was fixed either, but I'm inclined to believe that.
Closing issue now, let me know if it even makes sense to create a ticker about bringing back ghosting code or just forget about it entirely.

closed 2019-04-21 by nazar-pc