kiibohd/controller #199: Upstreaming udev rules

nazar-pc, 2017-05-09:
Have you attempted or wanted to do this?
I could possible work on this if needed.

haata, 2017-05-12:
Sure! I bundle the rules for convenience. Let me check if I haven't done any more rule updates recently.

haata, 2017-05-12:
These are the useful rules. There are a few more, but those are Teensy specific (I'd hope those are already merged).

```udev
# Kiibohd Serial Interface
SUBSYSTEM=="tty" ATTRS{idVendor}=="1c11" ATTRS{idProduct}=="b04d" SYMLINK+="kiibohd" MODE:="0666"

# Kiibohd Device
SUBSYSTEM=="usb" ATTR{idVendor}=="1c11" ATTR{idProduct}=="b04d" MODE:="0666"

# DFU Bootloader (MCHCK)
SUBSYSTEM=="usb" ATTRS{idVendor}=="2323" ATTRS{idProduct}=="0001" ENV{ID_MM_DEVICE_IGNORE}="1"
SUBSYSTEM=="usb" ATTRS{idVendor}=="2323" ATTRS{idProduct}=="0001" ENV{MTP_NO_PROBE}="1"
SUBSYSTEM=="usb" ATTR{idVendor}=="2323" ATTR{idProduct}=="0001" MODE:="0666"

# Kiibohd DFU Bootloader
SUBSYSTEM=="usb" ATTRS{idVendor}=="1C11" ATTRS{idProduct}=="b007" ENV{ID_MM_DEVICE_IGNORE}="1"
SUBSYSTEM=="usb" ATTRS{idVendor}=="1C11" ATTRS{idProduct}=="b007" ENV{MTP_NO_PROBE}="1"
SUBSYSTEM=="usb" ATTR{idVendor}=="1c11" ATTR{idProduct}=="b007" MODE:="0666"

# Kiibohd Force Gauge
# e.g. udevadm info -a -p /sys/bus/usb/devices/3-5
# sudo udevadm test /sys/bus/usb/devices/3-5:1.0/tty/ttyACM0
# Use ATTR on root, ATTRS on leaves
SUBSYSTEM=="tty"  ATTRS{idVendor}=="1c11" ATTRS{idProduct}=="f05c" MODE:="0666" SYMLINK+="force"
SUBSYSTEM=="usb" ATTR{idVendor}=="1c11" ATTR{idProduct}=="f05c" MODE:="0666"
```

nazar-pc, 2017-07-01:
Some folks on systemd's IRC said that since it only affect very few people (keyboard itself works fine without them), it doesn't make sense to upstream these rules.

closed 2017-07-01 by nazar-pc