kiibohd/controller #204: Recent changes have broken Infinity Ergodox support.

kll, 2017-05-31:
From a fresh clone of the repository master branch (commit 51b6b431199e1fd9bbadfd997cbe258923aa0e15 as of the time I'm writing this), building with ergodox.bash and flashing the resulting firmware causes 2 weird issues. I'm getting the wrong output offset by one key and its also outputting the corresponding key from both sides of the keyboard. For example if I press the escape key I get an output of 56, if I press the 5 key I get 47, etc.  I connected to the debug console and turned on matrixDebug and it is reporting the wrong scan codes as well.  For the escape key which should be scan code 0x2 it shows 0x3 and for the 5 key I get 0x4 instead of 0x3. Moving over to the last column I get no output at all and no scan codes reported. The behavior is the same on all rows and both halves of the keyboard.

I've tried to track down when the change occurred by building and flashing each recent commit and checking for when the behavior changed. It was a bit tricky because the kll submodule also has to be manipulated to get a successful build. Here is what I came up with. The last commit where things work as expected is c7fb6d1c3be29de1bcb0b9c7201448e66fb2ca33. Starting on commit 0cbb451642152fc2cb867e2c72153b9a4c0e522c the scan code reported is wrong, but I don't get the output from both sides of the keymap. In this case pressing escape outputs only 5 instead of 56. The commit for the kll submodule that let me compile both of those commits was 2062be08e34430d523b20947e9eb5a3ec3948331.

Hopefully that information is helpful. I did a diff between those two commits to try to narrow it down further but there was a lot more changes than I was expecting so I'm now out of my depth I believe. I'll do whatever is needed to help track down and fix the issue.  I've been really looking forward to the new features in KLL 0.5!

haata, 2017-05-31:
Yeah, Ergodox is currently broken. I'm working on fixes so it shouldn't be too much longer (interconnect is tricky).

For now I'd recommend using the 'old' branch if you're on an Infinity Ergodox

kll, 2017-05-31:
Ok, I was getting ahead of myself then. :( Unfortunately for me that means back to a fairly buggy TMK port. I'm lost without "tap keys"!  So I've been eyeing your KLL 0.5 progress ever since I got this Infinity Ergodox and realized the official firmware didn't support them.

If there's anything at all I might be able to do to help let me know. I'm not very strong in C but I do have some experience.

jceaser, 2017-06-03:
I guess this is why I can't compile when I have the following in my KII file:

`U"LBRACE" : U"NEXT LAYER";`

I found  `U"Next Layer" :: layerRotate( 0 ); # 0 is Next`

and switched to: `U"LBRACE" : layerRotate( 0 );`

Got past at least one error. Now complaining about undefined reference to `Pixel_AnimationSettings'

jceaser, 2017-06-03:
also, no mouseOut

haata, 2017-06-03:
Which keyboard are you compiling for? If it's an Ergodox or Whitefox you'll
need to select the PixelMap Macro module instead of the PartialMap module
during compilation if you haven't done so yet.

I'll have to look into mouseOut and layerRotate.

jceaser, 2017-06-04:
yea, I'm one PixelMap. I'm compiling for Ergodox.

# Keyboard Module Configuration
ScanModule="Infinity_Ergodox"
MacroModule="PixelMap"
OutputModule="pjrcUSB"
DebugModule="full"

jceaser, 2017-06-04:
what do I use for 
Default Map? Im starting to just throw everything at it: Originally I used just lcdFuncMap

DefaultMap="mine/MDErgo1-base stdFuncMap lcdFuncMap animation_test iced_func k-type/unset_v1 k-type/rainbow_wipe k-type/color_painter"

cogliostro, 2017-07-06:
The old branch doesn't work either it seems. I get a KeyError: 'AnimationList' when compiling the ergodox kll's.

curiously-b2, 2017-07-07:
> May 31 ... shouldn't be too much longer

Got any ETA on a fix?

danspiff, 2017-07-17:
I just wanted to post this for anyone that wanted to build the last working revisions above.  I struggled to figure this out and thought I would share:

1.  git clone https://github.com/kiibohd/controller
2.  cd controller/
3.  git checkout c7fb6d1c3be29de1bcb0b9c7201448e66fb2ca33
4.  cd Keyboards/
5.  ./ergodox.bash
Expect build to fail.
6.  cd ../kll
7.  git checkout 2062be08e34430d523b20947e9eb5a3ec3948331
8.  cd ../Keyboards
9.  ./ergodox.bash
10.  Customize from here and rebuild.

daantjie, 2017-07-18:
@danspiff; I fail on (9):

~~~
[ 96%] Linking C executable kiibohd.elf
CMakeFiles/kiibohd.elf.dir/Scan/ISSILed/led_scan.c.obj: In function `I2C_Send':
/home/daniel/hotfix/controller/Scan/ISSILed/led_scan.c:591: undefined reference to `I2C_BufferCopy'
CMakeFiles/kiibohd.elf.dir/Scan/ISSILed/led_scan.c.obj: In function `LED_setup':
/home/daniel/hotfix/controller/Scan/ISSILed/led_scan.c:392: undefined reference to `I2C_setup'
CMakeFiles/kiibohd.elf.dir/Scan/STLcd/lcd_scan.c.obj: In function `LCD_clear':
/home/daniel/hotfix/controller/Scan/STLcd/lcd_scan.c:241: undefined reference to `LCD_clearPage'
CMakeFiles/kiibohd.elf.dir/Scan/STLcd/lcd_scan.c.obj: In function `LCD_setup':
/home/daniel/hotfix/controller/Scan/STLcd/lcd_scan.c:298: undefined reference to `SPI_setup'
CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/trigger.c.obj: In function `Macro_evalTriggerMacro':
/home/daniel/hotfix/controller/Macro/PartialMap/trigger.c:293: undefined reference to `Macro_evalLongTriggerMacroVote'
/home/daniel/hotfix/controller/Macro/PartialMap/trigger.c:294: undefined reference to `Macro_evalShortTriggerMacroVote'
/home/daniel/hotfix/controller/Macro/PartialMap/trigger.c:350: undefined reference to `Macro_isLongResultMacro'
CMakeFiles/kiibohd.elf.dir/Macro/PartialMap/trigger.c.obj: In function `Trigger_process':
/home/daniel/hotfix/controller/Macro/PartialMap/trigger.c:474: undefined reference to `Macro_updateTriggerMacroPendingList'
CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.obj: In function `CLI_init':
/home/daniel/hotfix/controller/Debug/cli/cli.c:79: undefined reference to `prompt'
CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.obj: In function `CLI_tabCompletion':
/home/daniel/hotfix/controller/Debug/cli/cli.c:393: undefined reference to `prompt'
CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.obj: In function `CLI_retreiveHistory':
/home/daniel/hotfix/controller/Debug/cli/cli.c:446: undefined reference to `prompt'
CMakeFiles/kiibohd.elf.dir/Debug/cli/cli.c.obj: In function `CLI_process':
/home/daniel/hotfix/controller/Debug/cli/cli.c:188: undefined reference to `prompt'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/kiibohd.elf.dir/build.make:688: kiibohd.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/kiibohd.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Error in make. Exiting...
~~~

daantjie, 2017-07-18:
Actually, pretty much the same thing happens on the `old` branch. I don't know if anyone has a fix.

crystalhand, 2017-07-19:
This is happening on the older ones because it is downloading the latest kll files when you are running make.  You need to download the older ones from 2062be0 and manually put them in the kiibohd folder.

jceaser, 2017-07-21:
am I reading things right: `reset --hard 2062be0` should get me to the last ergo dox build that works?

wailashi, 2017-07-28:
Following the steps described by danspiff I get the same error as daantjie. Any advice?

daantjie, 2017-07-28:
@wailashi; have a look at **crystalhand**'s  response above. (I haven't tried it yet.)

wailashi, 2017-07-29:
@daantjie, I'm having a hard time understanding @crystalhand's advice. Which files I am supposed to place where? There is no `kiibohd` folder that I can find.

I tried modified the makefiles to get  2062be0 instead of the latest version, but I get the same errors.

crystalhand, 2017-07-29:
I just double checked to make sure it works and it does.  This is what you need to do: 
**Note**: This assumes you are running in either linux (preferably) or win 10 using bash.  Dont try this in win 7 as it is much more trouble then grabbing  vmplayer and ubuntu. 

1. Dowload this repository as a zip and extract it
https://github.com/kiibohd/controller/tree/c7fb6d1c3be29de1bcb0b9c7201448e66fb2ca33
Call this folder KiibohdControler (name does not matter all that much)

2. download this repository and extract it as a subfolder of /KiibohdControler
https://github.com/kiibohd/controller/tree/2062be08e34430d523b20947e9eb5a3ec3948331

 3. Rename the folder extracted in 2 to "kll" (The name of this folder is critical)

4.  navigate to the "keyboards" folder ie /KiibohdControler/keyboards   
 
5. open  terminal/bash within this folder and enter "./ergodox.bash"  this will build the bin file for you.  They will be located within two folders that are created.  One for the left side and the other for the right.  You only have to flash the master side.

wailashi, 2017-07-29:
@crystalhand following your steps exactly it still fails for me. :/ Same linker errors as daantjie had earlier. I'm doing this on arch linux.
Thanks for the help anyway!

daantjie, 2017-07-29:
@wailashi I'm also on Arch. (*Coincidence?*)

crystalhand, 2017-07-29:
I did my test this morning in windows but most of the time I am using a Ubuntu VM.  I would give that a try.  The error referenced is a link error so my only thought is to double check that you are downloading the right version.

wailashi, 2017-07-31:
@daantjie Not a coincidence. 
I found a fix and I'm now able to compile the controller. This seems to be a portability problem with `inline` (see [relevant stack overflow question](https://stackoverflow.com/q/18635665/485784)). Changing to `static inline` fixes the problem for me.

[This](https://gist.github.com/wailashi/3f4322985c2f752bc8e76e1bfb4a127b) patch file contains the fixes that worked for me.

andru, 2017-08-03:
The steps @crystalhand outlined gets things building for me again, but if I map any of the functions defined in `stdFuncMap.kll` or `lcdFuncMap.kll` to a key then pressing that key does a big nothing, so unfortunately I've lost all my layers.

nayr1991, 2017-09-08:
What are the chances that this is affecting a teensy3.2 custom wired setup? 
After updating the controller repo, my previously working matrix seems to be offset by 1 and a couple of the lower keys are reported as out of bounds by matrixDebug. Is crystalhand's workaround the only way to get around this currently?

crystalhand, 2017-09-08:
The chances of this affecting a teensy ergodox are pretty close to 100%.  the issue is not hardware based but software.  Because its the same code across all of the ditterent chips for ergodox layout it is going to happen.  I would suggest taking a look at QMK.  while the kiibohd firmware is much more elegant the dev support seems to be much less.  I assume their current efforts are focused on the k type keyboard that was on massdrop recently.  We will probably get more attention once the dust from that settles.

kll, 2017-09-08:
I agree with @crystalhand . I gave up and switched to QMK long ago.

jceaser, 2017-09-08:
How is the QMK support compared to this software as far as stability goes?

I do have a working setup following the steps outlined here, but I'm desperately waiting for the k-type code to be merged and successfully integrated with other boards. Frankly the people here should have not merged the k-type branch into master until everything was working. Don't get me wrong, I love the concept of what is trying to be done with kii but breaking ergodox is not cool.

nayr1991, 2017-09-08:
@crystalhand I thought qmk didn’t have very good support for a teensy 3.2? Forgot to mention mine isn’t an ergodox board, just a hand wired Planck.

kll, 2017-09-08:
I'm not sure not sure about Teensy 3.2 specifically, but QMK has been great on my original Ergodox as well as my Infinity Ergodox. Its nice having the exact same code-base and keymap work on both of mine.

crystalhand, 2017-09-08:
oh if it is a planck I dont imagine you would see the same issue but it is possible. From what I understand the problem discussed here is due to the split nature of the keyboard and the communication between the two halves.  However I am running kiibohd only on my infinite ergodox.

QMK is very stable from my experience.  I am running it on a numpad and a lets split.  It is more cumbersome to program but  has better documentation and support.  I rather enjoy playing with it when work is a bit slow.  I would at least check it out

LaikeSF, 2017-09-15:
I seem to be plagued with similar issues. Compiling my layout with the instructions provided by crystalhand worked fine, but the most recent build spat out the following error 

```
**linux-gnu/Linux is supported.
Selected Generator: Unix Makefiles
linux-gnu.ICED-L.gcc.make
-- Build CPU Detected:
x86_64
-- Build OS Detected:
Linux 4.4.0-43-Microsoft
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk20dx
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/mk20dx.c;Lib/delay.c;Lib/entropy.c;Lib/time.c
-- Bootloader Type:
dfu
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc
-- Detected Scan Module Source Files:
Scan/Devices/ISSILed/i2c.c;Scan/Devices/ISSILed/led_scan.c;Scan/Devices/MatrixARM/matrix_scan.c;Scan/Devices/STLcd/lcd_scan.c;Scan/Devices/UARTConnect/connect_scan.c;Scan/Infinity_Ergodox/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c;Macro/PixelMap/pixel.c;/mnt/x/ErgoDox/controller/Keyboards/linux-gnu.ICED-L.gcc.make/generatedPixelmap.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/pjrcUSB/output_com.c;Output/pjrcUSB/arm/usb_desc.c;Output/pjrcUSB/arm/usb_dev.c;Output/pjrcUSB/arm/usb_joystick.c;Output/pjrcUSB/arm/usb_keyboard.c;Output/pjrcUSB/arm/usb_mem.c;Output/pjrcUSB/arm/usb_mouse.c;Output/pjrcUSB/arm/usb_rawio.c;Output/pjrcUSB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/led/led.c;Debug/print/print.c
-- Downloading latest kll version:
fatal: destination path 'kll' already exists and is not an empty directory.
-- Detected Layout Files:
/mnt/x/ErgoDox/controller/Scan/Devices/ISSILed/capabilities.kll
/mnt/x/ErgoDox/controller/Scan/Devices/MatrixARM/capabilities.kll
/mnt/x/ErgoDox/controller/Scan/Devices/STLcd/capabilities.kll
/mnt/x/ErgoDox/controller/Scan/Devices/UARTConnect/capabilities.kll
/mnt/x/ErgoDox/controller/Macro/PartialMap/capabilities.kll
/mnt/x/ErgoDox/controller/Macro/PixelMap/capabilities.kll
/mnt/x/ErgoDox/controller/Output/HID-IO/capabilities.kll
/mnt/x/ErgoDox/controller/Output/pjrcUSB/capabilities.kll
/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/scancode_map.kll
/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/leftHand.kll
/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/slave1.kll
/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/rightHand.kll
/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-0.kll
/mnt/x/ErgoDox/controller/kll/layouts/lcdFuncMap.kll
/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-0.kll
/mnt/x/ErgoDox/controller/kll/layouts/lcdFuncMap.kll
/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-2.kll
/mnt/x/ErgoDox/controller/kll/layouts/lcdFuncMap.kll
/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-3.kll
/mnt/x/ErgoDox/controller/kll/layouts/lcdFuncMap.kll
/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-4.kll
/mnt/x/ErgoDox/controller/kll/layouts/lcdFuncMap.kll
/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-5.kll
/mnt/x/ErgoDox/controller/kll/layouts/lcdFuncMap.kll
-- Could NOT find DFU_SUFFIX (missing:  DFU_SUFFIX_EXECUTABLE)
CMake Warning at Lib/CMake/build.cmake:84 (message):
  DFU Binary has not been signed, requires dfu-suffix...
Call Stack (most recent call first):
  CMakeLists.txt:171 (include)
 
 
-- Configuring done
-- Generating done
[... 69 lines cut ...]
./generatedKeymap.h:383:68: error: (near initialization for 'rm100_guide[2]')
make[2]: *** [CMakeFiles/kiibohd.dir/Macro/PartialMap/macro.c.o] Error 1
make[1]: *** [CMakeFiles/kiibohd.dir/all] Error 2
make: *** [all] Error 2
Error in make. Exiting...`**
```

I used the following config for the bash files
```
DefaultMap="mine/MDErgo1-Default-0 lcdFuncMap"
 
# This is where you set the additional layers
# NOTE: Indexing starts at 1
# NOTE: Each new layer is another array entry
# e.g.  PartialMaps[1]="layer1 layer1mod"
#       PartialMaps[2]="layer2"
#       PartialMaps[3]="layer3"
PartialMaps[1]="mine/MDErgo1-Default-0"
PartialMaps[2]="mine/MDErgo1-Default-2"
PartialMaps[3]="mine/MDErgo1-Default-3"
PartialMaps[4]="mine/MDErgo1-Default-4"
PartialMaps[5]="mine/MDErgo1-Default-5"
```

Originally, it was thought to be a problem with my layout, but it seems like all my layouts are using U"LCK7" and similar keymaps, so that does not appear to be the issue.

crystalhand, 2017-09-16:
Try U"LCK2" instead of U"KEY_LCK1".  If I remember correctly the aliasing is weird (That may be completely wrong however)

Even if you do get a successful compile the keymap is not going to be accurate.  With the most recent builds all of the keys are shifted making it unusable.  The only way to resolve this is to start tracing where/why this is occurring.  Because there is a working solution and I have to many other projects I lack the desire to attempt this. 

IIRC I made some progress several months ago by modifying the left-hand.kll and the right hand.kll files.  I would suggest starting there.

LaikeSF, 2017-09-16:
Unfortunately, these errors are coming up when the layout is using U"LCK2" instead of U"KEY_LCK1". HaaTa thought it was the same problem, but I can't find any mention of KEY_LCK1 in my layout. The same layout were able to compile just fine with your workaround, so I guess the issue may not be the layout.

crystalhand, 2017-09-16:
Try spelling it "LOCK2" I know I was able to get it to compile successfully somehow.  Make sure there are no other aliases in the documentation/examples.  

Try only compiling a base layer with no references to any other layers.  Or better yet compile it with only one or two keys mapped.

I can double check my code on Monday when I go to work.  I did most of my tinkering with kiibohd while at work.  The only version I have here is the older release that compiles correctly

jceaser, 2017-09-19:
This issue of seams to be on going, where is this issue of ErgoDox support being discussed? People who have bought ergodox keyboards off massdrop may not understand why some keyboards work while others do not. Those in charge, could you not make a branch for ergodox instead of asking people to make multiple checkouts? (I follow the steps pretty easy, but I deal with GIT all day, others may not)

metsie2912, 2017-09-26:
Thanks danspiff and wailashi, I'm an Arch Linux user and with both your guidelines, I'm now able to compile my own layout. 
@the developers, Is it correct I'm currently not able to make use of keypress states? On my Ergodox EX I use that a lot and I'd love to use that on my Infinity Ergodox too. And do you have any idea whan these issues will be fixed? I'd love to use my Infinity Ergodox to it's full potential!

haata, 2017-09-26:
I'm actively working to get this working (#1 priority). Hopefully sometime this week.

haata, 2017-10-02:
While I'm not quite ready to say Infinity Ergodox is ready. I have fixed the INVALID HID(USBCode) issue.

Was entirely a kll compiler issue: https://github.com/kiibohd/kll/commit/8fe216c06de598b79292708708c046d58a609037

I'm now working on fixing the double-keypress issue on Infinity Ergodox (again, another bug in the kll compiler, the firmware should be fine). I've already identified where to fix it, so I should have a fix in the next couple days.

dfaught, 2017-10-02:
I can confirm the INVALID HID(USBCode) issue is gone.  I can build and flash with customized layouts.  After flashing however, keys are incorrect.  All keys appear to be shifted right by one...  a=s, s=d, d=f, etc.

haata, 2017-10-02:
Haha, ok. I think I know what that problem is. Should have a fix this evening.

Thanks for testing.

@dfaught Confirming, you have an Infinity Ergodox?

dfaught, 2017-10-02:
@haata, yes, I do have an Infinity Ergodox.  Latest Massdrop.  I'm happy to test.

haata, 2017-10-03:
Pushed a few more fixes. Ready for testing :D

dfaught, 2017-10-03:
@haata, building straight after a pull works with the default layout files.  Flashing the "stock" layout to my board works and proper characters are generated.

If I attempt to build with my layouts I get the following.

``` -- Configuring done
-- Generating done
-- Build files have been written to: /home/derek/home-ext/workspaces/git.local/controller/Keyboards/linux-gnu.ICED-R.gcc.make
[  3%] Generating KLL Layout
kll BETA 0.5c.668027c4050c63c051a126bf3ff30002b2a4bf28 - 2017-10-02 23:18:13 -0700
Traceback (most recent call last):
  File "../../kll/kll", line 159, in <module>
    control.process()
  File "/home/derek/home-ext/workspaces/git.local/controller/kll/common/stage.py", line 123, in process
    stage.process()
  File "/home/derek/home-ext/workspaces/git.local/controller/kll/common/stage.py", line 2753, in process
    self.analyze()
  File "/home/derek/home-ext/workspaces/git.local/controller/kll/common/stage.py", line 2725, in analyze
    self.generate_trigger_lists()
  File "/home/derek/home-ext/workspaces/git.local/controller/kll/common/stage.py", line 2461, in generate_trigger_lists
    if self.trigger_lists[index][identifier.get_uid()] is None:
IndexError: list index out of range
make[2]: *** [CMakeFiles/kiibohd.dir/build.make:84: generatedKeymap.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:301: CMakeFiles/kiibohd.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Error in make. Exiting...
```

kdayns, 2017-10-03:
don't know if related, but compilation of the rigth side fails with last head c25f15c, but the problem manifests only when I add something more than default layer and happens only to the rigth side build, I have kll files generated with IC web configurator.

the interesing thing is that the scancode alters between script run from: S0x029 to: S0x056

-- Build files have been written to: /home/mp/xxx/controller/Keyboards/linux-gnu.ICED-R.gcc.make                     
[  3%] Generating KLL Layout                                                                                         
kll BETA 0.5c.668027c4050c63c051a126bf3ff30002b2a4bf28 - 2017-10-02 23:18:13 -0700                                   
Traceback (most recent call last):                                                                                   
  File "../../kll/kll", line 159, in <module>
    control.process()
  File "/home/mp/xxx/controller/kll/common/stage.py", line 123, in process
    stage.process()
  File "/home/mp/xxx/controller/kll/common/stage.py", line 2753, in process
    self.analyze()
  File "/home/mp/xxx/controller/kll/common/stage.py", line 2725, in analyze
    self.generate_trigger_lists()
  File "/home/mp/xxx/controller/kll/common/stage.py", line 2458, in generate_trigger_lists
    trigger_index = self.trigger_index_lookup[sub_expr.kllify()]
KeyError: 'S0x056 : layerShift(1) + LCDLayerDisplay();'
make[2]: *** [CMakeFiles/kiibohd.dir/build.make:78: generatedKeymap.h] Error 1                                       
make[1]: *** [CMakeFiles/Makefile2:301: CMakeFiles/kiibohd.dir/all] Error 2                                          
make: *** [Makefile:84: all] Error 2                                                                                 
Error in make. Exiting...

 the old branch seems to have not locked to kll it needs, latest does not work, maybe specifying particular commit would help others test with older versions.
[  3%] Generating KLL Layout                                                                                         
make[2]: ../../kll/kll.py: Command not found                                                                         
make[2]: *** [CMakeFiles/kiibohd.elf.dir/build.make:75: generatedKeymap.h] Error 127                                 
make[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/kiibohd.elf.dir/all] Error 2                                      
make: *** [Makefile:84: all] Error 2                                                                                 
Error in make. Exiting...                                                                           
    
                                                                     
diff --git a/Keyboards/ergodox.bash b/Keyboards/ergodox.bash                                  
index 05923ef..93c18c1 100755
--- a/Keyboards/ergodox.bash
+++ b/Keyboards/ergodox.bash
@@ -25,7 +25,8 @@ BaseMap="scancode_map leftHand slave1 rightHand"
 # This is the default layer of the keyboard
 # NOTE: To combine kll files into a single layout, separate them by spaces
 # e.g.  DefaultMap="mylayout mylayoutmod"
-DefaultMap="mdergo1Overlay lcdFuncMap"
+#DefaultMap="mdergo1Overlay lcdFuncMap"
+DefaultMap="MDErgo1-Default-0 lcdFuncMap"
 
 # This is where you set the additional layers
 # NOTE: Indexing starts at 1
@@ -33,8 +34,8 @@ DefaultMap="mdergo1Overlay lcdFuncMap"
 # e.g.  PartialMaps[1]="layer1 layer1mod"
 #       PartialMaps[2]="layer2"
 #       PartialMaps[3]="layer3"
-PartialMaps[1]="iced_func"
-PartialMaps[2]="iced_numpad"
+PartialMaps[1]="MDErgo1-Default-1"
+PartialMaps[2]="MDErgo1-Default-2"

btw, I noticed that for the rigth side there is   BaseMap="defaultMap... but for the left, it is scancode_map

need my kll files?

haata, 2017-10-03:
@dfaught and @kdayns please provide your KLL files.

@dfaught please provide the order of the KLL files.

kdayns, 2017-10-03:
[http://public.pwp.lv/MDErgo1-Default-48da951272c84ef1132d300a803fca21.zip](url)

dfaught, 2017-10-03:
[ic-generated-kll.tar.gz](https://github.com/kiibohd/controller/files/1353512/ic-generated-kll.tar.gz)

einalex, 2017-10-03:
I don't get any errors compiling my KLL files and keys are mostly correct. It's just the innermost row on both halves that is erroneous. The left half keys produce outputs for both sides (first left, then right), right side keys produce no output.

haata, 2017-10-03:
@einalex it sounds like you haven't updated the kll git repo (I think I've fixed that issue).

@kdayns @dfaught thanks! I'll look at this over the next couple days.

einalex, 2017-10-03:
I did update the kll git repo, before that I got kll compile errors.

einalex, 2017-10-03:
commit 668027c4050c63c051a126bf3ff30002b2a4bf28 (HEAD, origin/master, origin/HEAD)
Author: Jacob Alexander <haata@kiibohd.com>
Date:   Mon Oct 2 23:14:38 2017 -0700

einalex, 2017-10-03:
I interpret that as some boundary check or loop problem.
When I press the innermost row keys on the right side, some check prohibits the codes from getting generated, when I press the innermost row keys on the left side, it sends keys for both halves...so the check/boundary for the right half would have to be moved one over. (Only happens in master slave mode btw, if I run the ride side in master mode, all is well on that half)

metsie2912, 2017-10-04:
I have similar problems like dfaught.
I'm using arch linux and all compile successfully. But my left keyboard has the keys shifted to the left. (e=r, w=e, s=d, etc.) The right part seems to work fine though. The exact same layout compiles correctly using the older software (following danspiff's instructions). Attached you'll find my kll files and ergodox.bash.
[kiibohdlayout.tar.gz](https://github.com/kiibohd/controller/files/1355254/kiibohdlayout.tar.gz)

dfaught, 2017-10-04:
@metsie2912, I know longer experience the offset keys after haata's latest commit: https://github.com/kiibohd/controller/commit/c25f15ccfde4667ff73691a9bf9105b6d5ae7229

I only have a compilation error for the right half.  Have you updated your kll folder and recompiled?

metsie2912, 2017-10-04:
@dfaught Unfortunately yes. I tried it again just to be sure. I'm using the following commits:
* kiibohd/controller
3299f4d (HEAD -> master, origin/master, origin/HEAD) Fixing more unit tests
* kll
cb1e0dd (HEAD -> master, origin/master, origin/HEAD) Fixing offset issue generating Partial Map trigger lists
I don't have compilation errors.

einalex, 2017-10-05:
I looked into the generatedKeymap.h and found what's causing the double sent keys. On the .h file for the left half, the respective keys get additional Macros assigned:
	/* 0 */ Define_TM( 0, 56 ), // (S3) : (HID(USBCode)"KEY_BACKSLASH");           <-- this shoud be on another key.
	/* 1 */ Define_TM( 0, 59 ), // (S3) : (HID(USBCode)"KEY_BACKTICK");
	/* 2 */ Define_TM( 0, 121 ), // (S3) : (layerShift(6) + LCDLayerDisplay());

haata, 2017-10-10:
I've fixed some more issues. Still need to verify if I've fixed the problems here.

haata, 2017-10-10:
I believe I've fixed this issue. @kdayns @dfaught please let me know if you're still seeing issues with the latest controller and kll repos.

LaikeSF, 2017-10-10:
Seems like I've hit a snag with the latest controller and kll repos. Using the default setting compiles okay, but something about my old layouts is throwing up errors.

`-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/x/ErgoDox/controller/Keyboards/linux-gnu.ICED-R.gcc.make
[  3%] Generating KLL Layout
kll BETA 0.5c.43a671b99cdd0a92ce3ad0d783764ba94e12f44b - 2017-10-10 00:23:21 -0700
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/ISSILed/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/MatrixARM/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/STLcd/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/UARTConnect/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Macro/PartialMap/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Macro/PixelMap/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Output/HID-IO/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Output/pjrcUSB/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/scancode_map.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/rightHand.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/slave1.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/leftHand.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-0.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/infinity_ergodox/lcdFuncMap.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-1.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-2.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-3.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-4.kll'
Scanning dependencies of target kiibohd
[  6%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/ISSILed/i2c.c.o
[  9%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/ISSILed/led_scan.c.o
[ 12%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/MatrixARM/matrix_scan.c.o
[ 15%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/STLcd/lcd_scan.c.o
[ 18%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/UARTConnect/connect_scan.c.o
[ 21%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/macro.c.o
[ 25%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/result.c.o
[ 28%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/trigger.c.o
[ 31%] Building C object CMakeFiles/kiibohd.dir/Macro/PixelMap/pixel.c.o
[ 34%] Building C object CMakeFiles/kiibohd.dir/generatedPixelmap.c.o
[ 37%] Building C object CMakeFiles/kiibohd.dir/Output/HID-IO/hidio_com.c.o
[ 40%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/output_com.c.o
[ 43%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_desc.c.o
[ 46%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_dev.c.o
[ 50%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_keyboard.c.o
[ 53%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_mouse.c.o
[ 56%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_rawio.c.o
[ 59%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_serial.c.o
[ 62%] Building C object CMakeFiles/kiibohd.dir/Debug/cli/cli.c.o
Linking C executable kiibohd.elf
Creating dfu binary file: kiibohd.dfu.bin
Creating secure dfu binary file: kiibohd.dfu.bin
Setting new key
Key Prepended: 0000000000000000
Creating Extended Listing:     kiibohd.lss
Creating Symbol Table:         kiibohd.sym
[ 93%] Built target kiibohd
[ 96%] Message for DFU targets
NOTE: kiibohd.secure.dfu.bin is required for secure Kiibohd Bootloaders.
Only keyboards from late 2017 and onwards, support this.
See lsusb -d 1c11: -v and look for iInterface 4 Kiibohd DFU Secure.
Otherwise, use the kiibohd.dfu.bin file when it says iInterface 4 Kiibohd DFU.
Use the physical reset button on the back of the keyboard to temporarily disable secure mode.
NOTE2: Secure mode is disabled currently, but will be enabled when key negotiation is supported.
[ 96%] Built target DFUMessage
[100%] Chip usage for mk20dx256vlh7
         SRAM:  16%     10956/65536     bytes
        Flash:  23%     59000/253952    bytes
[100%] Built target SizeAfter
Firmware has been compiled into: 'linux-gnu.ICED-R.gcc.make'
/mnt/x/ErgoDox/controller/Keyboards
laike@DESKTOP-N0037SR:/mnt/x/ErgoDox/controller/Keyboards$
laike@DESKTOP-N0037SR:/mnt/x/ErgoDox/controller/Keyboards$ ./ergodox.bash
linux-gnu/Linux is supported.
[... 81 lines cut ...]
ERROR: 'HID(USBCode)"KEY_LCK2"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK3"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK4"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK5"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK6"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK7"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: Invalid stage status 'Incomplete' for 'CodeGenerationStage'.
make[2]: *** [generatedKeymap.h] Error 1
make[1]: *** [CMakeFiles/kiibohd.dir/all] Error 2
make: *** [all] Error 2`

The layout in question
`U"5" : U"Lock5";
U"4" : U"Lock4";
U"3" : U"Lock3";
U"2" : U"Lock2";
U"1" : U"Lock1";
U"EQUALS" : flashMode();
U"T" : U"0";
U"R" : U"9";
U"E" : U"8";
U"W" : U"7";
U"Q" : U"6";
U"6" : U"Lock6";
U"7" : U"Lock7";`

einalex, 2017-10-10:
works for me :) thank you!

[17 more replies]