kiibohd/controller #206: Keys getting stuck and keyboard freezing

daantjie, 2017-07-18:
~~~
                   -`                 
                  .o+`                 daniel@tony
                 `ooo/                 OS: Arch Linux 
                `+oooo:                Kernel: x86_64 Linux 4.11.9-1-ARCH
               `+oooooo:               Uptime: 2h 51m
               -+oooooo+:              Packages: 1540
             `/:-:++oooo+:             Shell: zsh 5.3.1
            `/++++/+++++++:            Resolution: 1920x1080
           `/++++++++++++++:           DE: XFCE4
          `/+++ooooooooooooo/`         WM: Xfwm4
         ./ooosssso++osssssso+`        WM Theme: axiom
        .oossssso-````/ossssss+`       GTK Theme: Xfce-4.2 [GTK2]
       -osssssso.      :ssssssso.      Icon Theme: gnome
      :osssssss/        osssso+++.     Font: Sans 10
     /ossssssss/        +ssssooo/-     CPU: Intel Core i7-2600S @ 8x 3.8GHz [43.0°C]
   `/ossssso+/:-        -:/+osssso+-   GPU: intel
  `+sso+:-`                 `.-/+oso:  RAM: 3132MiB / 7875MiB
 `++:.                           `-/+/
 .`                                 `/
~~~

It's now getting ridiculous; *extremely* often now, while I'm typing, keys suddenly become stuck and continue pressing (meaning I have to unplug and then replug the 'dox), or the kkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkeyboard (it                                                                                                                            just happened twice, again after it with the spacebar) suddenly doesn't accept any input, although it does still switch layers.

I don't seem to be able to reliably reproduce it; sometimes it happens almost immediately after I plug the keyboard back in, and sometimes it can be a couple of minutes before it ‘triggers’.

The funny thing is, I *think* it still works on Windows. I'll be able to check this tomorrow, but I had no problems at work earlier today.

To add to the strangeness, I had no problems with the 'dox before I recently started bringing it into work and plugging it into the Windows machines there. (I'm not sure that's the easiest thing to debug, but maybe it'll help!)

I'm getting quite worked up about it, because I love this keyboard! And it's never gone wrong before (apart from needing to be un/replugged at startup, which I kinda accepted as a long-term bug). Help!

jceaser, 2017-07-19:
I think this is the same issue that I am having and reported in issue #196, however for me it always happens. Are you using the latest software and the master branch?

daantjie, 2017-07-19:
@jceaser; I'm unable to build master (and besides, it's apparently broken in a different way -- #204). So I'm running *either* the normal I:C configurator, *or* the alpha version of the configurator.

crystalhand, 2017-07-19:
Just an FYI the latest build does not work correctly with the infinity ergodox.  The latest one that works is c7fb6d1. So i would recommend using that or an earlier build.

I have not experienced the issue that you have described with my ergodox.  I use it in windows 10.

If you could somehow replicate it when running the serial debugger you can see if the key release command is not registering.

daantjie, 2017-07-19:
I can confirm that I don't get the problem with Windows 10. I've just booted up this Linux machine, so I'm waiting to see if the problem's still here. If it is, I'll run the serial debugger like you (@crystalhand) said.

surlyjake, 2017-07-21:
My experience has been that 'stuck key' issue frequently as well, but for me, it only happens if the KBD was plugged in during boot. If i unplug and replug it in once the OS is running, the keyboard works properly. I experience this in Win7 x64 and arch linux x86_64 on the same hardware. 

I wonder if the stuck key and NKRO symptoms are related (https://github.com/kiibohd/controller/wiki/Known-USB-Issues). take a look at the BOOT/NKRO mode with the `kbdProtocol` command.  For me, it says 'BOOT' if the kb is plugged in during boot, and 'NKRO' after i unplug-replug. @daantjie , you mentioned that you DO unplug-replug, but are you sure that it's making its way into NKRO mode reliably?

daantjie, 2017-07-22:
@surlyjake could you show me how to use the `kbdProtocol` command?

surlyjake, 2017-07-22:
@daantjie connect to the serial debugger port as documented here https://github.com/kiibohd/controller/wiki/Debugging. Then type kbdProtocol and press enter.

daantjie, 2017-07-23:
@surlyjake okay, it's telling me I'm in NKRO right now. (When booting up without the dox plugged in.) And I'm not currently experiencing the problem.

daantjie, 2017-07-25:
@surlyjake still in NKRO even when booting up with the dox plugged in, then un/re-plugging. (Haven't yet experienced the problem.)

chgue, 2017-07-27:
My DIY, hand-wired keyboard experiences the same problem with stuck keys on Windows 7 and Kubuntu 16.10 64-bit. Issue seems to occur mostly when moving/strafing in Overwatch or other shooters (multiple keys are pressed at the same time or in rapid succession). 

Replugging fixes the issue. kbdProtocol shows the keyboard being in BOOT mode before replugging it. Afterwards it's in NKRO as expected.

crystalhand, 2017-07-27:
So it sounds like the ergodox is crashing and then rebooting into boot mode or something similar.  The key is being held down because the down action for the key was registered but not the release.  My knowledge of how hardware input works is rather limited so I am unsure of where the command is being held ie keyboard, motherboard, os.  In my quick testing, I was not able to force it to hold a key when entering boot mode through keystroke or reset button.
The write up at [https://github.com/kiibohd/controller/wiki/Known-USB-Issues](url) mentions something about having USB mass storage plugged in.  Is this the case for anyone with the issue? 

Can people replicate this issue across multiple different computers (regardless of os)?  The main thing I am curious about is if the issues occurs with different motherboards/BIOS.

haata, 2017-10-03:
@daantjie are you using the firmware from the configurator? Or compiled yourself?

haata, 2019-02-18:
I've fixed a lot of these issues in the past year and a bit (including some really nasty ones that required perfect timing to hit them...).
Please reopen if flashing with the latest firmware doesn't fix the issue.

closed 2019-02-18 by haata