kiibohd/controller #212: Infinity Ergodox: code transfered, but not working

olivierpons, 2017-09-19:
I'm trying with the latest source code:

    git clone https://github.com/kiibohd/controller.gi
    cd Keyboards
    ./ergodox.bash

Plug the left hand of the Infinity Ergodox, push reset and do:

    sudo dfu-util -D linux-gnu.ICED-L.gcc.make/kiibohd.dfu.bin

It works, it's transfered, but my ergodox doesnt work and send bad keys (just typing once sends 2 keys for almost all the keyboards keys).
Any solution?

jceaser, 2017-09-19:
I don't think ergo dox fixes have been merged into the master branch, I have followed the suggestions of the user "danspiff" in ticket https://github.com/kiibohd/controller/issues/204 and have gotten working code.  Try that and report back your success. Now, the code is not perfect at that branch, but my keyboard is usable. and I'm typing on it now. I'm not sure what progress is being made to resolve this.

olivierpons, 2017-09-19:
This is a burning shame. It's like a car's vendor who sells car and say "ok you've bought you car 8 months ago, now you just can use its basic functions, even though you paid the maximum price, because we're working on other stuff".

How can someone be a professional and sells things like this? As an amateur, as an enthusiast, this is ok. But when people **pay** for something they should get the quality you pay for. The Infinity Ergodox is one of the most expensive keyboard of the world, and you dont have the tools to customize it *properly*. What a shame.

jceaser, 2017-09-19:
Amateur, I think that is by definition, I agree with you, but again by definition. we all told the "man" to f off when we signed up with this outfit. However, this group can improve it's image by communicating what the plan is, I don't fault anyone for wanting to write 0.5, I love the concept, but it should not have been merged into master till it was ready to be in master. kiibohb can solve this in my opinion by

1. creating a branch for each working keyboard till it all gets worked out
2. showing a plan to deal with this, (road map maybe with rough dates)
3. ask for help, people like me might be able to get up to speed on the code if we know what is wrong
4. build a community like olkb maybe even on redit or some other public place. IRC is not helpful for many.

As for your statement about code not working, have you tried going back to the specific revision mentioned in #204? That worked for me.

olivierpons, 2017-09-19:
Hi jceaser,

You're 100% right!

One branch for each keyboard (maybe each version), and let it like this: "working branch". And as soon as it's working, dont touch it for this particular keyboard (kind of a "frozen branch" I dont know if its feasible with git).

This way, each keyboard could clone its own "100% working" branch, or maybe try the main branch.

As for the test I'm not motivated anymore, I still have a lot of things to do for my job, because I'm serious and it's a job, I get paid for this so I'm trying to make a good job (see what I mean?).

Thank you very much for your commitment, but **they have to solve the bug on their own** because it's about money, more than 2000 people paid for the Infinity Ergodox. They **_paid_**. They have to make this working for the Infinity Ergodox. **It's their job, it's no longer a hobby**, they should have realized this two years ago (until now).

jceaser, 2017-09-19:
While you wait, you can install QMK.

olivierpons, 2017-09-19:
Already done QMK. But they dont support they new chip:

```
sudo dfu-programmer mk20dx256vlh7 flash ergodox_infinity_default.hex
[sudo] password for olivier: 
Unsupported target 'mk20dx256vlh7'.
targets:
    at89c51snd1c       at89c51snd2c       at89c5130          at89c5131       
    at89c5132          at90usb1287        at90usb1286        at90usb1287-4k  
    at90usb1286-4k     at90usb647         at90usb646         at90usb162      
    at90usb82          atmega32u6         atmega32u4         atmega32u2      
    atmega16u4         atmega16u2         atmega8u2          at32uc3a0128    
    at32uc3a1128       at32uc3a0256       at32uc3a1256       at32uc3a0512    
    at32uc3a1512       at32uc3a0512es     at32uc3a1512es     at32uc3a364     
    at32uc3a364s       at32uc3a3128       at32uc3a3128s      at32uc3a3256    
    at32uc3a3256s      at32uc3a4256s      at32uc3b064        at32uc3b164     
    at32uc3b0128       at32uc3b1128       at32uc3b0256       at32uc3b1256    
    at32uc3b0256es     at32uc3b1256es     at32uc3b0512       at32uc3b1512    
    at32uc3c064        at32uc3c0128       at32uc3c0256       at32uc3c0512    
    at32uc3c164        at32uc3c1128       at32uc3c1256       at32uc3c1512    
    at32uc3c264        at32uc3c2128       at32uc3c2256       at32uc3c2512    
    atxmega64a1u       atxmega128a1u      atxmega64a3u       atxmega128a3u   
    atxmega192a3u      atxmega256a3u      atxmega16a4u       atxmega32a4u    
    atxmega64a4u       atxmega128a4u      atxmega256a3bu     atxmega64b1     
    atxmega128b1       atxmega64b3        atxmega128b3       atxmega64c3     
    atxmega128c3       atxmega256c3       atxmega384c3    

```

olivierpons, 2017-09-20:
Hi @jceaser I have followed the comment with 10 steps of the user "danspiff" in ticket #204.

I've added a heart too because it seems to work.
I'll try to customize tonight my Infinity Ergodox. I'll tell you more tonight.
Do you have any link to a tutorial to make macros on the Infinity using kll "language"?

Nevertheless, this is not normal at all to sell products and claim they're customizable without giving the "working" tools to customize them.

jceaser, 2017-09-20:
I agree that it is not normal, I hope that by leaving polite comments here someone will notice and try to improve things. As I said, I like the ideas here, but QMK is run and functions so much better. I hear mass drop is mad at them now too, so something is going on. Lets hope they fix things.

As for macros, KLL has "capabilities" but I have not actually written any:

https://github.com/kiibohd/controller/wiki/Layers,-Macros

olivierpons, 2017-09-20:
I'm sorry it's not my kind of behavior to be aggressive, but when you sell product to a scale like 1000 keyboards / months you have to handle problems and bugs in a more professional way and this is not the case right now.

FYI, here's what I've done:
```
git clone --recursive https://github.com/qmk/qmk_firmware.git
cd qmk_firmware/
```
Then I've plugged the right side of my Infinity Ergodox, pushed its Reset button and typed:
```
sudo make ergodox_infinity-default-dfu-util MASTER=right
```
Then same opération for the left:
```
sudo make ergodox_infinity-default-dfu-util MASTER=left
```
Not only everything ran smoothly, but the LCD screen:
- changes its whole image every 2s, this implies that you can change the **whole image however you like** (on the contrary to kiibohd where you just can change layer 0)
- is off all the time except when you go to layer 1, where it's RED. This implies the code can change the colors too.

And the macro which prints the version of the qmk firmware is working too! Press one key = send the version of the qmk firmware.

I guess this will code will be hard for me to understand, but I'm pretty sure I'll be able to do whatever I want with the qmk firmware.

So I'll do what I do for my old Ergodox (not infinity): go for the qmk firmware!

haata, 2017-09-20:
Sorry all, I've been neglecting these problems, I was hoping that I'd be
able to jump on them quickly... Unfortunately, the legal issues with
Massdrop kept getting worse and took up more and more of my time...
However, the ergodox issues are my priority now. So hopefully I'll have
some fixes in the next few days.

ur5us, 2017-09-20:
@haata Out of curiosity, can you elaborate a bit on “legal issues with Massdrop” or do you have a link, probably to Massdrop’s forum where these matters are discussed?

haata, 2017-09-20:
https://input.club/official-halo-switch-statement/ is where most of the info is collected.

kemal1923, 2017-09-21:
@haata Just wondering why you entered into a revokable license back?  This allows Massdrop to terminate your license for any or no reason.

haata, 2017-09-21:
@kemal1923 I can't really go into details about the contract (there are legal implications on doing that). But it's not black and white like that in contract law. And as of now, the license has not been revoked. We were just refused our order of switches (these are two very different things).

gdelugre, 2017-09-29:
I concur with @olivierpons, as a customer of an expensive product one should expect a working firmware.

At the very least there should be a stable branch, as well as stable releases with version numbers and changelogs. I should be able to know which version I am running, whether it is stable or not, and not just a commit number. Rolling back to the latest known stable version should be something easily doable.

Having a stable branch would also allow more people to get involved in the code. Right now compiling and flashing the latest version just makes your keyboard unusable. That makes it totally discouraging to even start looking into it.

IMO, that would be standard development practices for a commercial product, and it would benefit both IC as well as their customers.

haata, 2017-09-29:
Btw, there is a stable branch: https://github.com/kiibohd/controller/tree/old

But yeah, I agree. I should have handled this more nicely. I was hoping to push this giant merge of the K-Type update (which includes firmware and a totally rewritten KLL compiler) and stomp out all the bugs. Then that didn't happen...so yeah, I'm sorry.

However, Infinity Ergodox is now my top priority (all related bugs) as it's gating the release of the new configuration software we're working on. I've found some pretty deep KLL compiler issue which has held back fixes for a while (fortunately starting to get to the bottom of it).

olivierpons, 2017-09-29:
FYI I've done exactly what I want with my keyboard using qmk firmware. I'm using right now my infinity ergodox to type. It's exactly the same, I never watch my hand and never watch the LCD it's just to show off actually. And to be honest, in the dark, too flashy colors are annoying, to less flashy colors, like it's done with the qmk, are better.

commit c25f15cc (2017-10-03): Small fix on the master interconnect node

Needs a recent kll compiler.

Should fix #212, #204
closed 2017-10-03 by haata