kiibohd/controller #218: Invalid USB HID code on Compile

LaikeSF, 2017-10-10:
Operating system: Windows 10
Compile method: Bash on Ubuntu on Windows 10 

`-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/x/ErgoDox/controller/Keyboards/linux-gnu.ICED-R.gcc.make
[ 3%] Generating KLL Layout
kll BETA 0.5c.43a671b99cdd0a92ce3ad0d783764ba94e12f44b - 2017-10-10 00:23:21 -0700
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/ISSILed/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/MatrixARM/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/STLcd/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Devices/UARTConnect/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Macro/PartialMap/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Macro/PixelMap/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Output/HID-IO/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Output/pjrcUSB/capabilities.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/scancode_map.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/rightHand.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/slave1.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/Scan/Infinity_Ergodox/leftHand.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-0.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/infinity_ergodox/lcdFuncMap.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-1.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-2.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-3.kll'
ERROR: Failed to write to file '/mnt/x/ErgoDox/controller/kll/layouts/mine/MDErgo1-Default-4.kll'
Scanning dependencies of target kiibohd
[ 6%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/ISSILed/i2c.c.o
[ 9%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/ISSILed/led_scan.c.o
[ 12%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/MatrixARM/matrix_scan.c.o
[ 15%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/STLcd/lcd_scan.c.o
[ 18%] Building C object CMakeFiles/kiibohd.dir/Scan/Devices/UARTConnect/connect_scan.c.o
[ 21%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/macro.c.o
[ 25%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/result.c.o
[ 28%] Building C object CMakeFiles/kiibohd.dir/Macro/PartialMap/trigger.c.o
[ 31%] Building C object CMakeFiles/kiibohd.dir/Macro/PixelMap/pixel.c.o
[ 34%] Building C object CMakeFiles/kiibohd.dir/generatedPixelmap.c.o
[ 37%] Building C object CMakeFiles/kiibohd.dir/Output/HID-IO/hidio_com.c.o
[ 40%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/output_com.c.o
[ 43%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_desc.c.o
[ 46%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_dev.c.o
[ 50%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_keyboard.c.o
[ 53%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_mouse.c.o
[ 56%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_rawio.c.o
[ 59%] Building C object CMakeFiles/kiibohd.dir/Output/pjrcUSB/arm/usb_serial.c.o
[ 62%] Building C object CMakeFiles/kiibohd.dir/Debug/cli/cli.c.o
Linking C executable kiibohd.elf
Creating dfu binary file: kiibohd.dfu.bin
Creating secure dfu binary file: kiibohd.dfu.bin
Setting new key
Key Prepended: 0000000000000000
Creating Extended Listing: kiibohd.lss
Creating Symbol Table: kiibohd.sym
[ 93%] Built target kiibohd
[ 96%] Message for DFU targets
NOTE: kiibohd.secure.dfu.bin is required for secure Kiibohd Bootloaders.
Only keyboards from late 2017 and onwards, support this.
See lsusb -d 1c11: -v and look for iInterface 4 Kiibohd DFU Secure.
Otherwise, use the kiibohd.dfu.bin file when it says iInterface 4 Kiibohd DFU.
Use the physical reset button on the back of the keyboard to temporarily disable secure mode.
NOTE2: Secure mode is disabled currently, but will be enabled when key negotiation is supported.
[ 96%] Built target DFUMessage
[100%] Chip usage for mk20dx256vlh7
SRAM: 16% 10956/65536 bytes
Flash: 23% 59000/253952 bytes
[100%] Built target SizeAfter
Firmware has been compiled into: 'linux-gnu.ICED-R.gcc.make'
/mnt/x/ErgoDox/controller/Keyboards
laike@DESKTOP-N0037SR:/mnt/x/ErgoDox/controller/Keyboards$
laike@DESKTOP-N0037SR:/mnt/x/ErgoDox/controller/Keyboards$ ./ergodox.bash
[... 84 lines cut ...]
ERROR: 'HID(USBCode)"KEY_LCK4"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK5"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK6"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: 'HID(USBCode)"KEY_LCK7"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: Invalid stage status 'Incomplete' for 'CodeGenerationStage'.
make[2]: *** [generatedKeymap.h] Error 1
make[1]: *** [CMakeFiles/kiibohd.dir/all] Error 2
make: *** [all] Error 2`

The layout in question
U"5" : U"Lock5"; 
U"4" : U"Lock4"; 
U"3" : U"Lock3"; 
U"2" : U"Lock2"; 
U"1" : U"Lock1"; 
U"EQUALS" : flashMode(); 
U"T" : U"0"; 
U"R" : U"9"; 
U"E" : U"8"; 
U"W" : U"7"; 
U"Q" : U"6"; 
U"6" : U"Lock6"; 
U"7" : U"Lock7";

As a temporary fix, I've added infinity_ergodox/lcdFuncMap to the problematic layout. Seems to have sorted the issue out for now.

kll, 2017-11-01:
I'm getting this now as well. It is specifically when I try to map U"Equals" to anything else. It results in

```
ERROR:  'HID(USBCode)"KEY_FUN6"' Invalid USB HID code, missing FuncMap layout (e.g. stdFuncMap, lcdFuncMap)
ERROR: Invalid stage status 'Incomplete' for 'CodeGenerationStage'.
```

If I comment out the line mapping U"Equals" it will compile. This is on my DefaultMap which already includes infinity_ergodox/lcdFuncMap so that mentioned work around isn't helping me.

haata, 2017-11-01:
Hmm, I've seen this a few times now. Looks like a kll compiler bug.

Kevin, can you post your layout files for me?

haata, 2017-11-02:
Should be fixed with https://github.com/kiibohd/kll/commit/5002a268e8504b6ddc926e37e4277a2d4b0c5049

closed 2017-11-02 by haata

kll, 2017-11-02:
I actually can't reproduce it now. I started again with a fresh clone of master in both controller and kll and have created more of my layout than I had before and haven't hit this issue again. I must have messed up and not gotten the latest kll before, but I really thought that I had. If I hit it again I'll be sure to save everything.

kll, 2017-11-02:
Hah, I totally missed your update saying you fixed it. Ignore my stupid reply saying I couldn't reproduce it. 😄